Ysurac/openmptcprouter #380: Bandwidth monitor

Ajowan84, 2019-02-26:
Suggestion:

Add a bandwidth monitor like Netlink but with more features.
data quota per day
data quota per users
eMail warning when a certain amount data is used.

A good thing for 4G connection users with limited data

alexbidule, 2019-02-27:
+1 on this feature !

Also having a quota per week day (Monday, Tuesday, Wednesday...) would be great. On my side I have unlimited data in weekends :-)

Gordon0472, 2019-05-30:
Hi,
+1 on this feature for 4G users !!!

Is it possible to install Yamon (http://usage-monitoring.com/index.php) for example : ?
It perfectly matches what I'm looking for ...

Best regardes

jayanta525, 2019-05-30:
> Is it possible to install Yamon (http://usage-monitoring.com/index.php) for example : ?

Yamon is not maintained properly. Targetted for DD-WRT users. OpenWrt has many inferior packages.
For bandwidth monitoring, check out `luci-app-nlbwmon` and `luci-app-wrtbwmon`.

luci-app-nlbwmon provides traffic per host and usage according to protocols. Also supports IPv6.

[image]

luci-app-wrtbwmon provides live data usage tracking of clients along with the total usage.

[image]

> Also having a quota per week day (Monday, Tuesday, Wednesday...) would be great. On my side I have unlimited data in weekends :-)

OMR does have monthly quota support. This could easily be adjusted for daily quota.

Ajowan84, 2019-05-30:
Thank you so much jayanta525 !
I installed both packages.
luci-app-wrtbwmon works perfectly but luci-app-nlbwmon does not display anything.
it stays on "Data Recovery ..."

jayanta525, 2019-05-30:
Check whether nlbwmon startup script is enabled.

Ajowan84, 2019-05-30:
How to check this ?

I used this to start nldwmon :
/etc/init.d/nlbwmon enable
/etc/init.d/nlbwmon start

But i don't know if is ok...

Ajowan84, 2019-05-31:
When i install nldwmon, i have the error "uci - Entry not found"

closed 2019-07-04 by Ysurac