Ysurac/openmptcprouter #381: Add a speedtest from router to server with all interface

Roken120, 2019-02-28:
Hello,
I have a recommandation you can add a speed monitor from router to server for test the speed of all link and aggregation ?

RomanJos, 2019-03-03:
Thats what i think too a speedtest on each wan would be awesome !

forid786, 2019-03-25:
Iperf? It's already installed.

Roken120, 2019-04-22:
Yes,
It would be nice to be able to test directly with the openmctpvps in order to check the aggregation
Example:
WAN1 -> VPS
WAN2 -> VPS
Aggregation -> VPS
For check the aggregation is correct

Ysurac, 2019-04-22:
True Iperf is installed on both VPS and OpenMPTCProuter side.
```omr-iperf``` can be used to check aggregation speed, in next version it should be able to test link speed.

Ysurac, 2019-04-22:
Can be used in fact by using ```omr-iperf vps -B ipofwan``` (and ```omr-iperf vps``` to only check aggregation).

closed 2019-07-04 by Ysurac