Ysurac/openmptcprouter #382: Speed error

Roken120, 2019-02-28:
## Expected Behavior
My download dont up 50mbps...

I have two 100mbps download an one 4G link 

## Steps to Reproduce the Problem

  1.Speed test

## Specifications

  - OpenMPTCProuter version: 0.46.3
 - OpenMPTCProuter VPS version: 0.95
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-02-28:
Disable SQM if enabled.

Roken120, 2019-02-28:
SQM is disable on all interface.

Ysurac, 2019-02-28:
I would need the result of ```uci show network``` via SSH on OpenMPTCProuter.
You can also test wan speed on OpenMPTCProuter via SSH using ```omr-test-speed wan1``` where wan1 is the interface name.

Roken120, 2019-02-28:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].ipset='/tpeweb1.e-transactions.fr/omr_rules_dst_bypass_all,omr_rules6_dst_bypass_all' '/tpeweb.e-transactions.fr/omr_rules_dst_bypass_all,omr_rules6_dst_bypass_all' '/e-transactions.fr/omr_rules_dst_bypass_all,omr_rules6_dst_bypass_all' '/all/omr_rules_dst_bypass_all,omr_rules6_dst_bypass_all'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].network='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].network='wan' 'wan6' 'wan1' 'wan2' 'wan3'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
firewall.@rule[0].proto='udp'
firewall.@rule[0].dest_port='68'
firewall.@rule[0].target='ACCEPT'
firewall.@rule[0].family='ipv4'
firewall.@rule[1]=rule
firewall.@rule[1].name='Allow-Ping'
firewall.@rule[1].src='wan'
[... 951 lines cut ...]
unbound.@unbound[0].protocol='ip4_only'
unbound.@unbound[0].enabled='1'
unbound.@unbound[0].recursion='aggressive'
upnpd.config=upnpd
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.enabled='1'
upnpd.config.uuid='6cf30390-c525-44f4-a412-c19ea66a90e2'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth0' 'wan1' 'wan2' 'wan3'

Roken120, 2019-02-28:
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='1'
network.globals=globals
network.globals.ula_prefix='fd0e:38bf:d744::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='5'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.congestion='bbr'
network.globals.mptcp_scheduler='default'
network.globals.mptcp_fullmesh_num_subflows='1'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='00:26:b9:e7:c2:23'
network.lan.metric='2'
network.lan.ipv6='0'
network.lan.ipaddr='192.168.0.33'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.defaultroute='0'
network.wan1.type='macvlan'
network.wan1.ifname='wan1'
network.wan1.masterintf='eth0'
network.wan1.metric='3'
network.wan1.netmask='255.255.255.0'
network.wan1.macaddr='6e:7a:c4:dc:91:05'
network.wan1.ipaddr='192.168.1.70'
network.wan1.gateway='192.168.1.1'
network.wan1.multipath='on'
network.wan2=interface
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.defaultroute='0'
network.wan2.type='macvlan'
network.wan2.ifname='wan2'
network.wan2.masterintf='eth0'
network.wan2.metric='4'
network.wan2.netmask='255.255.255.0'
network.wan2.macaddr='d6:1d:46:9a:bd:37'
network.wan2.gateway='192.168.3.1'
network.wan2.ipaddr='192.168.3.50'
network.wan2.multipath='master'
network.wan3=interface
network.wan3.proto='static'
network.wan3.type='macvlan'
network.wan3.ip4table='wan'
network.wan3.defaultroute='0'
network.wan3.ifname='wan3'
network.wan3.masterintf='eth0'
network.wan3.metric='5'
network.wan3.macaddr='3e:26:43:55:b8:3a'
network.wan3.ipaddr='192.168.8.109'
network.wan3.netmask='255.255.255.0'
network.wan3.gateway='192.168.8.1'
network.wan3.multipath='on'
network.wan1_dev=device
network.wan1_dev.name='wan1'
network.wan1_dev.type='macvlan'
network.wan1_dev.ifname='eth0'
network.wan1_dev.macaddr='6e:7a:c4:dc:91:05'
network.wan2_dev=device
network.wan2_dev.name='wan2'
network.wan2_dev.type='macvlan'
network.wan2_dev.ifname='eth0'
network.wan2_dev.macaddr='d6:1d:46:9a:bd:37'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.proto='dhcp'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.metric='6'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.gateway='fe80::a00:1'
network.omr6in4.ip6addr='fe80::a00:2'
network.omr6in4.metric='7'
network.wan3_dev=device
network.wan3_dev.name='wan3'
network.wan3_dev.type='macvlan'
network.wan3_dev.ifname='eth0'
network.wan3_dev.macaddr='3e:26:43:55:b8:3a'

Roken120, 2019-02-28:
root@OpenMPTCProuter:~# omr-test-speed wan1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 67.6M    0     0  10.8M      0  0:15:39  0:00:06  0:15:33 11.2M

Roken120, 2019-02-28:
C
root@OpenMPTCProuter:~# omr-test-speed wan2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 86.6M    0     0  10.3M      0  0:16:25  0:00:08  0:16:17 11.2M

Roken120, 2019-02-28:
root@OpenMPTCProuter:~# omr-test-speed wan3
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 70.1M    0     0  2528k      0  1:09:06  0:00:28  1:08:38 3883k

Ysurac, 2019-02-28:
And ```omr-test-speed``` without wan interfaces ?

Roken120, 2019-02-28:
root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 5694k    0     0  7453k      0  0:23:26 --:--:--  0:23:26 6453k

Ysurac, 2019-02-28:
Not really good. What is the platform of OpenMPTCProuter (rpi, wrt, x86,...) and the VPS (atom,  memory,...)?

Roken120, 2019-02-28:
I have changed today to vm with 2 vcpu

Roken120, 2019-02-28:
Model	QEMU Standard PC (i440FX + PIIX, 1996)
Architecture	Intel(R) Xeon(R) CPU E5520 @ 2.27GHz
I changed to 2 socket and 2 core.

Roken120, 2019-02-28:
Is it possible that the problem comes from vps?

Roken120, 2019-02-28:
oot@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 10.0G    1  157M    0     0  6870k      0  0:25:26  0:00:23  0:25:03 6346k

Ysurac, 2019-02-28:
yes it's possible, you can try disable encryption on shadowsocks interface

Roken120, 2019-02-28:
Or am I going to disable that ?

Ysurac, 2019-02-28:
in services->shadowsocks-libev->distant server

Roken120, 2019-02-28:
root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  4 10.0G    4  430M    0     0  19.1M      0  0:08:54  0:00:22  0:08:32 22.1M

Much better by cons when I download with my computer download start very slowly

Roken120, 2019-02-28:
Retrieving speedtest.net configuration...
Testing from Netrix (154.49.213.44)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Orange (Paris) [15.19 km]: 27.568 ms
Testing download speed................................................................................
Download: 50.49 Mbit/s
Testing upload speed................................................................................................
Upload: 28.90 Mbit/s
And with speedtest i have

Ysurac, 2019-02-28:
speedtest download small amount of data, so it's not always very good to test MPTCP.
But I can say you have the VPS CPU or the VM CPU that limit the speed.

Roken120, 2019-02-28:
You think the problem comes from the server? because I always had the same and before I could reach 200 to 250mbps

Ysurac, 2019-02-28:
It's a server or a VPS ? If it's a VPS, CPU can be shared with many VPS.

Roken120, 2019-02-28:
VPS

Ysurac, 2019-02-28:
I don't know which side is the problem, but this seems to be a CPU power problem, encryption need some CPU.
I think I would need to make something to test CPU power with encryption...

Roken120, 2019-02-28:
Ok i wait.

Roken120, 2019-02-28:
I strongly think that the problem comes from the vps, I will inquire with the host

Ysurac, 2019-03-01:
You can try to run ```glorytun-udp bench chacha``` on OpenMPTCProuter and ```glorytun bench chacha``` on the VPS. You should then know where is the problem.

Roken120, 2019-03-01:
In vps : root@VPS:~# glorytun bench chacha

  bench      chacha20poly1305
  libsodium  1.0.16

  precision  2^(-10)
  bufsize    65536 bytes
  duration   1 second

------------------------------------------------------------
 2^n       min            avg            max          delta
------------------------------------------------------------
   0      5.52 Mbps      6.79 Mbps      7.00 Mbps  3.10e-03
   1     13.19 Mbps     13.20 Mbps     14.79 Mbps  1.09e-03
   2     24.99 Mbps     27.59 Mbps     28.42 Mbps  6.67e-03
   3     52.99 Mbps     53.94 Mbps     60.57 Mbps  1.18e-02
   4    101.42 Mbps    110.51 Mbps    112.68 Mbps  5.86e-03
   5    185.91 Mbps    185.91 Mbps    191.63 Mbps  5.62e-02
   6    356.03 Mbps    383.64 Mbps    405.76 Mbps  2.65e-01
   7    562.53 Mbps    585.50 Mbps    608.93 Mbps  1.26e-01
   8    760.34 Mbps    835.54 Mbps    835.98 Mbps  4.38e-01
   9    909.98 Mbps    937.45 Mbps    978.77 Mbps  2.80e-01
  10    999.12 Mbps   1060.45 Mbps   1062.80 Mbps  6.77e-01
  11   1140.94 Mbps   1156.12 Mbps   1204.34 Mbps  5.15e-01
  12   1172.14 Mbps   1191.83 Mbps   1278.05 Mbps  5.62e-01
  13   1061.93 Mbps   1180.06 Mbps   1214.66 Mbps  7.64e-01
  14   1248.90 Mbps   1260.51 Mbps   1286.59 Mbps  7.23e-01
  15   1289.71 Mbps   1290.34 Mbps   1344.88 Mbps  6.34e-01
  16   1216.87 Mbps   1236.41 Mbps   1267.86 Mbps  9.16e-01

Roken120, 2019-03-01:
bench      chacha20poly1305
  libsodium  1.0.16

  precision  2^(-10)
  bufsize    65536 bytes
  duration   1 second

------------------------------------------------------------
 2^n       min            avg            max          delta
------------------------------------------------------------
   0      4.52 Mbps      4.60 Mbps      4.72 Mbps  8.11e-04
   1      7.95 Mbps      9.12 Mbps      9.12 Mbps  1.55e-03
   2     18.55 Mbps     18.89 Mbps     18.89 Mbps  1.39e-02
   3     35.47 Mbps     37.35 Mbps     37.46 Mbps  1.69e-02
   4     70.37 Mbps     70.42 Mbps     76.74 Mbps  5.49e-02
   5    127.87 Mbps    138.21 Mbps    138.21 Mbps  9.47e-02
   6    267.35 Mbps    289.45 Mbps    289.45 Mbps  2.18e-01
   7    447.80 Mbps    461.08 Mbps    461.08 Mbps  2.25e-01
   8    947.74 Mbps    969.26 Mbps    969.26 Mbps  5.01e-01
   9   1399.16 Mbps   1412.80 Mbps   1413.88 Mbps  1.08e+00
  10   1842.70 Mbps   1877.88 Mbps   1878.20 Mbps  3.20e-01
  11   2125.61 Mbps   2183.70 Mbps   2218.32 Mbps  5.91e-02
  12   2418.07 Mbps   2447.67 Mbps   2515.08 Mbps  5.03e-01
  13   2428.09 Mbps   2543.76 Mbps   2623.81 Mbps  1.43e+00
  14   2645.00 Mbps   2688.12 Mbps   2688.12 Mbps  1.80e+00
  15   2687.54 Mbps   2687.54 Mbps   2725.83 Mbps  2.03e-01
  16   2678.33 Mbps   2699.50 Mbps   2699.50 Mbps  1.77e+00
In router

Roken120, 2019-03-01:
Where is problem ???

Roken120, 2019-03-01:
root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 10.0G    1  164M    0     0  7424k      0  0:23:32  0:00:22  0:23:10 5956k

Roken120, 2019-03-01:
C
root@OpenMPTCProuter:~# omr-test-speed wan1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 10.0G    2  262M    0     0  10.7M      0  0:15:48  0:00:24  0:15:24 11.0M

Roken120, 2019-03-01:
C
root@OpenMPTCProuter:~# omr-test-speed wan2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 89.8M    0     0  10.4M      0  0:16:19  0:00:08  0:16:11 10.6M

Roken120, 2019-03-01:
root@OpenMPTCProuter:~# omr-test-speed wan3
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 66.6M    0     0  5171k      0  0:33:47  0:00:13  0:33:34 6563k

Ysurac, 2019-03-01:
I have no idea for now...

closed 2019-07-04 by Ysurac