Ysurac/openmptcprouter #384: Download speed Container LXC

Dragonox77, 2019-03-03:
## Expected Behavior
Hi; Thank you for your work.

Get an aggregated connection.

## Actual Behavior
Actually, in the status page, All is green, no error BUT when I download, do speedtest etc, it only download with my wan1.

If you want to get acces to my Container LXC feel free to ask me, I'm hosting it on a dedicated server from OVH with an ip FO.

## Specifications

  - OpenMPTCProuter version: 0.46.3
  - OpenMPTCProuter VPS version: 0.95
  - OpenMPTCProuter platform: RPI3 x64

Ysurac, 2019-03-03:
The host have a MPTCP kernel ?

Dragonox77, 2019-03-03:
I think yes,
uname -a:
Linux openmptcpserver 4.15.18-11-pve #1 SMP PVE 4.15.18-33 (Tue, 05 Feb 2019 07:36:16 +0100) x86_64 x86_64 x86_64 GNU/Linux

Ysurac, 2019-03-03:
No. It's not a kernel with MPTCP support.

Dragonox77, 2019-03-04:
Ok sorry then, I will try the docker version.

Ysurac, 2019-03-04:
This will be the same problem with Docker. Docker also use the host kernel.

Dragonox77, 2019-03-04:
So for what is the docker version ?

Ysurac, 2019-03-04:
The docker version is an alpha version, not yet finished (no support for omr-admin yet).
The docker version will also be usable on host with MPTCP kernel.

Dragonox77, 2019-03-04:
So there is no way to have a working openmptcprouter on a container ? Sad.

Ysurac, 2019-03-04:
MPTCP is a kernel patch, so this work with containers if the host kernel is patched.
Else you can use OpenMPTCProuter without MPTCP, by using only Glorytun UDP or MLVPN, but it's slower.

Dragonox77, 2019-03-04:
I've set default vpn to glorytun UDP but it's not working too.

Ysurac, 2019-03-04:
shadowsocks need to be disabled.
If configuration is good this should work. Check logs if there is any errors.

Dragonox77, 2019-03-04:
When i disable shadowsocks, i can't go to any website and there is no error.

Dragonox77, 2019-03-04:
Like if there is a dns problem

Ysurac, 2019-03-04:
UDP DNS always use the VPN in every case.

Dragonox77, 2019-03-04:
So where the problem can come from ?

Ysurac, 2019-03-04:
From your configuration.
To check that I would need the result of ```uci show``` by SSH on OpenMPTCProuter (remove at least server ip)

Dragonox77, 2019-03-04:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.relier_box=dhcp
dhcp.relier_box.leasetime='12h'
dhcp.relier_box.limit='150'
dhcp.relier_box.interface='relier_box'
dhcp.relier_box.start='1'
dhcp.relier_box.dynamicdhcp='0'
dhcp.relier_box.dhcp_option='6,192.168.1.1'
dhcp.@domain[0]=domain
dhcp.tv_orange=dhcp
dhcp.tv_orange.leasetime='12h'
dhcp.tv_orange.limit='150'
dhcp.tv_orange.interface='tv_orange'
dhcp.tv_orange.start='10'
dhcp.tv_orange.dynamicdhcp='0'
dhcp.tv_orange.dhcp_option='6,192.168.1.1'
dhcp.@host[0]=host
dhcp.@host[0].dns='1'
dhcp.@host[0].gw='192.168.1.1'
dhcp.@host[0].leasetime='24h'
dhcp.@host[0].name='decodeur_orange'
dhcp.@host[0].mac='2C:39:96:57:89:C2'
dhcp.@host[0].ip='192.168.1.43'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[0].network='lan'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
[... 1046 lines cut ...]
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='tun0' 'wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='11'
wireless.radio0.hwmode='11g'
wireless.radio0.path='platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:000                                                                                                                                                             1:1'
wireless.radio0.htmode='HT20'
wireless.radio0.disabled='1'

Ysurac, 2019-03-04:
In the config shadowsocks is not disabled. So you must have errors in logs.
You can also have info about glorytun udp using ```glorytun-udp show dev tun0``` and ```glorytun-udp path dev tun0```

Dragonox77, 2019-03-04:
client tun0:
  pid:       28647
  bind:      0.0.0.0 port 5000
  peer:      193.70.101.33 port 65001
  mtu:       1450
  cipher:    chacha20poly1305

path UP
  status:   OK
  bind:     192.168.1.19 port 5000
  public:   90.3.138.25 port 5000
  peer:     193.70.101.33 port 65001
  mtu:      1472 bytes
  rtt:      28.003 ms
  rttvar:   13.486 ms
  upload:   18520 bytes/s (max: 27880)
  download: 23670 bytes/s (max: 27110)
  output:   10938 packets
  input:    5801 packets
path UP
  status:   OK
  bind:     192.168.2.160 port 5000
  public:   176.167.144.50 port 5000
  peer:     193.70.101.33 port 65001
  mtu:      1472 bytes
  rtt:      43.569 ms
  rttvar:   15.066 ms
  upload:   18410 bytes/s (max: 32860)
  download: 27490 bytes/s (max: 43060)
  output:   10121 packets
  input:    16626 packets

Dragonox77, 2019-03-04:
Also I reactivate shadowsock to send you this.

Ysurac, 2019-03-04:
This is not a correct IP: network.tv_orange.ipaddr='192.168.1.0'

Dragonox77, 2019-03-04:
This is just to set up a dhcp server for one peripheral that I use this.

Dragonox77, 2019-03-04:
Well, it's ok, i will install it on a kvm vps.
Have you a donation link ?

Ysurac, 2019-03-04:
Yes, on https://www.openmptcprouter.com/ via paypal, flattr, liberapay,...

closed 2019-03-04 by Ysurac