Ysurac/openmptcprouter #385: omr-service invoked oom-killer on VPS

user747, 2019-03-04:
## Expected Behavior
Sustained download without oom-killer being invoked

## Actual Behavior
oom-killer starts when using aria2 to download a large file , around 2-4GB

## Steps to Reproduce the Problem

  1. Download large file using https://aria2.github.io/
  2.  oom-killer kills omr-service, glorytun-tcp, python3, ss-server
  3. Verify via /var/log/kern.log

## Specifications

  - OpenMPTCProuter version: v0.46.3
  - OpenMPTCProuter VPS version: 93
  - OpenMPTCProuter platform: (RPI3
[kern.log](https://github.com/Ysurac/openmptcprouter/files/2928035/kern.log)
)

Ysurac, 2019-03-04:
You download over TCP or UDP ?

user747, 2019-03-04:
> You download over TCP or UDP ?

It looks like TCP over 80 to deploy.static.akamaitechnologies.com

See lots of this in capture from WAN's to VPS 65101
MPTCP [TCP Dup ACK]
MPTCP [TCP Retransmission]

As soon as the download starts, free memory goes from 700ish Mb to 0, and then crashes.

user747, 2019-03-04:
Tested downloading a 10Gb file http://www.ovh.net/files/10Gio.dat using Chrome, and memory usage on VPS is stable (around 600-650Mb free)

I think aria2 is culprit, but I don't understand why its draining memory so fast.

Ysurac, 2019-03-04:
I think aria2 is opening connections to destination website, maybe too much ?

user747, 2019-03-04:
> I think aria2 is opening connections to destination website, maybe too much ?

It's possible. Something with the way aria2 sending packets to ss-server too quickly, causing some sort of memory leak, and then crashing. 

I will do more testing.

user747, 2019-03-05:
Tested on a new Debian VPS with mptcp kernel 4.14.104, shadowsocks-libev-3.2.4.

Lots of memory still being used, but free memory didn't drop below 300Mb.

closed 2019-03-15 by Ysurac