Ysurac/openmptcprouter #387: Some URLS won't load with openmptcprouter

victorleduc, 2019-03-06:
OpenMPTCProuter is mostly working great since a few weeks, but I had issues on a few websites, where I always have a Timeout error on chrome. I know the website is working because i tried sharing my phone 4G and the website does load. As soon as I connect to my router with OpenMPTCProuter, Timeout error :(

Here is an exemple of a website that does not work : https://agenda2.securitest.org/login

[image]

Can you check if it is working on your OpenMPTCProuter installation ?
And what information do you need to be able to see where is the issue ?

## Specifications

  - OpenMPTCProuter version: v0.46.3
  - OpenMPTCProuter VPS version: 0.93
  - OpenMPTCProuter platform: Linksys WRT3200ACM

Ysurac, 2019-03-06:
You can try omr-bypass, the website may ban VPS IPs.

cron410, 2019-03-14:
I have the same issue with Github website. I tested this on a server at work. Only have this problem after upgrading VPS script to .96 and MPTCProuter to .47 yesterday.

Ysurac, 2019-03-14:
Check if it's not an IPv6 issue if it's enabled, else you may have to bypass it but I don't think Github ban some IP so easily.

cron410, 2019-03-15:
Neither of my connections support IPv6 and it is disabled in the Settings Wizard. Is there something else I can do besides OMR bypass?

Ysurac, 2019-03-15:
You could check from the VPS if you can access the website (using curl for example).

cron410, 2019-03-15:
I went and unplugged that test server, plugged in my laptop with the same public IP set and was able to access the site normally. I also have not made any changes to firewall. It was installed on a fresh Ubuntu 18.04 VM connected directly to the internet with a public static IPv4.

xabix, 2019-03-16:
Hello, I have the same problem with some sites. If I do OMR bypass it works but I don't understand why I need to do this. Example with mutuelledesmotards.fr it's not an issue of DNS. I have also all ipv6 disabled in all interfaces. What else can be checked to understand the issue?

Ysurac, 2019-03-16:
Some website ban all VPS or dedicated server IPs to allow only IP from residential ISP. 
Nothing I can do about that, it's some stupid website/firewall choices.

victorleduc, 2019-03-16:
You're right, I checked to wget some urls directly from the VPS and I get 403 errors.
The best solution for me is to use a VPN (NordVPN for example) on those particular websites.
I wanted to use the VPN globally, but I think it's only using my VDSL connection and not my 4G...
Is there any way to use the NordVPN on both my connexion ? Can i setup something, even directly on the VPS ?

Ysurac, 2019-03-16:
A VPN over a VPN is always a bad idea. So OMR-ByPass is the best solution for these websites.

cron410, 2019-03-16:
The following is a hack/band-aid. 

Since NordVPN uses OpenVPN, you can disable the default route of the VPN and only route certain IPs through it as described here: https://superuser.com/questions/457020/openvpn-only-route-a-specific-ip-addresses-through-vpn

I've done this before but will have to find that config. The drawback is that you will need to find all IPs those sites use and will have to update them manually if it ever changes. This will also be useless for clients on your network that do not use the VPN.

cron410, 2019-03-18:
@victorleduc Here's a specific config I used to only have Put.io traffic go across the VPN, everything else goes out in the clear. Keep in mind this is an old config. The IP is probably wrong by now.

```
route-nopull
route 95.211.93.133 255.255.255.255
route ftp.put.io 255.255.255.255
route db.put.io 255.255.255.255
route put.io 255.255.255.255
```

cron410, 2019-03-26:
I can confirm that a fresh install of OpenMPTCProuter 0.47.6 and an upgrade of the VPS to latest 0.991 has resolved the issue for me. No changes from a base router install other than adding VPS IP and Admin Key. I can visit https://agenda2.securitest.org/login and github.com with no issues.

victorleduc, 2019-03-28:
Yes all the website are working now with the latest version ! (0.47.6 and VPS 0.991 )
I think we can close the issue now

victorleduc, 2019-03-28:
@Ysurac Do you know what was the issue that was fixed in the last version ? It's strange because I still get 403 when I try to do this on the VPS for exemple :
`wget https://agenda2.securitest.org/images/picto_agenda.png`
But it works fine on my computer. How did you do it ?

closed 2019-03-28 by victorleduc

Ysurac, 2019-03-28:
I modified "net.mptcp.mptcp_syn_retries" to a lower value, because too much retries give a timeout.
I don't know why we have this problem on some websites, I need to investigate when I will have some times.