Ysurac/openmptcprouter #388: Error connection status on android hotspot

manueldev, 2019-03-07:
Hi. I'm trying to add a 4G connection through a hotspot of a moto g5 plus smartphone. The status page shows me that the connection is not successful although there is a connection on the smartphone. Is there any additional configuration to perform in these cases? (I thought maybe the mobile operator might restrict this type of connection).

[image]

## Specifications

  - OpenMPTCProuter version: 0.46.3
  - OpenMPTCProuter VPS version: 0.93 (ssh show me 0.95)
  - OpenMPTCProuter platform: RPI3

Ysurac, 2019-03-07:
I need the result of ```uci show network``` via SSH on OpenMPTCProuter

manueldev, 2019-03-07:
`network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='1'
network.globals=globals
network.globals.ula_prefix='fd9b:8389:a84c::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='default'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_syn_retries='5'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.mptcp_debug='1'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='b8:27:eb:90:17:d1'
network.lan.metric='2'
network.lan.ipv6='0'
network.lan.dns='8.8.8.8 1.1.1.1'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.multipath='master'
network.wan1.defaultroute='0'
network.wan1.type='macvlan'
network.wan1.ifname='wan1'
network.wan1.masterintf='eth0'
network.wan1.metric='3'
network.wan1.netmask='255.255.255.0'
network.wan1.macaddr='4e:24:5e:a5:e8:a0'
network.wan1.label='VTR'
network.wan1.ipaddr='192.168.2.80'
network.wan1.gateway='192.168.2.1'
network.wan1_dev=device
network.wan1_dev.name='wan1'
network.wan1_dev.type='macvlan'
network.wan1_dev.ifname='eth0'
network.wan1_dev.macaddr='4e:24:5e:a5:e8:a0'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.proto='dhcp'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.metric='4'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.gateway='fe80::a00:1'
network.omr6in4.ip6addr='fe80::a00:2'
network.omr6in4.metric='5'
network.wwan=interface
network.wwan.proto='dhcp'
network.wwan.metric='6'`

Ysurac, 2019-03-07:
multipath need to be defined for wwan interface (In Network->Interface->wwan or in Network->MPTCP)

manueldev, 2019-03-07:
Thanks! Now it is working without problem. I think I also generated problems with the chrome autocompleter. When saving the form, the interface did not get ip again. I deleted the "@" that was inserted automatically and now the wifi interface works without problems. Thank you!

[image]

closed 2019-03-11 by Ysurac