Ysurac/openmptcprouter #389: WAN1 (key 4G) near to zero !

CoSysopK, 2019-03-11:
## Actual Behavior

[image]

Connexion WAN1 it does not work well !
Often at zero, and from time to time, a peak at 0.1 Mb/s with nperf.
It's hard to analyse the network with WAN1 because the connexion is to 0 and, there is micros connexions
to 0.1 Mb/s

Connexion WAN2 it's WORK !

## Steps to Reproduce the Problem

  1. Raspberry PI 3/3B+ (64-bit) and, ext4 image: https://www.openmptcprouter.com/files/openmptcprouter-v0.46.3-r09228-a3ccac6b1d-brcm2708-bcm2710-rpi-3-ext4-sysupgrade.img.gz
  2. VPS Ubuntu Server 18.04 x86_64 and, VPS script : wget -O - http://www.openmptcprouter.com/server/ubuntu18.04-x86_64.sh | sh
  2. Configure the router with the wan1 static and, wan2 static
  2. Add the IP address VPS and, all the informations (shadowsocks key, glorytun key, MLVPN password and OpenMPTCProuter VPS Admin key) retrived on the VPS
  3. 1 key 4G on usb PI 3B and, 1 key 4g wifi

## Specifications

  - OpenMPTCProuter version: v0.46.3
  - OpenMPTCProuter VPS version: install the 10 march with this link:  https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS
  - OpenMPTCProuter platform: RPI3

Ysurac, 2019-03-11:
You can try ```omr-test-speed wan1```, ```omr-test-speed wan2``` and ```omr-test-speed```. This will check wan1,wan2 and aggregated speed.

CoSysopK, 2019-03-11:
I don't have this command.

In the OpenMPTCProuter interface, Services > iPerf :

WAN1: (When WAN1 is the alone connexion)
Upload -
error - control socket has closed unexpectedly
Download -
error - control socket has closed unexpectedly

WAN2: (When WAN2 is the alone connexion)
Upload - 
Server: ping.online.net - Sender: 23.07Mb/s - Receiver: 17.16Mb/s
Download - 
Server: ping.online.net - Sender: 21.89Mb/s - Receiver: 21.17Mb/s
[image]

renaudadu, 2019-03-11:
Hello, for omr-speed-test command, you use ssh to your openmptcprouter.

CoSysopK, 2019-03-11:
[image]

I started doing a test with the following commands but it is necessary to wait for 10 GB of data, it is a lot of data whereas I am in 4G and not in unlimited.
How to make a test that uses less data?

renaudadu, 2019-03-11:
No wait. Just launch the command for both Wan interface.

renaudadu, 2019-03-11:
Try with thé second name of the interface. Like eth1 for wan1.

CoSysopK, 2019-03-11:
WAN1:
[image]

WAN2:
[image]

CoSysopK, 2019-03-11:
After several errors, here is the test:
WAN1:
[image]

renaudadu, 2019-03-11:
I have the same speed when my 4g quota is full and my 4g speed pass in low speed.
My connection in low speed mode was not good. Up and down very often.

CoSysopK, 2019-03-11:
Phew, It's exactly that !
I just changed sim and it's perfect.

Thanks for your work !

Have a good day

closed 2019-03-11 by Ysurac