Ysurac/openmptcprouter #390: mptcprouter  v0.47  make menuconfig !!!

kapook007, 2019-03-11:
mptcprouter  v0.47  with myplatform
OMR_TARGET="myplatform" ./build.sh

make menuconfig 
No menu openmptcprouter-full !!!
11-3-2019

mmm@debian:/tmp/openmptcprouter/myplatform/source$ make menuconfig
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/utils/nftables/Makefile' has a dependency on 'jansson', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/nginx/Makefile' has a dependency on 'uwsgi-cgi', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/nginx/Makefile' has a dependency on 'uwsgi-cgi-luci-support', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/nginx/Makefile' has a dependency on 'uwsgi-cgi', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/nginx/Makefile' has a dependency on 'uwsgi-cgi-luci-support', which does not exist
WARNING: Reducing ABI version '2018-05-18-cb119deddee5f0f8f1da883b20c60aea7611b175' of package 'liblucihttp' to '20180518'

Ysurac, 2019-03-11:
You should use your platform name instead of "myplatform" ;)
As some packages depends on arch, it's possible that openmptcprouter menu is not always available, you have to use that in source directory after arch is set in menuconfig:
```
scripts/feeds clean
scripts/feeds update -a
scripts/feeds install -a -d y -f -p openmptcprouter
```

kapook007, 2019-03-12:
scripts/feeds clean
scripts/feeds update -a
scripts/feeds install -a -d y -f -p openmptcprouter

No menu openmptcprouter-full !!!

<*> omr-6in4........................................... OpenMPTCProuter 6in4
<*> omr-update...................................... OpenMPTCProuter updater
<*> openmptcprouter............................. OpenMPTCProuter config

Ysurac, 2019-03-12:
True there is a problem. It's why you shouldn't use not released version :)

kapook007, 2019-03-12:
my board is not have released version T T

Ysurac, 2019-03-12:
I was speaking about the v0.47 that is not yet available.
I have fixed the issue, should be better now.

closed 2019-03-19 by Ysurac