Ysurac/openmptcprouter #391: V0.47 upgrade fail. Could SSH in but no luci

dnwk, 2019-03-13:
I try to upgrade to most recent V0.47 on WRT32X. No response after flashing system interface. I could SSH in and it says 0.47. however, web interface is not available. DHCP doesn't work. How should I proceed from here?

dnwk, 2019-03-13:
Upgrade from 0.46.3

Ysurac, 2019-03-13:
Fixed in latest available image.

closed 2019-03-13 by Ysurac

xslite, 2019-03-16:
Hello, issue is not resolved, as same problem append to me yesterday : used latest image openmptcprouter-v0.47-r09530-64bb88841f-x86-64-combined-squashfs.img.gz  and OpenMptcpRouter is working (as I asked to keep all config parameters) but **no Luci web interface**, only able to connect to Router via SSH !

Ysurac, 2019-03-16:
i'm using this image and I have luci interface. The same image is used for virtualbox and vmware.

Ysurac, 2019-03-16:
You should check via SSH if uhttpd is running and available in /usr/sbin/uhttpd
You can also check using logread if you see any errors.

xslite, 2019-03-16:
Hi !

Yes uhttpd is running, but maybe the probem come from listenning on 0.0.0.0
? instead of the lan address 192.168.3.200 ??

[image: Screenshot from 2019-03-16 15-37-53.png]

xslite, 2019-03-16:
and I see a lot of errors from logread about unknown iface no

[image: Screenshot from 2019-03-16 15-43-46.png]

Ysurac, 2019-03-16:
0.0.0.0 mean all IPs. So no problem here.
You are sure it's no and not lo ? lo mean loopback interface.

xslite, 2019-03-16:
yes, it's written "daemon.err omr-tracker[...] : ping : unknown iface no"

Ysurac, 2019-03-16:
Strange, can you give me all the config in System->OpenMPTCProuter->Show all config. Remove Server IP before paste.

xslite, 2019-03-16:
Yes, but where do I find the Config file via SSH?

Ysurac, 2019-03-16:
True I forget that, it's ```uci show```