Ysurac/openmptcprouter #393: IPv6 fallback issue

dnwk, 2019-03-14:
In v0.47, I turned back on ipv6. Connectivity is fine. However, when there is a problem, it doesn't fall back to ipv4. It get timed out. I have native ipv6 on VPS. Using DHCPv6. Server version 0.96

dnwk, 2019-03-19:
I am on 0.47.1 and 0.97 with WRT32x. It still won't fall back to IPv4 when IPv6 connection failed.

Ysurac, 2019-03-19:
What you want to fallback ? You are using IPv6 as VPS address ?
If it's when you browse website, it's not an OpenMPTCProuter problem but browser/OS problem.

dnwk, 2019-03-19:
I'm using ipv4 as VPS address. Normally when I visit a website, if there is AAAA record in DNS, it will prefer IPv6. But if ipv6 connection failed, browser will revert to using ipv4 for connection. I have my android device doing the fail cover in LTE connection. But fail over is not happening while I connect to openmptcprouter.

Ysurac, 2019-03-19:
Strange. You can test fallback using http://ipv6-test.com/
You can try to remove ipv6_first on ss_redir in shadowsocks interface.

dnwk, 2019-03-19:
I unchecked prefer ipv6 and tested with ipv6-test. It didn't resolved the fallback issue.

Ysurac, 2019-04-08:
Know issue with shadowsocks-libev.
I'm trying to find a way to resolve this.

closed 2019-07-04 by Ysurac