Ysurac/openmptcprouter #394: VPS Update error

Seb944, 2019-03-14:
## Expected Behavior
Hi,
After update (client ans server) i have this message when i want to update the VPS

The package linux-image-4.14.105-mptcp-3f01458 needs to be reinstalled, but I can't find an archive for it
## Actual Behavior

## Specifications

  - OpenMPTCProuter version: 0,47
  - OpenMPTCProuter VPS version: 0,96
  - OpenMPTCProuter platform: RPI2

Ysurac, 2019-03-14:
What command give you that ?

Seb944, 2019-03-14:
wget -O - http://www.openmptcprouter.com/server/debian9-x86_64.sh | sh

Seb944, 2019-03-14:
URL transformed to HTTPS due to an HSTS policy
--2019-03-14 18:42:07--  https://www.openmptcprouter.com/server/debian9-x86_64.sh
Résolution de www.openmptcprouter.com (www.openmptcprouter.com)… 94.23.252.192
Connexion à www.openmptcprouter.com (www.openmptcprouter.com)|94.23.252.192|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 24148 (24K) [application/octet-stream]
Sauvegarde en : « STDOUT »

-                   100%[===================>]  23,58K  --.-KB/s    in 0s      

2019-03-14 18:42:08 (101 MB/s) — envoi vers sortie standard [24148/24148]
       
Hit:1 http://deb.debian.org/debian stretch-backports InRelease                 
Hit:2 http://security.debian.org stretch/updates InRelease                     
Ign:3 http://debian.mirrors.ovh.net/debian stretch InRelease                   
Get:4 http://debian.mirrors.ovh.net/debian stretch-updates InRelease [91.0 kB]
Get:5 http://debian.mirrors.ovh.net/debian stretch Release [118 kB]
Fetched 209 kB in 1s (117 kB/s)                             
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: The package linux-image-4.14.105-mptcp-3f01458 needs to be reinstalled, but I can't find an archive for it.

Ysurac, 2019-03-14:
strange. I'm not able to reproduce the issue.
I think it's after a failed install. You can check the status of the package in  /var/lib/dpkg/status

Seb944, 2019-03-14:
I have that : Package: linux-image-4.14.105-mptcp-3f01458
Status: install reinstreq half-installed
Priority: optional
Section: kernel
Architecture: amd64
Version: 4.14.105-mptcp-3f01458

Ysurac, 2019-03-14:
So it's really the problem ;)
Try ```dpkg --remove --force-remove-reinstreq linux-image-4.14.105-mptcp-3f01458``` and run the VPS install script again.

Seb944, 2019-03-14:
When i install the script after :0 upgraded, 0 newly installed, 0 to remove and 24 not upgraded.
Selecting previously unselected package linux-image-4.14.105-mptcp-3f01458.
(Reading database ... 84828 files and directories currently installed.)
Preparing to unpack .../linux-image-4.14.105-mptcp-3f01458.amd64.deb ...
Ok, aborting, since modules for this image already exist.
dpkg: error processing archive /tmp/linux-image-4.14.105-mptcp-3f01458.amd64.deb (--install):
 subprocess new pre-installation script returned error exit status 1
Examining /etc/kernel/postrm.d .
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 4.14.105-mptcp-3f01458 /boot/vmlinuz-4.14.105-mptcp-3f01458
run-parts: executing /etc/kernel/postrm.d/zz-update-grub 4.14.105-mptcp-3f01458 /boot/vmlinuz-4.14.105-mptcp-3f01458
Errors were encountered while processing:
 /tmp/linux-image-4.14.105-mptcp-3f01458.amd64.deb

Ysurac, 2019-03-14:
Try: ```dpkg -P --force-all /tmp/linux-image-4.14.105-mptcp-3f01458.amd64.deb```

Seb944, 2019-03-14:
dpkg: erreur: --purge requiert un nom de paquet légal. « /tmp/linux-image-4.14.105-mptcp-3f01458.amd64.deb » ne l'est pas ; nom de paquet illégal dans la spécification « /tmp/linux-image-4.14.105-mptcp-3f01458.amd64.deb » : doit commencer par un caractère alphanumérique

Ysurac, 2019-03-14:
```dpkg -P --force-all linux-image-4.14.105-mptcp-3f01458```

Seb944, 2019-03-14:
dpkg: avertissement: requête ignorée : suppression de linux-image-4.14.105-mptcp-3f01458, qui n'est pas installé

Ysurac, 2019-03-14:
Try: ```dpkg --force-all -E -i /tmp/linux-image-4.14.105-mptcp-3f01458.amd64.deb```

Seb944, 2019-03-14:
It's Ok :)
Thanks for your help !

closed 2019-03-14 by Ysurac