Ysurac/openmptcprouter #395: Glorytun Fixed IP addres not working

jamesfranck, 2019-03-15:
## Expected Behavior

Add static IP on each Glorytun clients.

## Actual Behavior

DHCP disabled on VPS in /lib/systemd/network/glorytun-tcp.network
Static IP fixed on Glorytun clients (
[image]

I have 2 online clients
The first one has 10.255.255.2
The second one I fixed 10.255.255.3

But after reboot the second one takes 10.255.255.2 so there's some conflict and they are not working at the same time

## Specifications

  - OpenMPTCProuter version: v0.46.3
  - OpenMPTCProuter VPS version: No idea
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-03-15:
For now multiple OpenMPTCProuter on one VPS is not supported.

closed 2019-03-15 by Ysurac

jamesfranck, 2019-03-18:
Is there any way to fix IP for Glorytun ?

Ysurac, 2019-03-18:
Not for now.