Ysurac/openmptcprouter #396: Create a debian repo for all OpenMPTCProuter VPS utils

Ysurac, 2019-03-15:
This can make VPS install/update faster and mirrors can be created easily.

Ysurac, 2019-03-17:
Debian packages for Shadowsocks-libev with nocrypto patch, obfs plugin, glorytun-tcp and glorytun-udp are done.
Need v2ray-plugin package and to modify VPS install script.

dnwk, 2019-03-26:
How about kernels

Ysurac, 2019-03-26:
It will be in the repo to.

Ysurac, 2019-04-06:
A alpha version of the script using debian package is available here : 
```wget -O - http://www.openmptcprouter.com/server/debian9-x86_64-bin.sh | sh```

gwarf, 2019-04-09:
I quickly used it on my one and only VPS instance, it seems to work :)
The script is definitely much cleaner/shorter than the previous one. It will also be easier to know what to do about system updates (sometimes I was a bit unsure when seeing some system updates of components built manually).
One minor comment: I'm not very comfortable with the various hard/forced removals of the dpkg and apt lock files.

Ysurac, 2019-04-09:
I commented the remove of lock files, should not be needed in most case.
But for dpkg options "force-overwrite",... they are needed when a user use this script to update a VPS where previous script was used.

gwarf, 2019-04-30:
So a few comments (I'm using the script on the VPS I previously setup with the former "manual" script):
- it seems to work quite well :)
- running the script multiple times breaks on `rm /etc/kernel-img.conf` (need a `-f` or a test or handling of error
- this morning I tried to update and it broke when trying to update `shadowsocks-libev` as it conflicts with `omr-shadowsocks-libev`. So I removed `shadowsocks-libev`.

Ysurac, 2019-04-30:
* I added the "-f"
* I will force remove of shadowsocks-libev, this can't be used with omr-shadowsocks-libev (that have the nocrypto patch and some changes in configuration)