Ysurac/openmptcprouter #397: Can't get public IP address from ShadowSocks

suyuan168, 2019-03-18:
I upgraded the vps script. Has the configuration key changed during the first installation? Or is it different every time？
[image]
I used a lot of USB 4G network cards. But now it is not normal.

  - OpenMPTCProuter version: (Last is not a version) 0.47.1
  - OpenMPTCProuter VPS version: (Last is not a version) 0.94
  - OpenMPTCProuter platform: (RPI3B+

Ysurac, 2019-03-18:
After update there is no key change. But as there is a Kernel update you should reboot to use it.

Ysurac, 2019-03-18:
You can also check, on the VPS, via ```ps aux | grep omr``` that the script omr-admin.py is running. If not you can check log in /var/log/daemon.log.

closed 2019-03-18 by suyuan168