Ysurac/openmptcprouter #399: Can't reach website from VPS

dhumeau, 2019-03-20:
I tried to run **wget ip.changeip.com** directly from my VPS and it works fine.

Then I ran **shsudo sh debian9-x86_64.sh**

Now **wget ip.changeip.com** is falling down in timeout.

Any idea of what can cause this issue?

Best regards and much thanks for you work.

Ysurac, 2019-03-20:
You could check in _/etc/shorewall/params.net_ that it's the good interface used for internet connection.

dhumeau, 2019-03-21:
I forgot to say I can reach other website (wget google.fr).

Ysurac, 2019-03-22:
I'm able to reproduce this strange issue.

Ysurac, 2019-03-22:
Seems to be related to MPTCP, i have no problem when I disable it: ```sysctl -w net.mptcp.mptcp_enabled=0```, but it is really slow when enabled: ```sysctl -w net.mptcp.mptcp_enabled=1```.

dhumeau, 2019-03-22:
I confirm it's working with MPTCP disabled.
Thanks for your answer.

Ysurac, 2019-03-22:
This will be fixed in next release.

Ysurac, 2019-03-23:
Should be fixed in latest release (else need Multipath TCP SYN retries should be set to 1 or 2).

closed 2019-03-23 by Ysurac