Ysurac/openmptcprouter #401: Vultr Ubuntu 18.04 VPS install error

zicochaos, 2019-03-21:
## Expected Behavior

Sucessfull install 

## Actual Behavior
 Install failed on fresh Ubuntu 18.04 server 
  
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-egcrns4r/flask-jwt-simple/

## Steps to Reproduce the Problem
Follow instruction from Wiki to install VPS server on Vultr Ubuntu 18.04 

## Specifications

  - OpenMPTCProuter version: (Last is not a version)- N/A
  - OpenMPTCProuter VPS version: (Last is not a version) - 
wget http://www.openmptcprouter.com/server/ubuntu18.04-x86_64.sh
sudo sh ubuntu18.04-x86_64.sh
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) - x86_64

Ysurac, 2019-03-22:
When do you have this error ?
I have no error on OVH VPS and on fresh Ubuntu 18.04 install.

zicochaos, 2019-03-22:
fresh install on Vultr Ubuntu 18.04 

the first attempt straight with the error

Ysurac, 2019-03-22:
Can I have the exact version of Ubuntu 18.04 (18.04.2 ?) ?
All is updated before install ?

zicochaos, 2019-03-22:
18.04.02 yes apt update and apt upgrade with reboot done 

etting up python3-lib2to3 (3.6.7-1~18.04) ...
Setting up python3-itsdangerous (0.24+dfsg1-2) ...
Setting up python3-aniso8601 (2.0.0-1) ...
Setting up python3-tz (2018.3-2) ...
Setting up python3-distutils (3.6.7-1~18.04) ...
Setting up python-setuptools (39.0.1-2) ...
Setting up python3-flask (0.12.2-3) ...
Setting up python-gunicorn (19.7.1-4) ...
Setting up gunicorn (19.7.1-4) ...
Setting up python3-pip (9.0.1-2.3~ubuntu1) ...
Setting up python3-flask-restful (0.3.6-1) ...
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-45_pb5a8/flask-jwt-simple/
root@mrtcp2:~#

Ysurac, 2019-03-22:
ok, I will take a vultr VPS to test and fix that.

Ysurac, 2019-03-22:
Fixed in latest VPS script.

closed 2019-03-22 by Ysurac