Ysurac/openmptcprouter #402: Port forwding

flimo44, 2019-03-21:
## Expected Behavior

  Since 2 lasts update port forward fail.

## Actual Behavior
Actualy all is ok except port forward.
Not change in my rules was good before.
OpenMPTCProuter state all is green.

## Steps to Reproduce the Problem

  1. all time now
  2.
  3.

## Specifications

  - OpenMPTCProuter version: v0.47.4 
  - OpenMPTCProuter VPS version:  v 0.98
  - OpenMPTCProuter platform: x86_64 synology virtual Machine ( x86-64 UEFI (64-bit)
squashfs image)

Ysurac, 2019-03-22:
Port forward use the VPN, not shadowsocks.
Check that "Redirects all ports from server to OpenMPTCProuter" in enabled, in System->OpenMPTCProuter->Advanced router.
Check also on the VPS that last lines begin with DNAT and not #DNAT in /etc/shorewall/rules.

flimo44, 2019-03-22:
Hello,
 "Redirects all ports from server to OpenMPTCProuter"  are  enabled

and on VPS rules in   /etc/shorewall/rules. I have the 2 lines at the end with not # begin

For in formation with 47.1 ( snapshoot restored and ok before update ) i have the same issus with VPS 0.98. It's a OVH VPS

Ysurac, 2019-03-22:
If you have no error about the VPN, then all ports are directed to OpenMPTCProuter.
So the problem is now on OpenMPTCProuter side. Try to apply again firewall rules.

flimo44, 2019-03-22:
i have already try  to re-apply firewall rules several time :(

flimo44, 2019-03-22:
i have many error in /var/log/daemon.log ( VPS )

-bash: syntax error near unexpected token `('
root@vps612694:~# Mar 22 19:56:21 vps612694 snmpd[707]: error on subcontainer 'ia_addr' insert (-1)
-bash: syntax error near unexpected token `('
root@vps612694:~# Mar 22 19:56:51 vps612694 snmpd[707]: error on subcontainer 'ia_addr' insert (-1)
-bash: syntax error near unexpected token `('
root@vps612694:~# Mar 22 19:57:21 vps612694 snmpd[707]: error on subcontainer 'ia_addr' insert (-1)
-bash: syntax error near unexpected token `('Mar 22 19:57:51 vps612694 snmpd[707]: error on subcontainer 'ia_addr' insert (-1)

root@vps612694:~# Mar 22 19:57:51 vps612694 snmpd[707]: error on subcontainer 'ia_addr' insert (-1)
-bash: syntax error near unexpected token `('

Ysurac, 2019-03-22:
I would need to full config (without server ip) available in System->OpenMPTCProuter->Show all settings

flimo44, 2019-03-22:
[OpenMPTCProuter settings.txt](https://github.com/Ysurac/openmptcprouter/files/2998279/OpenMPTCProuter.settings.txt)

flimo44, 2019-03-23:
hello,
it works again but I have not found the real cause yet. I deleted the rules and recreated. it still did not work. the last thing I put 80 for SNAT port that was empty in Traffic Rules. Then if i put back empty, it still works again. As soon as I can I reboot everything to see if work fine again.

jusjus13, 2019-03-23:
salut j ai eu le problème j'ai réinstaller la maj du vps 0.991, il y a du avoir un bug dans installe

flimo44, 2019-03-23:
Update VPS 0.991 and  router 0.47.5  done. 
Everything seems (again)  to work well after reboot.
Thank you Ysurac for all the work and time spent on this project.

closed 2019-03-23 by Ysurac