Ysurac/openmptcprouter #403: The factory settings are automatically restored during the operation, and the backup file cannot be restored. The recovery fails.

suyuan168, 2019-03-22:
Question 1. I want to use the ngrok reverse proxy to remotely manage openmptcprouter. I found that after the sysupgrade.conf file was written to the file that needs to be saved, the file I need to save is not saved after the system firmware update.

Problem 2. After upgrading the firmware using sysupgrade, it is found that the script configuration files such as cfg and sh files are normal, but the executable file is changed, and the file size becomes smaller and becomes unexecutable.

Question 3. When I use the firmware upgrade file, I find that the system suddenly becomes the initial setting. I am prompted to set the password and all of the previous settings are gone. When I went to restore the backup, I found that the backup was restarted after the backup was restarted. Did not restore the backup.

## Steps to Reproduce the Problem

  1.Modify /etc/sysupgrade.conf to fill in the executable file ngrok you need to save, for example /root/route/ngrok
  2.Go to system firmware upgrade Select to keep the configuration, then upload a new upgrade package openmptcprouter-v0.47.4-r09530-64bb88841f-brcm2708-bcm2709-rpi-2-squashfs-sysupgrade.img.gz
  3.After the system upgrade is completed, the /root/route/ngrok file size is changed. Do not execute the command prompt. File error
4.Restart the system
5.The system was restored to the initial settings and all previous configurations were lost.
6.Go to the system settings Restore backup, restore and restart, the system is still initialized

## Specifications

  - OpenMPTCProuter version: 4.7.4
  - OpenMPTCProuter VPS version: 0.94
  - OpenMPTCProuter platform: RPI3

Ysurac, 2019-03-22:
1. Seems to be some problem here. This part come from OpenWrt. Check using "sysupgrade -l" via SSH on OpenMPTCProuter that you have your files added.
2. No idea
3. What do you have in your backup file ?

suyuan168, 2019-03-22:
1.I used to get into the habit of saving backups every time I successfully configured. The latest is a backup file three days ago.
3.ssh via ""sysupgrade -l" Show that I have added a file
2.I feel that the openwrt upgrade refresh firmware compresses my files when backing up files. Make my files unusable.

suyuan168, 2019-03-22:
I can't restore the backup now.

Ysurac, 2019-03-22:
What do you have inside the backup gz file ?

suyuan168, 2019-03-22:
is

suyuan168, 2019-03-22:
Segmentation fault

closed 2019-03-23 by suyuan168