Ysurac/openmptcprouter #404: Installation of OpenMPTCProuter on Proxmox VM

Caelion, 2019-03-23:
## Expected Behavior

Installation of OpenMPTCProuter

## Actual Behavior

Installation stopped at : (extract of console)
>random: crng init done
>adding VLAN 0 t oHW filter on device eth0
>adding VLAN 0 to HW filter on device wan1
>adding VLAN 0 to HW filter on device wan2

and then nothing happened
so the installation stopped after few seconds of the start of the VM on Proxmox

## Steps to Reproduce the Problem

  1. Creation of a new VM on proxmox :
  2. Parameters are :
Memory : 512MiB / 2,00GiB
Processors : 1 sockets / 2 cores
HDD : virtio0 size = 10G
Network : virtio vmbr0
  3. Connect to SSH to Proxmox server (not the vm) and upload img on folder /home/%user%/ and name it omr.img
  4. Dump the file to the disk directory : for me, the VM ID is 105 so :
dd bs=4M if=/home/%user%/omr.img of=/dev/pve/vm-105-disk-0
  5. Start the VM and look the consol

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.47.4-r09530-64bb88841f-x86-64-combined-squashfs.img
  - OpenMPTCProuter VPS version: not installed yet as OpenMPTCProuter installation failed
  - OpenMPTCProuter platform: x86_64 (VM on Proxmox on Intel NUC)

Ysurac, 2019-03-23:
This looks like message at the end of OpenMPTCProuter boot. Did you try "Enter" key ?

Caelion, 2019-03-23:
Loool
I'm a stupid boy
It's ok after I press Enter
Thank you Ysurac

sesipod, 2019-03-23:
Works just fine for me you just have to hit [enter] at the end of the output. 
[image]

closed 2019-03-23 by Caelion