Ysurac/openmptcprouter #407: Connection Reset

dnwk, 2019-03-24:
daemon.err /usr/bin/ss-local[1730]: getpeername: Socket not connected
daemon.err /usr/bin/ss-redir[1732]: remote recv: Connection reset by peer
I keep getting connection reset since upgrade. Revert back to 47.1 doesn't resolve the issue since I can't revert my VPS to 0.98.  Not sure why.

These are syslogs from VPS
Mar 24 17:25:39 murph ss-server[457]:  2019-03-24 17:25:39 ERROR: remote recv: Connection reset by peer
Mar 24 17:25:41 murph kernel: [  381.926388] Shorewall:rplog:DROP:IN=eth0 OUT= MAC=aa:00:00:15:34[....]SRC=2605:2700:0000:[....]DST=2605:2700:[....] LEN=80 TC=0 HOPLIMIT=63 FLOWLBL=615901 PROTO=TCP SPT=25092 DPT=853 WINDOW=65535 RES=0x00 SYN URGP=0 
Mar 24 17:25:45 murph ss-server[457]:  2019-03-24 17:25:45 ERROR: remote recv: Connection reset by peer
Mar 24 17:25:57 murph kernel: [  398.054575] Shorewall:rplog:DROP:IN=eth0 OUT= MAC=aa:00:00:15:34:[....]SRC=2605:2700:0000[....]DST=2605:2700:0000:[,,,,] LEN=80 TC=0 HOPLIMIT=63 FLOWLBL=42639 PROTO=TCP SPT=25092 DPT=853 WINDOW=65535 RES=0x00 SYN URGP=0 

## Specifications

  - OpenMPTCProuter version: (Last is not a version)  47.6
  - OpenMPTCProuter VPS version: (Last is not a version)  0.991
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) WRT32X

Ysurac, 2019-03-24:
"Connection reset by peer" is not an issue.
And nothing strange in shorewall logs.
What is the problem ?

dnwk, 2019-03-24:
shorewall-init.log doesn't have anything interesting. However, I frequent have website that won't load and I need to do couple refresh before it will load.  (error message: TLS handshake fail)

Ysurac, 2019-03-24:
Without logs or any data this will be difficult to find where is your problem.
You should have some data in OpenMPTCProuter system logs.
Do you have IPv6 enabled ?

dnwk, 2019-03-24:
I am looking at logs. Besides  /usr/bin/ss-local[8839]: getpeername: Socket not connected, I can't find anything related to why my connection is unstable and some will fail to load. Also, my port forwarding stop to work.   I have IPv6 enabled. I just disabled it and I am watching to see if things improves.

dnwk, 2019-03-24:
I frequently get error says "Can't get public IP from shadowsocks" on wizard and then switch to "Direct Output"

Ysurac, 2019-03-24:
You shouldn't have too much logs from ss-local. It's when shadowsocks is not reachable.
I would need your config in System->OpenMPTCProuter->Show all settings

dnwk, 2019-03-24:
All OpenMPTCProuter settings

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].server='1.1.1.1#53' '1.0.0.1#53'
dhcp.@dnsmasq[0].ipset='/vpn.******.edu/omr_dst_bypass_eth2,omr6_dst_bypass_eth2'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_management='1'
dhcp.lan.ndp='hybrid'
dhcp.lan.dns='2605:***::1'
dhcp.lan.ra_default='0'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='Tablo'
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='C4****'
dhcp.@host[0].ip='192**'
dhcp.@host[1]=host
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='24:5****'
dhcp.@host[1].ip='192.****'
dhcp.@host[1].name='cloudth******'
dhcp.@host[2]=host
dhcp.@host[2].name='piaware'
dhcp.@host[2].dns='1'
dhcp.@host[2].mac='B8:27:****'
dhcp.@host[2].ip='192.****'
dhcp.@host[3]=host
dhcp.@host[3].name='unifi_controller'
dhcp.@host[3].dns='1'
dhcp.@host[3].mac='02:42:***'
dhcp.@host[3].ip='192.168****'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='*****'
etherwake.@target[0].wakeonboot='off'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
[... 1258 lines cut ...]
wireless.radio0.hwmode='11a'
wireless.radio0.path='soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
wireless.radio0.legacy_rates='1'
wireless.radio0.htmode='VHT80'
wireless.radio0.country='US'
wireless.radio0.channel='48'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.hwmode='11g'
wireless.radio1.path='soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
wireless.radio1.country='US'
wireless.radio1.legacy_rates='1'
wireless.radio1.channel='auto'
wireless.radio1.htmode='HT40'
wireless.radio2=wifi-device
wireless.radio2.type='mac80211'
wireless.radio2.path='platform/soc/soc:internal-regs/f10d8000.sdhci/mmc_host/mmc0/mmc0:0001/mmc0:0001:1'
wireless.radio2.country='US'
wireless.radio2.legacy_rates='1'
wireless.radio2.hwmode='11g'
wireless.radio2.channel='auto'
wireless.radio2.htmode='HT40'
wireless.radio2.txpower='23'

Powered by LuCI Master (git-19.082.80236-26b738a) / openmptcprouter v0.47.6 r0+9530-64bb88841f

Ysurac, 2019-03-24:
I don't see problems. You should try with Glorytun TCP instead of Glorytun UDP.

dnwk, 2019-03-24:
Ok. Let me try. However, I don't think it ever fail over to VPN when shadowsocks failed.

Ysurac, 2019-03-24:
You could try also to set back shadowsocks to chacha20, maybe a change make the patch not working well.

dnwk, 2019-03-24:
I was using chacha20 and I get constant reset. So I am trying to set it to none and see if it work.

dnwk, 2019-03-24:
Turning off IPv6 has my port forwarding restored. I will have to see if my connection will stabilize.

dnwk, 2019-03-24:
Connection reset to an established connection on IPv4 only mode is slightly better. I still get connection refused and shadowsocks disconnect once in a while but less often.

dnwk, 2019-03-25:
If possible,  could you provide previous versions of VPS install scripts?

dnwk, 2019-03-25:
I was able to connect to an old vps that's still running 0.93 version of OMR-ADMIN and my connection is flawless. So maybe some newer version of shadowsocks are creating troubles?

Ysurac, 2019-03-25:
All VPS scripts are here: https://github.com/Ysurac/openmptcprouter-vps
I don't have the same troubles and there is no reports of such troubles in shadowsocks issues.
Can you check in /var/log/daemon.log on the VPS if you have some strange logs ?

TheCrach, 2019-03-25:
Restoring shadowsocks to version 3.2.3 instead of 3.2.5 fixes the issue for me.

TheCrach, 2019-03-25:
Concerning the logs part, using 3.2.5

> Mar 25 08:55:53 sd-xxxxx ss-server[19178]:  2019-03-25 08:55:53 ERROR: getpeername: Transport endpoint is not connected
Mar 25 08:55:53 sd-xxxxx ss-server[19178]:  2019-03-25 08:55:53 ERROR: getpeername: Transport endpoint is not connected
Mar 25 08:55:53 sd-xxxxx ss-server[19151]:  2019-03-25 08:55:53 ERROR: getpeername: Transport endpoint is not connected
Mar 25 08:55:53 sd-xxxxx ss-server[19178]:  2019-03-25 08:55:53 ERROR: getpeername: Transport endpoint is not connected
Mar 25 08:55:53 sd-xxxxx ss-server[19148]:  2019-03-25 08:55:53 ERROR: getpeername: Transport endpoint is not connected
Mar 25 08:55:53 sd-xxxxx ss-server[19163]:  2019-03-25 08:55:53 ERROR: getpeername: Transport endpoint is not connected
Mar 25 08:55:53 sd-xxxxx ss-server[19178]:  2019-03-25 08:55:53 ERROR: getpeername: Transport endpoint is not connected
Mar 25 08:55:53 sd-xxxxx ss-server[19151]:  2019-03-25 08:55:53 ERROR: getpeername: Transport endpoint is not connected
Mar 25 08:55:53 sd-xxxxx ss-server[19160]:  2019-03-25 08:55:53 ERROR: getpeername: Transport endpoint is not connected
Mar 25 08:55:53 sd-xxxxx ss-server[19160]:  2019-03-25 08:55:53 ERROR: getpeername: Transport endpoint is not connected
Mar 25 08:55:55 sd-xxxxx ss-server[19154]:  2019-03-25 08:55:55 ERROR: getpeername: Transport endpoint is not connected

These kind of errors disappear completely restoring 3.2.3. 
Could be related to IPv6, my server doesn't have IPv6 working and I understood you do @Ysurac

dnwk, 2019-03-25:
I have reinstall the VPS so the log is gone. Similar things have happened to me before after version upgrade. Maybe it is something related to the Linksys? I can't seem to find any clue.

dnwk, 2019-03-26:
Is https://www.openmptcprouter.com/kernel/linux-image-4.14.94-mptcp-a284ba1.amd64.deb Kernel still exist somewhere?

Ysurac, 2019-03-26:
No. But you can compile it again if you really want.

TheCrach, 2019-03-26:
I have 4.14.91 if you want:
https://we.tl/t-Kbdhg7hFYT

dnwk, 2019-03-27:
And for daemon.log
I rebuild the VPS and run for a while. These are the only log I could find and I can't find anything interesting in there

Mar 26 18:53:13 murph shorewall[404]: Compiling using Shorewall 5.0.15.6...
Mar 26 18:53:14 murph shorewall6[395]: Processing /etc/shorewall6/params ...
Mar 26 18:53:14 murph shorewall6[395]: Processing /etc/shorewall6/shorewall6.conf...
Mar 26 18:53:14 murph shorewall6[395]: Loading Modules...
Mar 26 18:53:14 murph shorewall6[395]: Compiling /etc/shorewall6/zones...
Mar 26 18:53:14 murph shorewall6[395]: Compiling /etc/shorewall6/interfaces...
Mar 26 18:53:14 murph shorewall6[395]: Determining Hosts in Zones...
Mar 26 18:53:14 murph shorewall6[395]: Locating Action Files...
Mar 26 18:53:14 murph shorewall6[395]: Compiling /etc/shorewall6/policy...
Mar 26 18:53:14 murph shorewall6[395]: Adding rules for DHCP
Mar 26 18:53:14 murph shorewall6[395]: Compiling TCP Flags filtering...
Mar 26 18:53:14 murph shorewall6[395]: Compiling /etc/shorewall6/snat...
Mar 26 18:53:14 murph shorewall6[395]: Compiling MAC Filtration -- Phase 1...
Mar 26 18:53:14 murph shorewall6[395]: Compiling /etc/shorewall6/rules...
Mar 26 18:53:14 murph ntpd[501]: Listen normally on 16 eth0 [fe80::a800:ff:fe15:34a3%2]:123
Mar 26 18:53:14 murph shorewall[404]: Processing /etc/shorewall/params ...
Mar 26 18:53:14 murph shorewall[404]: Processing /etc/shorewall/shorewall.conf...
Mar 26 18:53:14 murph shorewall[404]: Loading Modules...
Mar 26 18:53:14 murph shorewall6[395]: Compiling /etc/shorewall6/conntrack...
Mar 26 18:53:14 murph shorewall6[395]: Compiling MAC Filtration -- Phase 2...
Mar 26 18:53:14 murph shorewall6[395]: Applying Policies...
Mar 26 18:53:14 murph shorewall6[395]: Compiling /usr/share/shorewall/action.Drop for chain Drop...
Mar 26 18:53:14 murph shorewall6[395]: Compiling /usr/share/shorewall6/action.AllowICMPs for chain AllowICMPs...
Mar 26 18:53:14 murph shorewall6[395]: Compiling /usr/share/shorewall6/action.Broadcast for chain Broadcast...
Mar 26 18:53:14 murph shorewall6[395]: Generating Rule Matrix...
Mar 26 18:53:15 murph shorewall6[395]: Compiling /usr/share/shorewall/action.Reject for chain Reject...
Mar 26 18:53:15 murph shorewall6[395]: Creating ip6tables-restore input...
Mar 26 18:53:15 murph shorewall[404]: Compiling /etc/shorewall/zones...
Mar 26 18:53:15 murph shorewall[404]: Compiling /etc/shorewall/interfaces...
Mar 26 18:53:15 murph shorewall[404]: Determining Hosts in Zones...
Mar 26 18:53:15 murph shorewall[404]: Locating Action Files...
Mar 26 18:53:15 murph shorewall[404]: Compiling /etc/shorewall/policy...
Mar 26 18:53:15 murph shorewall[404]: Adding Anti-smurf Rules
Mar 26 18:53:15 murph shorewall[404]: Adding rules for DHCP
Mar 26 18:53:15 murph shorewall[404]: Compiling TCP Flags filtering...
Mar 26 18:53:15 murph shorewall[404]: Compiling Kernel Route Filtering...
Mar 26 18:53:15 murph shorewall[404]: Compiling Martian Logging...
Mar 26 18:53:15 murph shorewall[404]: Compiling Accept Source Routing...
Mar 26 18:53:15 murph shorewall[404]: Compiling /etc/shorewall/snat...
Mar 26 18:53:15 murph shorewall[404]: Compiling MAC Filtration -- Phase 1...
Mar 26 18:53:15 murph shorewall[404]: Compiling /etc/shorewall/rules...
Mar 26 18:53:15 murph shorewall[404]: Compiling /etc/shorewall/conntrack...
Mar 26 18:53:15 murph shorewall[404]: Compiling MAC Filtration -- Phase 2...
Mar 26 18:53:15 murph shorewall[404]: Applying Policies...
Mar 26 18:53:15 murph shorewall[404]: Compiling /usr/share/shorewall/action.Drop for chain Drop...
Mar 26 18:53:15 murph shorewall[404]: Compiling /usr/share/shorewall/action.Broadcast for chain Broadcast...
Mar 26 18:53:15 murph shorewall[404]: Generating Rule Matrix...
Mar 26 18:53:15 murph shorewall[404]: Compiling /usr/share/shorewall/action.Reject for chain Reject...
Mar 26 18:53:15 murph shorewall[404]: Creating iptables-restore input...
Mar 26 18:53:15 murph shorewall[404]: Compiling /etc/shorewall/stoppedrules...
Mar 26 18:53:15 murph shorewall[404]: Shorewall configuration compiled to /var/lib/shorewall/.start
Mar 26 18:53:15 murph shorewall[404]: Starting Shorewall....
Mar 26 18:53:15 murph shorewall[404]: Initializing...
Mar 26 18:53:15 murph shorewall[404]: Setting up Route Filtering...
Mar 26 18:53:15 murph shorewall[404]: Setting up Martian Logging...
Mar 26 18:53:15 murph shorewall[404]: Setting up Accept Source Routing...
Mar 26 18:53:15 murph systemd[1]: Reloading.
Mar 26 18:53:15 murph shorewall[404]: Preparing iptables-restore input...
Mar 26 18:53:15 murph shorewall[404]: Running /sbin/iptables-restore ...
Mar 26 18:53:15 murph shorewall[404]: IPv4 Forwarding Enabled
Mar 26 18:53:15 murph shorewall[404]: done.
Mar 26 18:53:15 murph systemd[1]: Started Shorewall IPv4 firewall.
Mar 26 18:53:15 murph systemd[1]: Reloading.
Mar 26 18:53:16 murph shorewall6[395]: Starting Shorewall6....
Mar 26 18:53:16 murph shorewall6[395]: Initializing...
Mar 26 18:53:16 murph shorewall6[395]: Preparing ip6tables-restore input...
Mar 26 18:53:16 murph shorewall6[395]: Running /sbin/ip6tables-restore ...
Mar 26 18:53:16 murph shorewall6[395]: IPv6 Forwarding Enabled
Mar 26 18:53:16 murph shorewall6[395]: Setting up IPv6 Interface Forwarding...
Mar 26 18:53:16 murph shorewall6[395]: done.
Mar 26 18:53:16 murph systemd[1]: Started Shorewall IPv6 firewall.
Mar 26 19:19:24 murph systemd[1]: keyboard-setup.service: Cannot add dependency job, ignoring: Unit keyboard-setup.service is masked.
[... cut ...]
Mar 26 19:19:24 murph ss-server[5501]:  2019-03-26 19:19:24 INFO: udp port reuse enabled
Mar 26 19:19:24 murph ss-server[5501]:  2019-03-26 19:19:24 INFO: udp server listening at 0.0.0.0:65101
Mar 26 19:19:24 murph ss-server[5501]:  2019-03-26 19:19:24 INFO: udp port reuse enabled
Mar 26 19:19:24 murph ss-server[5501]:  2019-03-26 19:19:24 INFO: running from root user
Mar 26 19:19:25 murph glorytun-tcp-run[365]: 66.87.138.64.5447: connected
Mar 26 19:19:56 murph glorytun-tcp-run[365]: 66.87.138.64.7616: connected

dnwk, 2019-03-27:
And these info are showing up on the out-of-band screen
I only found one that's SRC from my IP. Have to test if it is something on the VPS's network that's keep dropping my connection. 

[ 1986.715584] Shorewall:net-fw:DROP:IN=eth0 OUT= MAC=aa:00:00:15:34:a3:f4:52:14:97:07:d2:08:00 SRC=66.87.138.64 DST=71.19.154.208 LEN=129 TOS=0x00 PREC=0x20 TTL=51 ID=13284 DF PROTO=UDP SPT=10719 DPT=6881 LEN=109

dnwk, 2019-03-27:
I ran out of options to test. I have setup a brandnew VPS with a different provider running 0.991 and I still get constant shadowsock restart and reset. Right now one of my old VPS running 0.93 is working fine without problem. However, I can't reinstall 0.93 on any other VPS. For some reason, even if I use an old install script and kernel compiled by @TheCrach , I get a glorytun compile error. I just have to hand on to my old VPS that's running 0.93 for now.

Ysurac, 2019-03-27:
I think the problem come from omr services: ```systemctl stop omr``` and tell me if it's better. (don't change any VPN settings after that, this will not work).

dnwk, 2019-03-28:
I have tested with omr stopped. It seems better. It would not outright shut off shadowsocks. It seems keep trying in the background. As I observed that on the status page, sometimes it will switch to "direct output". For my connections in browser, it will not reset outright but will keep retrying in the back.  In the log, I could see OMR-ByPass keep rebooting. 

Wed Mar 27 19:10:47 2019 user.notice Shadowsocks: Rules UP
Wed Mar 27 19:10:51 2019 user.notice omr-bypass: Starting OMR-ByPass...
Wed Mar 27 19:11:25 2019 daemon.info dnsmasq[8940]: read /etc/hosts - 4 addresses
Wed Mar 27 19:11:25 2019 daemon.info dnsmasq[8940]: read /tmp/hosts/dhcp.cfg01411c - 6 addresses
Wed Mar 27 19:11:25 2019 daemon.info dnsmasq-dhcp[8940]: read /etc/ethers - 0 addresses
Wed Mar 27 19:11:25 2019 user.notice omr-bypass: OMR-ByPass is running
Wed Mar 27 19:11:39 2019 daemon.err omr-tracker-ss[21713]: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Wed Mar 27 19:11:39 2019 user.notice Shadowsocks: Rules UP
Wed Mar 27 19:11:42 2019 user.notice omr-bypass: Starting OMR-ByPass...
Wed Mar 27 19:12:17 2019 daemon.info dnsmasq[8940]: read /etc/hosts - 4 addresses
Wed Mar 27 19:12:17 2019 daemon.info dnsmasq[8940]: read /tmp/hosts/dhcp.cfg01411c - 6 addresses
Wed Mar 27 19:12:17 2019 daemon.info dnsmasq-dhcp[8940]: read /etc/ethers - 0 addresses
Wed Mar 27 19:12:17 2019 user.notice omr-bypass: OMR-ByPass is running
Wed Mar 27 19:12:30 2019 daemon.info dnsmasq[8940]: read /etc/hosts - 4 addresses
Wed Mar 27 19:12:30 2019 daemon.info dnsmasq[8940]: read /tmp/hosts/dhcp.cfg01411c - 6 addresses
Wed Mar 27 19:12:30 2019 daemon.info dnsmasq-dhcp[8940]: read /etc/ethers - 0 addresses
Wed Mar 27 19:12:30 2019 user.notice omr-bypass: OMR-ByPass is running
Wed Mar 27 19:12:30 2019 user.notice omr-bypass: Starting OMR-ByPass...
Wed Mar 27 19:12:36 2019 daemon.err omr-tracker-ss[21713]: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Wed Mar 27 19:12:36 2019 user.notice Shadowsocks: Rules UP
Wed Mar 27 19:12:39 2019 user.notice omr-bypass: Starting OMR-ByPass...

dnwk, 2019-03-28:
I could report back on some new findings. I think it probably has something to do with Linksys client. I reinstall my router on a x86_64 VM and it works fine. The only problem is when I use my USB LTE router, the  process to generate the status page will hang there and page will eventually timed out. But otherwise network connectivity is fine on x86_64 even when omr is turned on.

dnwk, 2019-03-29:
Walk back my previous comments a little bit. The x86 version still get browser sessions reset sometimes.  It's just way less often.

dnwk, 2019-04-05:
So there is another update on the hunt to find the root cause. Apparently, one of my LTE router seems to have problems. On status page, it will show up as not responding to pings once every couple minutes. So I disconnected the LTE router and the connection is way better stable. So as a conclusion, I suspect this version is very susceptible to a downed gateway or package loss.

dnwk, 2019-04-06:
One of the LTE router keep getting "no server ping after 1 second" error on the status page every couple other minutes. Is it a problem?

Ysurac, 2019-04-06:
This can be yes. This mean there is packets loss between this router and the server. But some ISP set ICMP (ping) as very low priority and some packets can be lost.
You can check if this is really a problem in the system log, if this interface is often detected as down and this is not the case you can change in omr-tracker the method used for this interface (you can set httping for exemple).

closed 2019-04-11 by dnwk