Ysurac/openmptcprouter #408: ERR_CONNECTION_CLOSED

KoRnEr007, 2019-03-24:
Hello,
Since the update of my router and VPS, the connection is not stable :(

My connection is reset ..

On logs i have:
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Sun Mar 24 18:53:47 2019 daemon.notice netifd: omr6in4 (15093): Command failed: Unknown error
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Sun Mar 24 18:53:47 2019 daemon.notice netifd: omr6in4 (15114): Command failed: Unknown error
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Sun Mar 24 18:53:47 2019 daemon.notice netifd: omr6in4 (15135): Command failed: Unknown error
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Sun Mar 24 18:53:47 2019 daemon.notice netifd: omr6in4 (15156): Command failed: Unknown error
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Sun Mar 24 18:53:47 2019 daemon.notice netifd: omr6in4 (15177): Command failed: Unknown error
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Sun Mar 24 18:53:47 2019 daemon.notice netifd: omr6in4 (15198): Command failed: Unknown error
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Sun Mar 24 18:53:47 2019 daemon.notice netifd: omr6in4 (15219): Command failed: Unknown error
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Sun Mar 24 18:53:47 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Sun Mar 24 18:53:47 2019 daemon.notice netifd: omr6in4 (15240): Command failed: Unknown error

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.47.6 r0+9530-64bb88841f / LuCI Master (git-19.082.80236-26b738a)
  - OpenMPTCProuter VPS version: 0.991
  - OpenMPTCProuter platform: 	Raspberry Pi 3 Model B Rev 1.2

KoRnEr007, 2019-03-24:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].ipset='/nflxvideo.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/netflix.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/jeuxvideo.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/discord.gg/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/netflix.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflxext.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflximg.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflximg.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflxso.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_default='0'
dhcp.lan.dhcpv6='disabled'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='KoRnEr007'
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='74:D4:35:E4:C3:FB'
dhcp.@host[0].ip='192.168.100.211'
dhcp.@host[0].gw='192.168.100.1'
dhcp.@host[1]=host
dhcp.@host[1].name='Shield'
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='00:04:4B:48:36:6E'
dhcp.@host[1].ip='192.168.100.3'
dhcp.@host[1].gw='192.168.100.1'
dhcp.@host[2]=host
dhcp.@host[2].name='googleHome'
dhcp.@host[2].dns='1'
dhcp.@host[2].mac='80:C5:F2:3D:5D:01'
dhcp.@host[2].ip='192.168.100.200'
dhcp.@host[2].gw='192.168.100.1'
dhcp.@host[3]=host
dhcp.@host[3].name='Imprimante'
dhcp.@host[3].dns='1'
dhcp.@host[3].mac='60:6B:FF:C1:2B:9B'
dhcp.@host[3].ip='192.168.100.203'
dhcp.@host[3].gw='192.168.100.1'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dropbear.@dropbear[0].Interface='lan'
dropbear.@dropbear[0].GatewayPorts='on'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='AABBCCDDEEFF'
[... 1196 lines cut ...]
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[0].int_addr='192.168.100.211'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
upnpd.@perm_rule[1].int_addr='192.168.100.211'
upnpd.@perm_rule[1].action='allow'
upnpd.config=upnpd
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.uuid='578535bd-6294-4e28-a4a7-dd1122465b60'
upnpd.config.enabled='1'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='11'
wireless.radio0.hwmode='11g'
wireless.radio0.path='platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='HT20'
wireless.radio0.disabled='1'

Ysurac, 2019-03-24:
You have a problem with omr6in4 and with the VPN, No idea how you can get an 10.255.254.3 IP...

KoRnEr007, 2019-03-24:
I don't change that ....

KoRnEr007, 2019-03-24:
What is the good parameters please ?

KoRnEr007, 2019-03-24:
[image]

Ysurac, 2019-03-24:
True, it's a problem. No idea how you can have something like that... And local IP should be 10.255.254.2.

Ysurac, 2019-03-24:
"Adresse IPv4 distante" should be 10.255.254.1 (because you use Glorytun UDP)

KoRnEr007, 2019-03-24:
and locale ?

Ysurac, 2019-03-24:
no problem for locale IPv6

KoRnEr007, 2019-03-24:
Sun Mar 24 19:16:18 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Sun Mar 24 19:16:18 2019 daemon.notice netifd: omr6in4 (8148): Command failed: Unknown error

it's look good, no issue

closed 2019-03-24 by KoRnEr007