Ysurac/openmptcprouter #410: Glorytun UDP   Don't work.

kapook007, 2019-03-27:
Only Glorytun TCP is  working.
but Glorytun UDP   Don't work.
and  MLVPN   Don't work.

My vps is behind nat 
i forward port
Shadowsocks port is 65101 (TCP & UDP)
Glorytun port is 65001 (TCP & UDP)
OMR JSON admin is 65500 (TCP)
OpenVPN port is 65301 (TCP)
MLVPN ports are 65201-65208 (UDP)

  - OpenMPTCProuter version: 0.46 ,  0.47.6
  - OpenMPTCProuter VPS version: 0.98 , 0.99
  - OpenMPTCProuter platform: rbp-R2 , x86_64

Ysurac, 2019-03-27:
All are working for me on latest release.
I would need the system log on OpenMPTCProuter.

cron410, 2019-03-28:
My VPS has a public IPv4. If I change Glorytun to UDP in settings I get VPN Tunnel Down. After stopping GloryTUN and changing back to TCP, I can start the service but OpenMPTCProuter status page shows VPN Tunnel Down as explained in issue #366

Ysurac, 2019-03-28:
@cron410 This issue will be solved in next release.

cron410, 2019-03-28:
Thanks for your hard work. This system is working leaps and bounds better with every release.

Ysurac, 2019-04-06:
Should be fixed in OpenMPTCProuter v0.48

closed 2019-04-06 by Ysurac