Ysurac/openmptcprouter #411: Small change for ppp connections

TheCrach, 2019-03-27:
Hello,

It would be great to make a change inside openmptcprouter.lua, pppoe interfaces are displayed as down on the status page while omr-tracker report it OK, I narrowed down the problem to that lua file.

Is it possible to get l3_device from get_device() function first instead of section["ifname"] ? At least when proto = 'pppoe'

uci show | grep "wan1"
> openmptcprouter.wan1=interface
openmptcprouter.wan1.mtu='1500'
openmptcprouter.wan1.state='up'
openmptcprouter.wan1.asn='AS3215'
openmptcprouter.wan1.mptcp_status='-tracebox'
openmptcprouter.wan1.publicip='83.204.xx.xx'
openmptcprouter.wan1.lc='1553706747'
openmptcprouter.wan1.latency_previous='17'
openmptcprouter.wan1.latency='22'
network.wan1=interface
network.wan1.proto='pppoe'
network.wan1.ifname='orangedsl'
network.wan1.force_link='1'
network.wan1.username='fti/xxxxxxx'
network.wan1.password='xxxxxx'
network.wan1.ipv6='auto'
network.wan1.macaddr='xx:xx:xx:xx:xx:xx'
network.wan1.metric='12'
network.wan1.peerdns='0'
network.wan1.dns='1.1.1.1'
network.wan1.multipath='on'

ubus call network.interface.wan1 status
> {
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 16448,
        **"l3_device": "pppoe-wan1",**
        "proto": "pppoe",
        "device": "orangedsl",
        "updated": [
                "addresses",
                "routes"
        ],
        "metric": 12,
        "dns_metric": 0,
        "delegation": true,
        "ipv4-address": [
                {
                        "address": "83.204.xx.xx",
                        "mask": 32,
                        "ptpaddress": "193.253.160.3"
                }
        ],
[...]

[image]

Ysurac, 2019-03-27:
Good catch. This will be fixed in next release.

closed 2019-03-27 by Ysurac

TheCrach, 2019-03-28:
Additional informations

_post-tracking_ also has to be modified to get a fully working solution

`interface_if=$(ifstatus "$OMR_TRACKER_INTERFACE" 2>/dev/null | jsonfilter -q -e '@["l3_device"]')`
must be checked before
`interface_if=$(uci -q get network.$OMR_TRACKER_INTERFACE.ifname)`

Then _ip route_ must use _$interface_if_ as _dev_ because all set_server_XXX() functions and main code are using _$OMR_TRACKER_DEVICE_ as _dev_ ( while you already started using _$interface_if_ inside _set_route()_ )

Ysurac, 2019-03-28:
I fixed this too. And OMR_TRACKER_DEVICE will be the correct device now.