Ysurac/openmptcprouter #412: Forwarding all ports to pfSense

forid786, 2019-03-28:
## Expected Behavior
I am attempting to forward all ports coming to the VPS to a pfSense firewall sitting behind OpenMPTCP router as I prefer/want to manage all the firewalling on the pfSense

Setup: pfSense getting DHCP from OpenMPTCP (192.168.100.139)

Port Forward: https://i.imgur.com/pmUSbj3.png
SNAT: https://i.imgur.com/jdrUjjl.png

## Actual Behavior
When hitting VPS IP from WAN, the pfSense box gets no traffic at all from MPTCP 

## Steps to Reproduce the Problem

  1. All above
  2.
  3.

## Specifications

  - OpenMPTCProuter version:  v0.47.6
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.991
  - OpenMPTCProuter platform: x86_64

If anyone has got a similar setup working, please do let me know!

Ysurac, 2019-03-28:
In System->OpenMPTCProuter->Advanced settings, do you check "Redirects all ports from server to OpenMPTCProuter" ?

forid786, 2019-03-28:
Thank you for your reply. Yes that is ticked and in the shorewall configs on the vps there's no '#' on the last two lines. Is there anything else I should be checking?

Thanks

Ysurac, 2019-03-28:
So all traffic should go to OpenMPTCProuter. What is the VPN used (Glorytun TCP, Glorytun UDP, MLVPN,...) ?

forid786, 2019-03-28:
The VPN I'm using is MLVPN but traffic doesn't seem to be going to the pfSense behind the OpenMPTCP Router.

Ysurac, 2019-03-28:
If you try Glorytun TCP it's working or same problem ?

Ysurac, 2019-03-28:
Yes Glorytun TCP is the recommended VPN for now.

forid786, 2019-03-28:
Alright, I have switched VPN to Glorytun TCP and the Port forwarding still seems to be an issue.

Setting up a packet capture on the WAN interface of the pfSense, I am getting no traffic at all. Something is wrong on the OpenMPTCP Router end. On my mobile which I am trying to connect with over data is getting a connection refused. Have you got any other ideas? I think it most likely is to do with the configuration of the OpenMPTCP end, but I am not sure what it could be. I would really appreciate any help with getting this up and running.

Ysurac, 2019-03-28:
Check that:
* in ```/etc/shorewall/params.vpn``` on the VPS, check that values are correct, with Glorytun TCP this should be _VPS_ADDR=10.255.255.1_,_ OMR_ADDR=10.255.255.2_ and _VPS_IFACE=gt-tun0_.
* try ```ping 10.255.255.2``` from the VPS
* OpenMPTCProuter use iptables, so check with ```iptables-save``` that your rule is here
* tcpdump is also available on OpenMPTCProuter, so ```tcpdump -i tun0``` should show you if something come to OpenMPTCProuter or not.

forid786, 2019-03-28:
The values in: /etc/shorewall/params.vpn all seem to be correct. I am able to ping 10.255.255.2 from the VPS. However the rules do not seem to exist when I run iptables-save. 

I'm terrible with iptables, but do you just want me to add the rules there? If so, how would it look like in order to forward all ports to 192.168.100.139 behind  OpenMPTCP?

Ysurac, 2019-03-28:
With your settings,you should have this in iptables:
```
-A zone_vpn_prerouting -p tcp -m comment --comment "!fw3: All" -j DNAT --to-destination 192.168.100.139
-A zone_vpn_prerouting -p udp -m comment --comment "!fw3: All" -j DNAT --to-destination 192.168.100.139
```
(it's the result of iptables-save | grep 192.168.100.139 for me with your rule)
If you don't have this, I would need the result of ```uci show firewall```

forid786, 2019-03-28:
I tried re-adding those rules, but running iptables-save | grep 192.168.100.139 still doesn't show the rules. Weird...

uci show firewall isn't recognised when run on the VPS, is this something you want me to run on the Open MPTCP Router?

Ysurac, 2019-03-28:
Both commands should be run on the router.

forid786, 2019-03-28:
Oops, my bad. I'll run both from Open MPTCP for you ASAP once I'm back home and post here for you.

forid786, 2019-03-28:
`
root@OpenMPTCProuter:~# iptables-save | grep 192.168.100.139
-A zone_lan_postrouting -d 192.168.100.139/32 -p tcp -m comment --comment "!fw3: test" -j SNAT --to-source 192.168.100.1
-A zone_lan_postrouting -d 192.168.100.139/32 -p udp -m comment --comment "!fw3: test" -j SNAT --to-source 192.168.100.1
-A zone_vpn_prerouting -p tcp -m comment --comment "!fw3: Forward all to pfsense" -j DNAT --to-destination 192.168.100.139
-A zone_vpn_prerouting -p udp -m comment --comment "!fw3: Forward all to pfsense" -j DNAT --to-destination 192.168.100.139
root@OpenMPTCProuter:~#
`

`
root@OpenMPTCProuter:~# uci show firewall
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@defaults[0].forward='ACCEPT'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[0].network=' '
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@zone[1].input='ACCEPT'
firewall.@zone[1].network='wan2'
firewall.@zone[1].forward='ACCEPT'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
firewall.@rule[0].proto='udp'
firewall.@rule[0].dest_port='68'
firewall.@rule[0].target='ACCEPT'
firewall.@rule[0].family='ipv4'
firewall.@rule[1]=rule
firewall.@rule[1].name='Allow-Ping'
firewall.@rule[1].src='wan'
firewall.@rule[1].proto='icmp'
firewall.@rule[1].icmp_type='echo-request'
firewall.@rule[1].family='ipv4'
firewall.@rule[1].target='ACCEPT'
firewall.@rule[2]=rule
firewall.@rule[2].name='Allow-IGMP'
firewall.@rule[2].src='wan'
firewall.@rule[2].proto='igmp'
firewall.@rule[2].family='ipv4'
firewall.@rule[2].target='ACCEPT'
firewall.@rule[3]=rule
firewall.@rule[3].name='Allow-DHCPv6'
firewall.@rule[3].src='wan'
firewall.@rule[3].proto='udp'
firewall.@rule[3].src_ip='fc00::/6'
firewall.@rule[3].dest_ip='fc00::/6'
firewall.@rule[3].dest_port='546'
firewall.@rule[3].family='ipv6'
firewall.@rule[3].target='ACCEPT'
firewall.@rule[4]=rule
firewall.@rule[4].name='Allow-MLD'
firewall.@rule[4].src='wan'
firewall.@rule[4].proto='icmp'
firewall.@rule[4].src_ip='fe80::/10'
firewall.@rule[4].icmp_type='130/0' '131/0' '132/0' '143/0'
firewall.@rule[4].family='ipv6'
firewall.@rule[4].target='ACCEPT'
firewall.@rule[5]=rule
firewall.@rule[5].name='Allow-ICMPv6-Input'
[... 96 lines cut ...]
firewall.allow_dhcp_request_vpn.family='ipv4'
firewall.miniupnpd=include
firewall.miniupnpd.type='script'
firewall.miniupnpd.path='/usr/share/miniupnpd/firewall.include'
firewall.miniupnpd.family='any'
firewall.miniupnpd.reload='1'
firewall.@redirect[0]=redirect
firewall.@redirect[0].target='SNAT'
firewall.@redirect[0].src='vpn'
firewall.@redirect[0].src_dip='192.168.100.1'
firewall.@redirect[0].dest='lan'
firewall.@redirect[0].proto='tcp udp'
firewall.@redirect[0].dest_ip='192.168.100.139'
firewall.@redirect[0].name='test'
firewall.@redirect[1]=redirect
firewall.@redirect[1].target='DNAT'
firewall.@redirect[1].src='vpn'
firewall.@redirect[1].dest='lan'
firewall.@redirect[1].proto='tcp udp'
firewall.@redirect[1].dest_ip='192.168.100.139'
firewall.@redirect[1].name='Forward all to pfsense'
firewall.@redirect[1].reflection='0'
`

It seems like it's there, but still can't get port forward working, I'm sure I'm missing something.

Ysurac, 2019-03-28:
You made the test with ```tcpdump -i tun0``` on the router ?

forid786, 2019-03-28:
I ran tcpdump -i tun0 | grep 8096 because that is the port I am trying to connect on, it seems to be coming through. See below:
`
root@OpenMPTCProuter:~# tcpdump -i tun0 | grep 8096
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on tun0, link-type RAW (Raw IP), capture size 262144 bytes
21:21:03.022612 IP 213.205.240.53.45330 > 10.255.255.2.8096: Flags [S], seq 1098871763, win 65535, options [mss 1362,sackOK,TS val 1056516 ecr 0,nop,wscale 8], length 0
21:21:03.022649 IP 10.255.255.2.8096 > 213.205.240.53.45330: Flags [R.], seq 0, ack 1098871764, win 0, length 0
21:21:03.022703 IP 213.205.240.53.45328 > 10.255.255.2.8096: Flags [S], seq 2844838175, win 65535, options [mss 1362,sackOK,TS val 1056515 ecr 0,nop,wscale 8], length 0
21:21:03.022730 IP 10.255.255.2.8096 > 213.205.240.53.45328: Flags [R.], seq 0, ack 2844838176, win 0, length 0
21:21:03.077084 IP 213.205.240.53.45332 > 10.255.255.2.8096: Flags [S], seq 2268483818, win 65535, options [mss 1362,sackOK,TS val 1056591 ecr 0,nop,wscale 8], length 0
21:21:03.077125 IP 10.255.255.2.8096 > 213.205.240.53.45332: Flags [R.], seq 0, ack 2268483819, win 0, length 0
21:21:03.281419 IP 213.205.240.53.45334 > 10.255.255.2.8096: Flags [S], seq 356340526, win 65535, options [mss 1362,sackOK,TS val 1056655 ecr 0,nop,wscale 8], length 0
21:21:03.281464 IP 10.255.255.2.8096 > 213.205.240.53.45334: Flags [R.], seq 0, ack 356340527, win 0, length 0
21:21:03.281597 IP 213.205.240.53.45336 > 10.255.255.2.8096: Flags [S], seq 3775803773, win 65535, options [mss 1362,sackOK,TS val 1056656 ecr 0,nop,wscale 8], length 0
21:21:03.281623 IP 10.255.255.2.8096 > 213.205.240.53.45336: Flags [R.], seq 0, ack 3775803774, win 0, length 0
21:21:04.868923 IP 213.205.240.53.45340 > 10.255.255.2.8096: Flags [S], seq 989282377, win 65535, options [mss 1362,sackOK,TS val 1057110 ecr 0,nop,wscale 8], length 0
21:21:04.868963 IP 10.255.255.2.8096 > 213.205.240.53.45340: Flags [R.], seq 0, ack 989282378, win 0, length 0
21:21:04.868972 IP 213.205.240.53.45338 > 10.255.255.2.8096: Flags [S], seq 2395300360, win 65535, options [mss 1362,sackOK,TS val 1057109 ecr 0,nop,wscale 8], length 0
21:21:04.868993 IP 10.255.255.2.8096 > 213.205.240.53.45338: Flags [R.], seq 0, ack 2395300361, win 0, length 0
^C61 packets captured
61 packets received by filter
0 packets dropped by kernel
`

forid786, 2019-03-28:
As a side note, since switching to GloryTun TCP - I am getting drop outs, every so often, web pages will freeze and stop loading. In terms of speedtests though, they appear to be pretty much the same as what I got with MLVPN.

Ysurac, 2019-03-28:
Glorytun TCP shouldn't change anything for page: it is used only for UDP traffic. All TCP traffic use Shadowsocks.
I tested your rule (DNAT to ip & SNAT src 192.168.100.1) on my install, and forwarding is working. Maybe I don't see something in your firewall rules...

forid786, 2019-03-28:
Firewall Zone Settings: https://i.imgur.com/o43RpU2.png
Port Forwards: https://i.imgur.com/hH3NWwn.png
Firewall traffic rules 1: https://i.imgur.com/ne17YeC.png
Firewall traffic rules 2: https://i.imgur.com/Y26eHWz.png

Please let me know if you need anything else. pfSense has a NAT rule to NAT anything coming into it's WAN interface on 8096 to go to my Emby media server internally. That side I know is fine, as pre-MPTCP implementation when it was directly connected to WAN. It worked fine without issues.

Ysurac, 2019-03-28:
You can try a tcpdump -i eth0 port 8096 to check if the traffic try to go out.
It's all ideas I have for now.

forid786, 2019-03-28:
`root@OpenMPTCProuter:~# tcpdump -i eth0 port 8096
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
^C
0 packets captured
47 packets received by filter
37 packets dropped by kernel
`

All blank at the moment, when trying to request the page from WAN.

forid786, 2019-04-21:
I decided to revisit MPTCP. I re-installed everything from scratch since a newer version was released and now port forward seems to be working perfectly. Using Glorytun TCP. 

I will go ahead and close the issue as I have achieved what I wanted. All the port forwarding is now controlled by my pfSense, and OpenMPTCP is just acting as a gateway and adding in the MPTCP support.

closed 2019-04-21 by forid786