Ysurac/openmptcprouter #413: Freezing happens

manueldev, 2019-03-28:
## Expected Behavior
Access to the luci interface without problems. Internet connection without interruptions.
I am connecting the openmptcprouter to two internet connections. One connection is HFC and another LTE from a Moto G5 Plus per hotspot.

[image]

## Actual Behavior
From one moment to another, I lose connection to the internet and access to openmtcprouter (ssh and luci) but my computer continues to respond pings from 192.168.100.1.

dmesg shows the following at that moment:

> [  197.865280] TCP: mptcp_fallback_infinite 0xcfe30eb6 will fallback - pi 1, src 192.168.43.212:51649 dst 130.104.228.140:80 rcv_nxt 1159364016 from tcp_rcv_established+0x3a8/0x4e8
> [  402.484379] Meta already closed i_rcv 1 i_snd 1 send_i 0 flags 0x2000309
> [  402.492122] ------------[ cut here ]------------
> [  402.497637] WARNING: CPU: 3 PID: 0 at net/mptcp/mptcp_ctrl.c:646 mptcp_sock_def_error_report+0xb0/0xf8
> [  402.508777] Modules linked in: rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial option ipw brcmfmac usb_wwan usb_serial_simple ti_usb_3410_5052 sierra rtlwifi rtl8xxxu rt2x00usb rt2x00lib pppoe ppp_async pl2303 oti6858 nf_nat_pptp nf_conntrack_pptp mt7601u mos7720 mmc_spi mct_u232 mac80211 keyspan iptable_nat ipt_REJECT ipt_MASQUERADE garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute cypress_m8 cp210x ch341 cfg80211 cdc_mbim belkin_sa bas_gigaset ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS
> [  402.591342]  xt_REDIRECT xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor usbserial usbhid usb_gigaset ts_fsm ts_bm sr9700 spidev spi_gpio spi_bitbang sierra_net ser_gigaset rtl8150 rndis_host r8712u r8152 qmi_wwan pppox ppp_generic pegasus of_mmc_spi nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda mcs7830 macvlan kaweth kalmia
> [  402.675503]  iptable_raw iptable_mangle iptable_filter ipt_ECN ipheth ip6table_raw ip_tables huawei_cdc_ncm hso hid_generic gigaset ezusb ethoc ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 dm9601 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_subset cdc_ncm cdc_ether cdc_eem cdc_acm brcmutil ax88179_178a asix arptable_filter arpt_mangle arp_tables sch_cake sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred snd_bcm2835 hid evdev i2c_gpio i2c_algo_bit i2c_dev i2c_core spi_ks8995 xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip
> [  402.760066]  ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 nf_nat nf_conntrack nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 isdn_bsdcomp dss1_divert isdn slhc ifb sit sctp libcrc32c tunnel4 ip_tunnel veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore mISDN_dsp l1oip mISDN_core nls_utf8 sha256_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gf128mul gcm ctr fbtft sysimgblt sysfillrect syscopyarea fb_sys_fops nls_iso8859_1 nls_cp437 xhci_plat_hcd xhci_hcd ehci_platform
> [  402.844588]  ehci_hcd gpio_button_hotplug ptp broadcom bcm_phy_lib pps_core
> [  402.852913] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.104 #0
> [  402.860326] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
> [  402.867458] task: ffffffc038ca8000 task.stack: ffffff8008828000
> [  402.874666] PC is at mptcp_sock_def_error_report+0xb0/0xf8
> [  402.881394] LR is at mptcp_sock_def_error_report+0xb0/0xf8
> [  402.888083] pc : [<ffffff80085306e0>] lr : [<ffffff80085306e0>] pstate: 60000145
> [  402.897789] sp : ffffff800801bd30
> [  402.902217] x29: ffffff800801bd30 x28: ffffff8008737008 
> [  402.908660] x27: ffffff8008738598 x26: ffffff8008755180 
> [  402.915048] x25: 0000000000000100 x24: 0000000000000010 
> [  402.921418] x23: 0000000000000000 x22: ffffff8008789d00 
> [  402.927748] x21: ffffffc033505f80 x20: ffffffc034caca40 
> [  402.934046] x19: ffffffc031a018c0 x18: 0000000000000000 
> [  402.940311] x17: ffffff80085b2ac4 x16: ffffff80085b2ac4 
> [  402.946538] x15: ffffff80085b2ac4 x14: ffffff80085b2ac4 
> [  402.952724] x13: ffffff80085b2ac4 x12: ffffff80085b2ac4 
> [  402.958859] x11: ffffffc038f820a0 x10: 0000000000000001 
> [  402.964953] x9 : 0000000000000000 x8 : 0000000000000000 
> [  402.971019] x7 : 0000000000000007 x6 : 00000000000002bd 
> [  402.977040] x5 : 0000000000000008 x4 : 0000000000000003 
> [  402.983023] x3 : 0000000000000000 x2 : 0000000000000004 
> [  402.988978] x1 : 0000000000000004 x0 : 000000000000003c 
> [  402.994902] Call trace:
> [  402.997891] Exception stack(0xffffff800801bbf0 to 0xffffff800801bd30)
> [  403.005000] bbe0:                                   000000000000003c 0000000000000004
> [  403.014040] bc00: 0000000000000004 0000000000000000 0000000000000003 0000000000000008
> [  403.023155] bc20: 00000000000002bd 0000000000000007 0000000000000000 0000000000000000
> [  403.032404] bc40: 0000000000000001 ffffffc038f820a0 ffffff80085b2ac4 ffffff80085b2ac4
> [  403.041807] bc60: ffffff80085b2ac4 ffffff80085b2ac4 ffffff80085b2ac4 ffffff80085b2ac4
> [  403.051331] bc80: 0000000000000000 ffffffc031a018c0 ffffffc034caca40 ffffffc033505f80
> [  403.061053] bca0: ffffff8008789d00 0000000000000000 0000000000000010 0000000000000100
> [  403.070925] bcc0: ffffff8008755180 ffffff8008738598 ffffff8008737008 ffffff800801bd30
> [  403.080973] bce0: ffffff80085306e0 ffffff800801bd30 ffffff80085306e0 0000000060000145
> [  403.091212] bd00: 0000000000000001 0000000000000000 0000007fffffffff 00000000f7d18add
> [  403.101534] bd20: ffffff800801bd30 ffffff80085306e0
> [  403.107685] [<ffffff80085306e0>] mptcp_sock_def_error_report+0xb0/0xf8
> [  403.115521] [<ffffff80084a9c44>] tcp_write_err+0x2c/0x48
> [  403.122099] [<ffffff80084aa2b8>] tcp_write_timeout+0x318/0x350
> [  403.129176] [<ffffff80084aa6d0>] tcp_retransmit_timer+0x1c8/0x380
> [  403.136489] [<ffffff80085383ec>] mptcp_sub_retransmit_timer+0x14/0xd0
> [  403.144135] [<ffffff80084aa964>] tcp_write_timer_handler+0xdc/0x248
> [  403.151581] [<ffffff80084aab1c>] tcp_write_timer+0x4c/0xf0
> [  403.158195] [<ffffff80080e4554>] call_timer_fn.isra.27+0x24/0x78
> [  403.165300] [<ffffff80080e4a2c>] run_timer_softirq+0x184/0x3a0
> [  403.172212] [<ffffff8008081040>] __do_softirq+0x100/0x1fc
> [  403.178652] [<ffffff8008099550>] irq_exit+0x80/0xd8
> [  403.184529] [<ffffff80080d2fa0>] __handle_domain_irq+0x88/0xb0
> [  403.191348] [<ffffff8008080cbc>] bcm2836_arm_irqchip_handle_irq+0xac/0xb8
> [  403.199114] Exception stack(0xffffff800882be30 to 0xffffff800882bf70)
> [  403.206523] be20:                                   0000000000000000 0000000000000000
[... 6 lines cut ...]
> [  403.274757] bf00: ffffff800801c000 f3fbfcbfff1579ff ffffffc038ca8000 ffffff800882bf70
> [  403.284876] bf20: ffffff800808475c ffffff800882bf70 ffffff8008084760 0000000000000145
> [  403.295141] bf40: ffffff8008018000 ffffff800801c000 ffffffffffffffff ffffff80080e15f4
> [  403.305484] bf60: ffffff800882bf70 ffffff8008084760
> [  403.311628] [<ffffff8008082870>] el1_irq+0xb0/0x140
> [  403.317772] [<ffffff8008084760>] arch_cpu_idle+0x10/0x18
> [  403.324327] [<ffffff80080c7f9c>] do_idle+0xec/0x1b8
> [  403.330393] [<ffffff80080c81a8>] cpu_startup_entry+0x20/0x28
> [  403.337239] [<ffffff800808c23c>] secondary_start_kernel+0x17c/0x188
[... cut ...]
  1. Flashing https://www.openmptcprouter.com/files/openmptcprouter-v0.47.6-r09530-64bb88841f-brcm2708-bcm2710-rpi-3-squashfs-sysupgrade.img.gz
  2. Disable DHCP on eth0 (I use static ips).
  3. Scan wifi android hotspot and connect.
  4. Enable MPTCP on wifi interface.
  5. Use wizard to setup openmptcpvps (for me dhcp on wan1 and wwan).
  6. And wait (make udp and tcp connections for test and check realtime graphs on luci).

## Specifications

  - OpenMPTCProuter version: 0.47.6
  - OpenMPTCProuter VPS version: 0.991
  - OpenMPTCProuter platform: RPI3 64 bit

Ysurac, 2019-03-28:
The VPS is rebooted to latest release and with ```uname -a``` you get the MPTCP kernel ?
Fo you have often the "mptcp_fallback_infinite" errors ? (It's when MPTCP is not detected, this use normal TCP). It's possible this happen when you are in 3G, should not be a big problem.
For the freeze, you should check power supply and RPI temperature. You have to reboot each times after a freeze ?

manueldev, 2019-03-28:
> The VPS is rebooted to latest release and with `uname -a` you get the MPTCP kernel ?

[image]

> Fo you have often the "mptcp_fallback_infinite" errors ? (It's when MPTCP is not detected, this use normal TCP). It's possible this happen when you are in 3G, should not be a big problem.

[image]

For 6 hours I only have 1 error mptcp_fallback_infinite in the kernellog (although I have more disconnections within those 6 hours).

> For the freeze, you should check power supply and RPI temperature. You have to reboot each times after a freeze ?

I'm going to check the temperature and the energy provider. I do not need to restart. The system returns to work by itself after a few seconds (30 approximate).

manueldev, 2019-03-29:
I stopped the server to try a direct connection for a few hours. For now I have no interruptions.

[image]

Ysurac, 2019-03-29:
With direct output nothing is encrypted, so no much use of CPU.

manueldev, 2019-03-29:
Now I get 50ºC cpu temp. There is not much difference.

manueldev, 2019-04-06:
I discovered that the router of my internet provider, despite changing it to a different network segment than the default, still continues to ping 192.168.100.1 through the ethernet ports. I think this generated the problems. I changed the ip of openmptcrouter to 192.168.99.1 and (for now) it works.
I made the change by ssh using:
```
uci set network.lan.ipaddr = '192.168.99.1'
uci commit && service network restart
reboot
```

closed 2019-04-08 by Ysurac