Ysurac/openmptcprouter #414: Suggestion: Disable TCP Fastopen for Shadowsocks on errors

TheCrach, 2019-03-31:
I suggest to auto-disable TCP Fastopen on Shadowsocks when there are few errors in the logs (let's say something like 5 / 10 lines in the last 10 minutes, something like that):

> Sun Mar 31 10:52:25 2019 daemon.err /usr/bin/ss-redir[9942]: remote recv: Operation timed out
> Sun Mar 31 10:57:56 2019 daemon.err /usr/bin/ss-redir[9942]: remote recv: Operation timed out
> Sun Mar 31 10:57:56 2019 daemon.err /usr/bin/ss-redir[9940]: remote recv: Operation timed out
> Sun Mar 31 10:57:59 2019 daemon.err /usr/bin/ss-redir[9941]: remote recv: Operation timed out
> Sun Mar 31 10:57:59 2019 daemon.err /usr/bin/ss-redir[9942]: remote recv: Operation timed out

It was driving me crazy because websites randomly did not work properly (SSL_VERSION_INTERFERENCE and so on) because the connection couldn't be established.
It could be related to the drivers of my Ethernet-USB dongles or something else but disabling TCP FastOpen on Shadowsocks fixed the problem, so if disabling was automatic it would probably prevent issues for others.

Another solution: disable FastOpen by default.

Ysurac, 2019-03-31:
FastOpen is working in most case. There is some know problem in China and on some 4G networks.
It's difficult to detect when fastopen is really the problem...

dnwk, 2019-04-01:
I thought this is related to my case. I try to disable Fast Open but no significant improvement.

user747, 2019-04-01:
I used to experience something similar to this. I narrowed it down to internal network DNS issue. I'm also using Google Wi-Fi with 3 Mesh AP's. Google Wi-Fi integration isn't the greatest (slow NAT discovery / closed system)

Since using 4.14.108, I haven't experienced many issues at all with pages not loading. My ping times to local city are around 30-40ms. 

I'm testing this 90-shadowsocks.conf and it's working well for me.

# local sysctl settings can be stored in this directory
# max open files
fs.file-max = 51200
# max read buffer
net.core.rmem_max = 67108864
# max write buffer
net.core.wmem_max = 67108864
net.core.optmem_max = 524287
# default read buffer
net.core.rmem_default = 32768
# default write buffer
net.core.wmem_default = 32768
# max processor input queue
net.core.netdev_max_backlog = 250000
# max backlog
net.core.somaxconn = 4096

# resist SYN flood attacks
net.ipv4.tcp_syncookies = 1
# reuse timewait sockets when safe
net.ipv4.tcp_tw_reuse = 1
# turn off fast timewait sockets recycling
net.ipv4.tcp_tw_recycle = 0
# short FIN timeout
net.ipv4.tcp_fin_timeout = 30
# short keepalive time
net.ipv4.tcp_keepalive_time = 1200
# outbound port range
net.ipv4.ip_local_port_range = 10000 65000
# max SYN backlog
net.ipv4.tcp_max_syn_backlog = 8192
# max timewait sockets held by system simultaneously
net.ipv4.tcp_max_tw_buckets = 5000
# turn on TCP Fast Open on both client and server side
net.ipv4.tcp_fastopen = 3
# TCP buffer
net.ipv4.tcp_mem = 25600 51200 102400
# UDP buffer
net.ipv4.udp_mem = 25600 51200 102400
# TCP receive buffer
net.ipv4.tcp_rmem = 4096 87380 67108864
# TCP write buffer
net.ipv4.tcp_wmem = 4096 65536 67108864
# turn on path MTU discovery
net.ipv4.tcp_mtu_probing = 0

# for low-latency network, use cubic instead
net.ipv4.tcp_congestion_control = bbr
net.core.default_qdisc = fq
# Default conntrack is too small
net.netfilter.nf_conntrack_max = 131072

# MPTCP settings
net.mptcp.mptcp_checksum = 0
net.mptcp.mptcp_syn_retries = 1

I have TCP Fast Open enabled on both rpi 3b+ and Debian 9.8VPS

root@OpenMPTCProuter:~# cat /proc/sys/net/ipv4/tcp_fastopen
3

debian@3:~$ cat /proc/sys/net/ipv4/tcp_fastopen
3

I'm also using another rpi 3 for Pi Hole DNS.

closed 2019-07-04 by Ysurac