Ysurac/openmptcprouter #415: DHCP don't work on wifi

bosseb, 2019-04-01:
Hello

i have a problem since i use OpenMPTCProuteur, on my PC with WIFI, DHCP don't works at all
  - OpenMPTCProuter version: 0.47.6-r09530-64bb88841f
  - OpenMPTCProuter VPS version: 0.991
  - OpenMPTCProuter platform: x86_64
Thanks

Ysurac, 2019-04-01:
Do you have DHCP enabled on Wifi interface in OpenMPTCProuter interfaces (Network->Interfaces) ?

bosseb, 2019-04-01:
i just have these interfaces 
[image]

Ysurac, 2019-04-01:
So you need to configure the wifi interface if you have one, because I think it's not just DHCP that is not working now.

bosseb, 2019-04-01:
on my INTEL NUC there is not wifi , but the wifi problem is with my 2 BOX (sfr box an 4g box)

Ysurac, 2019-04-01:
OK.
I don't know what is the configuration of these boxes. Maybe you can enable a DHCP server that listen only on wifi interface, or use DHCP relay.

Ysurac, 2019-04-02:
You can't also set change wifi network ? else you have to put it on same network as OpenMPTCProuter lan.

bosseb, 2019-04-02:
no i can't change wifi network on these 2 boxes, i can disable DHCP on them but there is nothing special for wifi dhcp ...

bosseb, 2019-04-02:
i have this NUC : https://www.intel.fr/content/www/fr/fr/products/boards-kits/nuc/kits/nuc7i3bnh.html
but , it have a wireless but i think OpenMPTCProuter platform: x86_64 don't install the driver ...

Ysurac, 2019-04-04:
iwl driver that should support this card will be available in next release.

closed 2019-04-05 by Ysurac