Ysurac/openmptcprouter #416: instable system

bosseb, 2019-04-02:
Hello

i have a problem my OpenMPTCProuter, it's very instable but i have no change something

OpenMPTCProuter version: 0.47.6-r09530-64bb88841f
OpenMPTCProuter VPS version: 0.991
OpenMPTCProuter platform: x86_64

Tue Apr  2 19:30:52 2019 daemon.err /usr/bin/ss-redir[6067]: remote recv: Operation timed out
Tue Apr  2 19:30:59 2019 daemon.err /usr/bin/ss-redir[6069]: remote recv: Operation timed out
Tue Apr  2 19:31:03 2019 daemon.err /usr/bin/ss-redir[6068]: remote recv: Operation timed out
Tue Apr  2 19:31:03 2019 daemon.err /usr/bin/ss-redir[6067]: remote recv: Operation timed out
Tue Apr  2 19:31:03 2019 daemon.err /usr/bin/ss-redir[6069]: remote recv: Operation timed out
Tue Apr  2 19:31:03 2019 daemon.err /usr/bin/ss-redir[6068]: remote recv: Operation timed out
Tue Apr  2 19:31:03 2019 daemon.err /usr/bin/ss-redir[6067]: remote recv: Operation timed out
Tue Apr  2 19:31:03 2019 daemon.err /usr/bin/ss-redir[6068]: remote recv: Operation timed out
Tue Apr  2 19:31:04 2019 daemon.err /usr/bin/ss-redir[6069]: remote recv: Operation timed out
Tue Apr  2 19:31:06 2019 daemon.err /usr/bin/ss-redir[6067]: remote recv: Operation timed out
Tue Apr  2 19:31:37 2019 daemon.err /usr/bin/ss-redir[6067]: remote recv: Operation timed out
Tue Apr  2 19:31:46 2019 daemon.err /usr/bin/ss-redir[6066]: remote recv: Operation timed out
Thanks

bosseb, 2019-04-02:
Tue Apr  2 20:16:19 2019 daemon.notice netifd: Network device 'eth0' link is down
Tue Apr  2 20:16:19 2019 daemon.notice netifd: Interface 'lan' has link connectivity loss
Tue Apr  2 20:16:19 2019 daemon.notice netifd: macvlan 'wan1' link is down
Tue Apr  2 20:16:19 2019 daemon.notice netifd: Interface 'wan1' has link connectivity loss
Tue Apr  2 20:16:19 2019 daemon.notice netifd: macvlan 'wan2' link is down
Tue Apr  2 20:16:19 2019 daemon.notice netifd: Interface 'wan2' has link connectivity loss
Tue Apr  2 20:16:23 2019 daemon.notice netifd: Network device 'eth0' link is up
Tue Apr  2 20:16:23 2019 daemon.notice netifd: Interface 'lan' has link connectivity
Tue Apr  2 20:16:23 2019 daemon.notice netifd: macvlan 'wan1' link is up
Tue Apr  2 20:16:23 2019 daemon.notice netifd: Interface 'wan1' has link connectivity
Tue Apr  2 20:16:23 2019 daemon.notice netifd: macvlan 'wan2' link is up
Tue Apr  2 20:16:23 2019 daemon.notice netifd: Interface 'wan2' has link connectivity
Tue Apr  2 20:16:23 2019 kern.info kernel: [ 2891.511142] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: No

Ysurac, 2019-04-02:
This looks like a hardware problem.
Ethernet link seems to have some problem.

bosseb, 2019-04-02:
what can i do ??

Ysurac, 2019-04-02:
check ethernet cables, switch,...

bosseb, 2019-04-02:
OK thanks

bosseb, 2019-04-02:
a little question : to make the router i use an INTEL NUC , your router image for x86_64 works fo all intel nuc or not ???

Ysurac, 2019-04-02:
If the image boot yes.

bosseb, 2019-04-02:
i want to install the radio (wifi) i have find the linux radio drive for Intel® Dual Band Wireless-AC 8265, how can i install it ... i can't access with filezilla in ssh mode port 22

Ysurac, 2019-04-03:
Support of this wifi interface will be added in next release (that will be out very soon).

bosseb, 2019-04-03:
Thank you so much :) 
i have always some errors
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909] e1000e 0000:00:1f.6 eth0: Detected Hardware Unit Hang:
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909]   TDH                  <ef>
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909]   TDT                  <67>
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909]   next_to_use          <67>
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909]   next_to_clean        <ed>
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909] buffer_info[next_to_clean]:
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909]   time_stamp           <1007bc2e4>
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909]   next_to_watch        <ef>
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909]   jiffies              <1007bc400>
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909]   next_to_watch.status <0>
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909] MAC Status             <40080083>
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909] PHY Status             <796d>
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909] PHY 1000BASE-T Status  <3800>
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909] PHY Extended Status    <3000>
Wed Apr  3 22:27:11 2019 kern.err kernel: [81409.847909] PCI Status             <10>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777] e1000e 0000:00:1f.6 eth0: Detected Hardware Unit Hang:
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777]   TDH                  <ef>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777]   TDT                  <67>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777]   next_to_use          <67>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777]   next_to_clean        <ed>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777] buffer_info[next_to_clean]:
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777]   time_stamp           <1007bc2e4>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777]   next_to_watch        <ef>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777]   jiffies              <1007bc4c8>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777]   next_to_watch.status <0>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777] MAC Status             <40080083>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777] PHY Status             <796d>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777] PHY 1000BASE-T Status  <3800>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777] PHY Extended Status    <3000>
Wed Apr  3 22:27:13 2019 kern.err kernel: [81411.847777] PCI Status             <10>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769] e1000e 0000:00:1f.6 eth0: Detected Hardware Unit Hang:
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769]   TDH                  <ef>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769]   TDT                  <67>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769]   next_to_use          <67>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769]   next_to_clean        <ed>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769] buffer_info[next_to_clean]:
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769]   time_stamp           <1007bc2e4>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769]   next_to_watch        <ef>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769]   jiffies              <1007bc590>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769]   next_to_watch.status <0>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769] MAC Status             <40080083>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769] PHY Status             <796d>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769] PHY 1000BASE-T Status  <3800>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769] PHY Extended Status    <3000>
Wed Apr  3 22:27:15 2019 kern.err kernel: [81413.847769] PCI Status             <10>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749] e1000e 0000:00:1f.6 eth0: Detected Hardware Unit Hang:
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749]   TDH                  <ef>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749]   TDT                  <67>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749]   next_to_use          <67>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749]   next_to_clean        <ed>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749] buffer_info[next_to_clean]:
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749]   time_stamp           <1007bc2e4>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749]   next_to_watch        <ef>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749]   jiffies              <1007bc658>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749]   next_to_watch.status <0>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749] MAC Status             <40080083>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749] PHY Status             <796d>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749] PHY 1000BASE-T Status  <3800>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749] PHY Extended Status    <3000>
Wed Apr  3 22:27:17 2019 kern.err kernel: [81415.847749] PCI Status             <10>
Wed Apr  3 22:27:18 2019 daemon.notice netifd: Network device 'eth0' link is down
[... cut ...]
Wed Apr  3 22:27:22 2019 kern.info kernel: [81420.699794] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
Wed Apr  3 22:27:24 2019 daemon.err /usr/bin/ss-redir[6096]: remote recv: Operation timed out
Wed Apr  3 22:27:56 2019 daemon.err glorytun[4811]: read: Operation timed out
Wed Apr  3 22:27:56 2019 daemon.info glorytun[4811]: STOPPED tun0
Wed Apr  3 22:27:57 2019 daemon.err glorytun[4811]: 51.15.229.76.65001: connected
Wed Apr  3 22:27:57 2019 daemon.info glorytun[4811]: STARTED tun0

Ysurac, 2019-04-03:
You can try ```ethtool -K eth0 gso off gro off tso off``` via SSH on OpenMPTCProuter and check if it's better or not (you will get lower perf). You can also try to disable enhanced C1 power saving state (C1E) in the BIOS settings.
This seems to be the same problem as https://github.com/Ysurac/openmptcprouter/issues/286

bosseb, 2019-04-03:
ok thank you 👍

bosseb, 2019-04-06:
hello i don't find Enhanced C1 power saving state (C1E) in the bios settings

Ysurac, 2019-04-06:
In fact any power saving should be stopped, at least for test.

bosseb, 2019-04-14:
hello power saving is stopped but i have always some instability
Sun Apr 14 19:41:30 2019 daemon.err /usr/bin/ss-redir[6547]: send: Broken pipe
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007] e1000e 0000:00:1f.6 eth0: Detected Hardware Unit Hang:
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007]   TDH                  <f8>
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007]   TDT                  <e3>
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007]   next_to_use          <e3>
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007]   next_to_clean        <f6>
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007] buffer_info[next_to_clean]:
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007]   time_stamp           <1003ef9be>
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007]   next_to_watch        <f8>
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007]   jiffies              <1003efa48>
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007]   next_to_watch.status <0>
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007] MAC Status             <40080083>
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007] PHY Status             <796d>
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007] PHY 1000BASE-T Status  <3800>
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007] PHY Extended Status    <3000>
Sun Apr 14 19:41:52 2019 kern.err kernel: [41572.091007] PCI Status             <10>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645] e1000e 0000:00:1f.6 eth0: Detected Hardware Unit Hang:
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645]   TDH                  <f8>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645]   TDT                  <e3>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645]   next_to_use          <e3>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645]   next_to_clean        <f6>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645] buffer_info[next_to_clean]:
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645]   time_stamp           <1003ef9be>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645]   next_to_watch        <f8>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645]   jiffies              <1003efb10>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645]   next_to_watch.status <0>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645] MAC Status             <40080083>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645] PHY Status             <796d>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645] PHY 1000BASE-T Status  <3800>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645] PHY Extended Status    <3000>
Sun Apr 14 19:41:54 2019 kern.err kernel: [41574.090645] PCI Status             <10>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585] e1000e 0000:00:1f.6 eth0: Detected Hardware Unit Hang:
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585]   TDH                  <f8>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585]   TDT                  <e3>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585]   next_to_use          <e3>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585]   next_to_clean        <f6>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585] buffer_info[next_to_clean]:
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585]   time_stamp           <1003ef9be>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585]   next_to_watch        <f8>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585]   jiffies              <1003efbd8>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585]   next_to_watch.status <0>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585] MAC Status             <40080083>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585] PHY Status             <796d>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585] PHY 1000BASE-T Status  <3800>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585] PHY Extended Status    <3000>
Sun Apr 14 19:41:56 2019 kern.err kernel: [41576.090585] PCI Status             <10>
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090351] e1000e 0000:00:1f.6 eth0: Reset adapter unexpectedly
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572] e1000e 0000:00:1f.6 eth0: Detected Hardware Unit Hang:
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572]   TDH                  <f8>
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572]   TDT                  <e3>
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572]   next_to_use          <e3>
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572]   next_to_clean        <f6>
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572] buffer_info[next_to_clean]:
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572]   time_stamp           <1003ef9be>
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572]   next_to_watch        <f8>
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572]   jiffies              <1003efca0>
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572]   next_to_watch.status <0>
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572] MAC Status             <40080083>
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572] PHY Status             <796d>
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572] PHY 1000BASE-T Status  <3800>
Sun Apr 14 19:41:58 2019 kern.err kernel: [41578.090572] PHY Extended Status    <3000>
[... cut ...]
Sun Apr 14 19:42:02 2019 daemon.notice netifd: macvlan 'wan3' link is up
Sun Apr 14 19:42:02 2019 daemon.notice netifd: Interface 'wan3' has link connectivity
Sun Apr 14 19:42:02 2019 kern.info kernel: [41581.802641] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
Sun Apr 14 19:44:51 2019 daemon.info dnsmasq-dhcp[6225]: DHCPREQUEST(eth0) 192.168.100.235 9c:b6:d0:65:fd:0b
Sun Apr 14 19:44:51 2019 daemon.info dnsmasq-dhcp[6225]: DHCPACK(eth0) 192.168.100.235 9c:b6:d0:65:fd:0b DESKTOP-E281IAJ

Ysurac, 2019-04-14:
And no change with ```ethtool -K eth0 gso off gro off tso off``` ?

bosseb, 2019-04-14:
could you tell me if i need to use this command each time i restart openmptcrouter ?

Ysurac, 2019-04-14:
It's a test, so yes...
Then you can test to disable only what is working best for you.

Ysurac, 2019-04-14:
You can also try ```ethtool -K eth0 tx off rx off```

bosseb, 2019-04-14:
ethtool -K eth0 gso off gro off tso off 
is now active, no bug at all, but it's very strange, because there was no bug during more 24 hours and Suddenly the bug is coming

bosseb, 2019-04-14:
could you explain me what this command line do ?

Ysurac, 2019-04-14:
gso = Generic Segmentation Offload
gro = Generic Reveive Offload
tso = TCP Segmentation Offload

bosseb, 2019-04-15:
this morning my numéricable box was out , impossible to have internet with the 2 orther box who have internet, so i had restore a backup and after it works
so , now i'm looking events of the day and there is a lot of these lines :
Mon Apr 15 11:11:39 2019 daemon.info dnsmasq-dhcp[17861]: DHCPREQUEST(eth0) 192.168.100.120 xxxxxxxxx
Mon Apr 15 11:11:39 2019 daemon.info dnsmasq-dhcp[17861]: DHCPACK(eth0) 192.168.100.120 xxxxxxxxx Chambre

Ysurac, 2019-04-15:
Nothing special here: the computer chambre request IP 192.168.100.120 to DHCP server, and the DHCP say "ok I give it to you"

bosseb, 2019-04-15:
ok but it request this 50 times in 10 hours and nobody was at home ! it's normal ??

Ysurac, 2019-04-15:
A bit too much, you can check the lease time in network->interface and LAN interface.
And it's a request from a computer, no need to be at home for that, but the computer is at home (else you may have a problem ;) )

bosseb, 2019-04-15:
IT's my 2 apple tv who send request to DHCP !!!

lease time : 12 hours

Ysurac, 2019-04-15:
ok, there is always problems with apple devices and DHCP...

bosseb, 2019-04-15:
what can i do ?

Ysurac, 2019-04-15:
set a static IP on the Apple TV if it's possible

bosseb, 2019-04-15:
Yes of course !! thank you

closed 2019-07-04 by Ysurac