Ysurac/openmptcprouter #417: daemon.err /usr/bin/ss-local[11640]: getpeername: Socket not connected

suyuan168, 2019-04-03:
The system log has a lot of getpeername: Socket not connected with daemon.err /usr/bin/ss-redir[11641]: remote recv: Operation timed out
It is RPI3. It is the latest system. Is it because I have a lot of WANs but one of them without a network will affect the work of openmptcprouter? Causes all networks to be lost.
I found that the operator has intercepted shadowsocks-libev. Enable Enable ShadowSocks Obfuscating MPTCP aggregation without line broadband increase. How can I avoid it?

suyuan168, 2019-04-03:
Failed to establish connection

dnwk, 2019-04-03:
I test yesterday when I reduced 3 WAN to 2 WAN, it stabilized the connection.

suyuan168, 2019-04-04:
Thank you for your reply, I am going to test it.

closed 2019-04-04 by suyuan168