Ysurac/openmptcprouter #418: Can you replace shadowsock with shadowsocksrShadowsocksR?

suyuan168, 2019-04-04:
Can you replace shadowsock with shadowsocksrShadowsocksR?
Because operators in China are limiting Shadowsocks. See if ShadowsocksR can solve the problem that Shadowsocks is slow and the network is not normal.

suyuan168, 2019-04-04:
https://github.com/MPSec/MPSec Or use PPTP or L2TP

suyuan168, 2019-04-04:
SS/SSR/V2RAY WS+TLS?

Ysurac, 2019-04-04:
ShadowsocksR is no more developed and was partly closed source, so not possible.
MPSec use IPsec. IPsec, PPTP and L2TP are detected by GFW.
TLS will be supported with V2ray in a next release.

suyuan168, 2019-04-05:
Opening a web page is very slow. For example, pulling streaming media RTMP back to local speed is also very slow. Should be related to UDP. . But opening the webpage is very slow. It feels the same as DNS. The page needs to be refreshed many times to get the card. How is this going. Yesterday I tried to turn off tpc_fast_open but it still looks like this. Opened v2ray still has not improved. Anything you can try is tried. Is it related to the DNS of the VPS? That is, the page speed is very slow and affects access.

suyuan168, 2019-04-05:
Is it the reason for SS? So why not replace Shadowsocks directly with V2Ray?

Ysurac, 2019-04-05:
You can try to change the DNS. By default root server are used.
V2Ray doesn't have any source package for OpenWrt and it's a nightmare to made them.
Shadowsocks is used because it's a fast socks proxy.

suyuan168, 2019-04-05:
I don't know if it's Shadowsocks or mptcp these days, it's very slow to open pages, it's not good for udp data, and sometimes it's extremely rare that broadband can be maximized. I hope you can come out earlier and improve on the next version.

closed 2019-04-05 by Ysurac