Ysurac/openmptcprouter #419: What speed can i get from a single core 2.4Ghz 32 Bit processor ?

RomanJos, 2019-04-05:
Hi i'm about to buy a dedicated pc for OpenMptcp, however i don't know if the two that i found can aggregate 2x 75Mbps.
I also didn't find any minimum requirement etc but if a raspberry can handle ~150Mbps i don't think i need a pc from NASA.
Here is the only specs that i have:
First one is a pentium4 2.4Ghz 32bit HT512K with only one cores with 512Mo of RAM and the second is a 1.8Ghz single core 64bit AMD sempron 3500+ with 1.8gb of RAM.
Thanks in advance :relaxed:
Ps: i don't know if i should put this in issue or not sorry if its the wrong section

Ysurac, 2019-04-05:
All I can say is that RPI3B+ can aggregate at least 150Mbps.
I have no idea how perform these processors versus the ARM Cortex-A53 in the RPI3, but the amd one should be good ;) Check that you have gigabit ethernet card.

RomanJos, 2019-04-08:
Hi, the seller didn't sell it to me because another guy took it before me so i searched again but the only pc under 50€ with some minimal performance was the pentium4 but at this moment i realized that some friends can have some old pc that they didn't use so i called one and now i have an AMD Athlon 64 3400+ that run at 2.2Ghz with 1g of RAM so i think its okay but one problem here is that i can't boot :
[image] its blocked here
i tried with my previous hdd wich has my config already installed and also reinstalling with etcher on another hdd and its giving me the same result. i tried also my opnsense hdd that i have cloned with dd one month ago and it went one line more than "FreeBSD F1 : boot" saying that he didn't find the boot or something like that so i don't know why it is doing this except maybe its related to the uefi or something like that but i'm not sure at all.

RomanJos, 2019-04-08:
It work ! But only with a usb key otherwise not. I'm going to try with a small hdd to see because i tried 2To and 500Go maybe its not supporting so much

closed 2019-04-08 by RomanJos

PhatHub, 2019-04-10:
I can confirm RomanJos's solution. I have a 1.2Ghz Celeron that couldn't boot from Compact Flash over ATA100, and a USB drive with a microSD card worked to full boot. 
As sad as it is that I can't use the compact flash card, I'll have to find a tiny USB drive instead! :)

Ysurac, 2019-04-10:
All drivers are not installed by default, you can try to find it and install it on the image on something that can boot it or you can tell me the driver needed and it will be available in next release.

PhatHub, 2019-04-10:
I'll try to do that in the future.
But for now I'm going to mount the thumbdrive internally to a motherboard USB header.

RomanJos, 2019-04-16:
For those who wondering what max speed i have get with the AMD Athlon 64 3400+ (Cpu to 100%) :
[image]
i have two Huawei bs525, two LowcostMobile rad58 and two FreeMobile unlimited 4G but the max speed i've got from free was 100Mbps (not often), i think they limit the speed per sector or per frequency ( if you want to know i tested 2600Mhz and 1800Mhz on only one device and i get 60-70Mbps for 1800Mhz and 20-30Mbps for 2600Mhz but it depend af the weather i didn't tested on a blue sky) if do a speed test on one huawei at a time, both give 100Mbps but if the two run a speedtest at the same time they drop to 50Mbps each (they are spaced from 80-100 meters to avoid any interference) so i've buyed a Red-By-SFR 100Go dataplan to be with a Free sim (because i need to have a minimum speed for my internet in case i download more than 100Go on the SFR sim lol) and by combining 1800Mhz and 800Mhz for the SFR cell i get 160-170Mbps 😆 so i replaced the amd athlon computer by another with  a 4 core AMD Phenom(tm) II X4 965 Processor and now i have this haha 😃 : 
[image]
Fortnite update never been so fast

PhatHub, 2019-04-16:
@RomanJos Very cool! 
I think my Celeron 220 (Tualatin era) is too slow (connecting to Digital Ocean single core droplet). When I have two cell phones I get decent speed (16mbps) but when I put on a third phone, it lowers to 11mbps! I will maybe disable the encryption like @Ysurac  says in another issue (#382) , or just use a multiple core router like you did.