Ysurac/openmptcprouter #420: Support Pi-hole

Ysurac, 2019-04-05:
[Pi-hole](https://github.com/pi-hole/pi-hole) can be installed on VPS side.
It need to listen to vpn interface (gt-tun0, gt-udp-tun0 or tun0) so this need to be changed each time user change the vpn used (or try to listen to all interfaces ?). Same for lighttpd that need to listen only on vpn interfaces.

Ysurac, 2019-04-05:
Possible to make dnsmasq listen to all interface by adding a file with all interfaces in ```/etc/dnsmsq.d/```.
So this seems easy to add support for that.

user747, 2019-04-09:
I'm testing local instance of Pi Hole on 2nd rpi 3 with static IP. On OMR I have disabled Unbound, disabled 'Rebind protection' , and DNS forwardings pointing to Pi Hole.

Ysurac, 2019-04-10:
In next release, this will be possible to install Pi-hole on the VPS and dnsmasq will forward request to it if detected.

closed 2019-07-04 by Ysurac