Ysurac/openmptcprouter #422: Shadowsocks-libev open page is not normal, if closing shadowsocks-libev with glorytun-tcp normal

suyuan168, 2019-04-07:
Shadowsocks-libev open page is not normal, if closing shadowsocks-libev with glorytun-tcp normal
RPI3
Simply use glorytun-tcp and everything works fine. If you open the webpage with MPTCP+SS it is slow and not normal. Whether it is RTMP streaming or http browsing. But glorytun-tcp is normal. Turn off the Shadowsocks-libev switch and forward it.
Hi, hope is getting better and better

Ysurac, 2019-04-07:
Please put logs, reproducible tests and versions used...

suyuan168, 2019-04-07:
VPS MPTCP 0.94.4 
OPENMPTCPROUTER  v0.48 
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='222'
dropbear.@dropbear[0].GatewayPorts='on'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='AABBCCDDEEFF'
etherwake.@target[0].wakeonboot='off'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].network='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].network='wan' 'wan6' 'wan1' 'wan2' 'wan3' 'wan4'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
[... 947 lines cut ...]
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth3' 'eth4'
Powered by LuCI Master (git-19.095.24565-19d15f2) / openmptcprouter v0.48 r0+9745-b0395cfc56

Ysurac, 2019-04-08:
I don't really understand your problem. But as China user I would say that you should disable fastopen because it's know to make some problems on some LTE providers in China.

suyuan168, 2019-04-09:
I have banned openfast. But the webpage is still not very smooth. It’s really weird, I don’t know why.

closed 2019-04-10 by suyuan168