Ysurac/openmptcprouter #423: Android USB tethering

gwenaelvigor, 2019-04-08:
## Expected Behavior
Add possibility to use android usb tethering to add a new wan for openmptcprouter.

## Actual Behavior
I have test on 2 different android phone, and with usb internet sharing ( tethering) it not possible to deactivate the DHCP on this connection.
So it's difficult to configure this wan in OpenMPTCProuter.

## Steps to Reproduce the Problem

  1. Connect phone to one of OpenMPTCProuter usb port
  2. Activate usb tethering on phone
  3. Try to use this connection as wan in OpenMPTCProuter

## Specifications

  - OpenMPTCProuter version: 0.47.6
  - OpenMPTCProuter VPS version: 0.991
  - OpenMPTCProuter platform: RPI2

Ysurac, 2019-04-08:
Not enough info here. You try to configure it on network page ?

manueldev, 2019-04-09:
I use a Moto G5 plus with internet tethering and it works. By default Android gives ip addresses in the range "192.168.43.0" (at least on my phone) and it is not possible to disable it. I am not sure if it is possible to use more than one connected android phone as an internet source with openmptcprouter.

manueldev, 2019-04-09:
I just tried. I connected my android phone by usb. I activated share connection for usb. On the configuration screen of openmptcprouter I added the interface usb0. I configured it with dhcp to get an ip address from the phone.
[image]

dnwk, 2019-04-10:
I think if an interface is set to DHCP, on the configure page, it will refuse to add the interface in as wan for aggregation

Ysurac, 2019-04-10:
Not at all. Why are you saying that ?

gwenaelvigor, 2019-04-10:
I've try many things, so i will to resume what i have done and result. 

Configuration : 
I have connected my RaspPi 2 to one of ethernet port of my isp box. On my Isp box i have previously deactivate DHCP. 
I have configured like in wiki with wizard.
Until that everything work on system->openmptcprouter->state.

After i have had my phone (Xiaomi redmi note 5 with MIUI 10.2.2.0 [Android 8.1.0]) connected to pi on usb and activate internet sharing through usb. 
I never find how to deactivate dhcp with this connection

Things i have try to use this connection as wan2 in wizard : 
DHCP drop down selection. Result : No static ip, No gateway
Static Ip drop down selection : 
i have try many value that i have found using route -n; traceroute... ( connecting phone through usb directly to my computer) . 
Ipv4 or ipv4 gateway field : 192.168.42.129 ; 172.20.0.0 ; 192.168.42.184; 192.168.42.130; (192.168.43.XX is for wifi tethering ) 
Result each time was Gateway Down in state page.

Below screenshot of my last attempt : 
[image]

I have also try with my wife phone ( LG G6) but the result was the same

Ysurac, 2019-04-10:
You must use Network->Interface to configure it. You can search the doc of OpenWrt about Android and USB tethering.

closed 2019-07-04 by Ysurac