Ysurac/openmptcprouter #424: Another DNS problem

RomanJos, 2019-04-08:
I have a opnsense machine that is my main dns for my network but openmptcp seem to block it or something like that i tried to untick "authoritative" in the DHCP and DNS section and also stop unbound and dnsmask in Startup section (i click on enabled to make them disabled and then reboot) but it didn't worked.
Also i didn't reboot opnsense but i don't think its related to this

Ysurac, 2019-04-08:
OpenMPTCProuter can't block any other DNS on the network. But you really should use OpenMPTCProuter as DNS server, even if you redirect requests to your DNS (even if I have no idea why you want that).
Authoritative is for DHCP, by default DHCP also set DNS. If you have another DHCP server you can disable it for lan interface on network page.

RomanJos, 2019-04-09:
In fact i created an account on opendns to have all the cool stats etc and i wanted to keep it because it was cool lol.
For if openmptcp block dns i really feel i was doing this because i can go directly on 1.1.1.1 but not on google for example so that was weird. Anyway i will retry it

user747, 2019-04-09:
What do you have set i  'DNS forwardings' ?

http://192.168.100.1/cgi-bin/luci/admin/network/dhcp

Can also ssh into omr run this and paste output

root@OpenMPTCProuter:~# cat /etc/config/dhcp

Ysurac, 2019-04-09:
1.1.1.1,8.8.8.8,... are also used by omr-tracker, it's often not a good idea to use them as DNS server because these can have strange results.

RomanJos, 2019-04-12:
Hi i've got some news, i don't completly know what the fuck i did to have this problem but i ended up wits reseting my opnsense router and rebuild my network lol so now i have internet if i connect my computer directly to openmptcprouter but if i'm behind my firewall i can't, i don't completly know why because the ip of openmptcp is 192.168.100.1 and i put 192.168.100.5/24 for the WAN IP with 192.168.100.1 for the gateway and my dns are already set (i now have dns over tls or something like that). I then tried to create a Firewall rule for the WAN to allow all traffic from any direction bla bla bla but it didn't work so i tried to do a floating rule to allow the LAN and the WAN to talk but it obviously does nothing because without all of this i was having internet if i put my box for the wan connection. I can of course access to the openmptcp webpage behind my opnsense and they can ping each other. Do you have any idea ? Because i'm sure its something dumb that i didn't see by now but i have no idea

user747, 2019-04-12:
> reseting my opnsense router and rebuild my network
Sounds like OPNsense configuration issue.

> connect my computer directly to openmptcprouter
What is OMR running on? It sounds like you have done some troubleshooting, but sometimes it's something simple at Physical layer that is overlooked. 

Here is example from my network topology 
CLIENT-->USB 2.0 ETHERNET DONGLE-->SWITCH2-->SWITCH1-->GOOGLE WIFI-->RPI3 (OMR)-->MODEM-->TELCO

OMR already has built in Firewall. It might be a case of tweaking rules so it plays nicely with 
https://openwrt.org/docs/guide-user/firewall/firewall_configuration

Not necessarily relevant to your situation, but here is a guide regarding OpenWRT with OPNsense
https://techblog.jeppson.org/2018/02/site-site-vpn-opnsense-openwrt-tinc/

Also some further docs that might have some fruit 
https://wiki.opnsense.org/
https://forum.opnsense.org/index.php?board=1.0

RomanJos, 2019-04-12:
I solved it ! i have no idea why i needed to do this but this is what solved my problem : 
[image]
when i tried to ping 1.1.1.1 it instantly returned no route to host instead of timeout or something "normal" so i just added the 0.0.0.0/0 network and the openmptcprouter as gateway for this route and now it work i have completely no idea why i needed to change that but it work so i'm pretty happy the next thing now is to figure it out why i have like one connection used haha but this might be easy i will try tomorrow with a quad core cpu because i have right now a single core Athlon 64 3400+. so thanks everyone who helped me i appreciate it 😄

Also : why i need to have openmptcprouter in dns to make omr-bypass work (with domain not with ip) ? it would be better if i can put a domain name and it check every x minutes the ip of this domain and update omr bypass ip list because i wanted to go to cpu-world and i have been blocked, the only option to make omr bypass was to get the ip of cpu-world and put it in the domain

closed 2019-04-12 by RomanJos

Ysurac, 2019-04-12:
OpenMPTCProuter need to be DNS for omr-bypass because I have no way to know what IP will be used if it's not the case. A domain can (and often have) multiple IPs. 
But dnsmasq can redirect to any DNS you want (by default it's root dns server).