Ysurac/openmptcprouter #425: Not bonding 2 DSL lines with 1 4G router

FREEEEMANX, 2019-04-09:
## Expected Behavior

Bonding those 3 connections

## Actual Behavior

It's either the 2 DSL lines are bonded and the 4G is excluded or only the 4G router connection is used.

## Steps to Reproduce the Problem

  1. When master is set to one of the 2 DSL lines bonding works only for those 2 connections
  2. When master is set to the 4G router, only the this connection is working
  3. Through separates speedtest via web services or omr through shell

## Specifications

  - OpenMPTCProuter version: 0.46.3 
  - OpenMPTCProuter VPS version: 0.991 BUT had same problem with older version
  - OpenMPTCProuter platform: x86_64

On the status, also says multipath seems to be blocked on the connection on the 4G router.

Here is the uci show network :

network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='1'
network.globals=globals
network.globals.ula_prefix='fd75:af3f:f2f9::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='default'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='5'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='00:0f:fe:c1:1b:bb'
network.lan.metric='2'
network.lan.ipv6='0'
network.wan1=interface
network.wan1.ifname='eth1'
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.defaultroute='0'
network.wan1.macaddr='6c:b3:11:50:82:de'
network.wan1.metric='3'
network.wan1.ipaddr='192.168.10.2'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.10.1'
network.wan1.label='ADSL 1 192.168.10.1'
network.wan1.multipath='on'
network.wan2=interface
network.wan2.ifname='eth2'
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.defaultroute='0'
network.wan2.macaddr='6c:b3:11:50:82:df'
network.wan2.metric='4'
network.wan2.ipaddr='192.168.11.2'
network.wan2.netmask='255.255.255.0'
network.wan2.gateway='192.168.11.1'
network.wan2.label='ADSL 2 192.168.11.1'
network.wan2.multipath='master'
network.wan3=interface
network.wan3.ifname='eth3'
network.wan3.proto='static'
network.wan3.ip4table='wan'
network.wan3.defaultroute='0'
network.wan3.macaddr='7c:8b:ca:03:a9:ff'
network.wan3.metric='5'
network.wan3.ipaddr='192.168.12.2'
network.wan3.netmask='255.255.255.0'
network.wan3.gateway='192.168.12.1'
network.wan3.label='CELLULAR 192.168.12.1'
network.wan3.multipath='on'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.proto='dhcp'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.metric='6'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.gateway='fe80::a00:1'
network.omr6in4.ip6addr='fe80::a00:2'
network.omr6in4.metric='7'

Ysurac, 2019-04-09:
I would need the openmptcprouter part of config (without server IP).
I would also need the result of Network->MPTCP->MPTCP Fullmesh
Then give me also "MPTCP Support Check" on wan3 also in Network->MPTCP (if there is a timeout, ```omr-tracebox wan3``` via SSH on OpenMPTCProuter)

FREEEEMANX, 2019-04-09:
[image]

mptcp fullmesh

Index, Address-ID, Backup, IP-address, if-idx
IPv4, next v4-index: 3
2, 3, 0, 192.168.12.2 8
6, 7, 0, 192.168.10.2 6
7, 8, 0, 192.168.11.2 7
IPv6, next v6-index: 0

support check wan3
1: * 
2: * 
3: * 
4: * 
5: * 
6: * 
7: * 
8: * 
9: * 
10: * 
11: * 
12: * 
13: * 
14: * 
15: * 
16: * 
17: * 
18: * 
19: * 
20: * 
21: * 
22: * 
23: * 
24: * 
25: * 
26: * 
27: * 
28: * 
29: * 
30: * 
31: * 
32: * 
33: * 
34: * 
35: * 
36: * 
37: * 
38: * 
39: * 
40: * 
41: * 
42: * 
43: * 
44: * 
45: * 
46: * 
47: * 
48: * 
49: * 
50: * 
51: * 
52: * 
53: * 
54: * 
55: * 
56: * 
57: * 
58: * 
59: * 
60: * 
61: * 
62: * 
63: * 
64: * 

Result:
nil

ssh command
root@OpenMPTCProuter:~# omr-tracebox eth3
tracebox to 130.104.228.140 (multipath-tcp.org): 64 hops max
1: *
2: *
3: *
4: *
5: *
6: *
7: *
8: *
9: *
10: *
11: *
12: *
13: *
14: *
15: *
16: *
17: *
18: *
19: *
20: *
21: *
22: *
23: *
24: *
25: *
26: *
27: *
28: *
29: *
30: *
31: *
32: *
33: *
34: *
35: *
36: *
37: *
38: *
39: *
40: *
41: *
42: *
43: *
44: *
45: *
46: *
47: *
48: *
49: *
50: *
51: *
52: *
53: *
54: *
55: *
56: *
57: *
58: *
59: *
60: *
61: *
62: *
63: *
64: *

Ysurac, 2019-04-10:
OK so no traces.
You always have the "cant' contact server admin script" ?
What I wanted about OpenMPTCProuter config is the "openmptcprouter.*" in "show all settings" tab in "System->OpenMPTCProuter".

FREEEEMANX, 2019-04-10:
There is only  :
   Settings Wizard
    Status
    Advanced Settings

on "System->OpenMPTCProuter" tab. Which one you need ? And is there a better way than screenshots for the settings you need ?

Thanks.

Ysurac, 2019-04-10:
Update to latest release.

FREEEEMANX, 2019-04-10:
All OpenMPTCProuter settings

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_default='0'
dhcp.lan.dhcpv6='disabled'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='AABBCCDDEEFF'
etherwake.@target[0].wakeonboot='off'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].network='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].network='wan' 'wan6' 'wan1' 'wan2' 'wan3'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@forwarding[0]=forwarding
[... 892 lines cut ...]
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth3'

Powered by LuCI Master (git-19.095.24565-19d15f2) / openmptcprouter v0.48 r0+9745-b0395cfc56

FREEEEMANX, 2019-04-10:
Updated to last version, 0.48 by the way.

FREEEEMANX, 2019-04-10:
All OpenMPTCProuter settings

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_default='0'
dhcp.lan.dhcpv6='disabled'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='AABBCCDDEEFF'
etherwake.@target[0].wakeonboot='off'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].network='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].network='wan' 'wan6' 'wan1' 'wan2' 'wan3'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@forwarding[0]=forwarding
[... 892 lines cut ...]
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth3'

Powered by LuCI Master (git-19.095.24565-19d15f2) / openmptcprouter v0.48 r0+9745-b0395cfc56

Ysurac, 2019-04-10:
Seems that VPS admin key was not set, so I don't have all infos about VPS status.

FREEEEMANX, 2019-04-10:
VPS admin key set.

All OpenMPTCProuter settings

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='AABBCCDDEEFF'
etherwake.@target[0].wakeonboot='off'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].network='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].network='wan' 'wan6' 'wan1' 'wan2' 'wan3'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@forwarding[0]=forwarding
[... 925 lines cut ...]
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth3'

Powered by LuCI Master (git-19.095.24565-19d15f2) / openmptcprouter v0.48 r0+9745-b0395cfc56

Ysurac, 2019-04-10:
MPTCP seems to be really blocked on the wan3 connection...
You can disable Shadowsocks and use Glorytun UDP or MLVPN for aggregation.

FREEEEMANX, 2019-04-10:
Could you explain how you check that based on the settings I posted ?
Do you have clues about the why ? Is it ISP side (WAN3 is cellular) , LAN side, or ?

Gonna try as you suggested (glory or mlvpn).

Thanks.

Ysurac, 2019-04-11:
Tracebox can't be used on your connection, the status page say multipath is disabled on the connection so the result of ```omr-mptcp-intf wan3``` via SSH should say that mptcp is not working.
I can't say where is the problem exactly, as tracebox fails since the beginning I would say there is a problem with the router.
Can you try tracebox on wan1 for example ? this should work.

FREEEEMANX, 2019-04-19:
root@OpenMPTCProuter:~# omr-mptcp-intf eth3
Nay, Nay, Nay, your have an old computer that does not speak MPTCP. Shame on you!
root@OpenMPTCProuter:~# omr-mptcp-intf eth2
Yay, you are MPTCP-capable! You can now rest in peace.
root@OpenMPTCProuter:~# omr-mptcp-intf eth1
Yay, you are MPTCP-capable! You can now rest in peace.

FREEEEMANX, 2019-04-19:
Eth3 is where the cellular connection is plugged (TP LINK MR 200).

FREEEEMANX, 2019-04-23:
?

Ysurac, 2019-04-23:
I can't say anything more. For unknown reason eth3 doesn't support MPTCP.
You can try to disable shadowsocks and use Glorytun UDP, or use eth3 only as backup.

closed 2019-05-25 by Ysurac