Ysurac/openmptcprouter #426: ip doorbell out

bosseb, 2019-04-10:
OpenMPTCProuter version: 0.48
OpenMPTCProuter VPS version: 0.992
OpenMPTCProuter platform: x86_64

Hello
since i have install OPENMPTCProuter, my ip doorbell doesn't work at all

if you could help me

Ysurac, 2019-04-10:
Model ? By ethernet or wifi ? DHCP or static IP ? Locally or via internet ?

bosseb, 2019-04-10:
this model : https://home-smart.fr/
DHCP, wifi, locally and via internet

Ysurac, 2019-04-10:
Check in system log if you see DHCP request from the doorbell.

bosseb, 2019-04-10:
i have always looking for, but i don't know the doorbell ip

Ysurac, 2019-04-10:
The doorbell should give a name when requesting an IP (else check the mac address of the doorbell if you have it somewhere).
You can also check in Network->DHCP & DNS if you see the doorbell.

If not check wifi settings and that DHCP server is enabled on wifi interface (if wifi is served by OpenMPTCProuter).

bosseb, 2019-04-10:
the door bell is in DHCP 192.168.0.149 is name is DEFAULT

Ysurac, 2019-04-10:
So at least it get an IP.
Maybe some ports need to be open ? I can't help you, the documentation is not available online.

bosseb, 2019-04-10:
no i think it send video on a specific server in the cloud, like new ipcam, but here i think the system block the video connection

Ysurac, 2019-04-10:
The server in the cloud can ban IP from VPS.

Ysurac, 2019-04-10:
You can use OMR-Bypass.

bosseb, 2019-04-10:
yes but how to find the server adress ??

Ysurac, 2019-04-10:
You can bypass everything that come from MAC or IP of the doorbell.

bosseb, 2019-04-10:
I have bypass his MAC address but it doesn't work

bosseb, 2019-04-10:
i have reboot the NUC and it works ! thank you so much

closed 2019-04-11 by Ysurac