Ysurac/openmptcprouter #427: upgrade problem

bojanlisa, 2019-04-11:
## Expected Behavior

You have OpenMPTCProuter version 0.47.6 and version 0.48 is now available

Model Raspberry Pi 3 Model B Plus Rev 1.3
Firmware Version openmptcprouter v0.47.6 r0+9530-64bb88841f / LuCI Master (git-19.082.80236-26b738a) 

Upgrade Firmware by uploading the img: (extracted)

squashfs image: openmptcprouter-v0.48-r09745-b0395cfc56-brcm2708-bcm2710-rpi-3-squashfs-sysupgrade.img.gz (2019/04/03 - 38.97Mo - sha256sum)

## Actual Behavior

The uploaded image file does not contain a supported format. Make sure that you choose the generic image format for your platform. 

## Specifications

  - OpenMPTCProuter version: 0.47.6
  - OpenMPTCProuter VPS version: 0.95
  - OpenMPTCProuter platform: (RPI3)

Worked fine for:
openmptcprouter-v0.46.3-r09228-a3ccac6b1d-brcm2708-bcm2710-rpi-3-squashfs-sysupgrade.img

to

openmptcprouter-v0.47.6-r09530-64bb88841f-brcm2708-bcm2710-rpi-3-squashfs-sysupgrade.img

Ysurac, 2019-04-11:
You have to keep the image in gz format.

umekzg, 2019-04-11:
And you are right. It works now :)

closed 2019-04-11 by Ysurac