Ysurac/openmptcprouter #428: wlan traffic MPTCP ALL graph missing

umekzg, 2019-04-11:
MPTCP - Realtime Traffic - all graph
/cgi-bin/luci/admin/network/mptcp/bandwidth?dev=all
wlan0 interface is not shown in the ALL graph traffic, only ethernet & vlans

when i select only wlan the graph shows traffic
/cgi-bin/luci/admin/network/mptcp/bandwidth?dev=wlan0

wan0 is configured with mptcp and works normaly

## Specifications

  - OpenMPTCProuter version: 0.47.6
  - OpenMPTCProuter VPS version: 0.95
  - OpenMPTCProuter platform: (RPI3)

Ysurac, 2019-04-11:
I would need a screenshot of ```/cgi-bin/luci/admin/network/mptcp/bandwidth``` page

umekzg, 2019-04-11:
[image]
Here you go

user747, 2019-04-12:
I think maybe you want this graph instead? 
cgi-bin/luci/admin/network/mptcp/bandwidth

closed 2019-07-04 by Ysurac