Ysurac/openmptcprouter #429: Add traffic shaping with DSCP on both side

Ysurac, 2019-04-11:
Todo:
* Create an interface on OpenMPTCProuter to set DSCP
* Add domains with defined DSCP via dnsmasq ipset
* Set download side rules on VPS via shorewall if possible and omr-admin script

jwthecolorist, 2019-04-21:
Hello Ysurac! I really appreciate your hard work on this project, overall it's working extremely well combining connections provided by ADSL, 3G, and 4G LTE modems. 

Regarding DSCP, would it be possible to have a section on OMR-Bypass to where I can have traffic flagged with a certain DSCP flag (say 48 etc) sent out a specific WAN? You pretty well have all circumstances covered with OMR-Bypass presently and it's working well, that's just one thing I use constantly on my network to route traffic from specific applications.

Only if you can, no worries if you can't. Thanks!

jwthecolorist, 2019-05-07:
Just curious: Do you think the DSCP OMR-Bypass redirection will be something possible to add? If not, no worries.

xabix, 2019-05-07:
I think this would be useful. What I mean is that I also have a mix of Wan between ADSL/4G and the overall jitter &Bufferbloat are very bad which impacts my Skype/VoIP traffic. Therefore I am looking for the best way to combine the 2 ADSL for Skype as 1st priority and then the 2 ADSL and 2 4G for the rest.
Therefore I was wondering the best way to do this and maybe DSCP tagging could help?

Ysurac, 2019-05-07:
Not possible for now. Maybe in a next version using Netlink API...

jwthecolorist, 2019-05-07:
Ah, I see. No problem.

tentious, 2019-05-30:
@Ysurac... First, thanks for your tremendous work on this, this is the only solution that's worked for me - 2x 4G/LTE connections. Aggregation and nearly all other features are working near flawlessly. Thanks again!
Now, this is really the next piece of the puzzle I'm missing. I also have Skype/MS Teams/VoIP and video conferencing that I use for several hours of the day. If there is absolutely no traffic, there's no problem, however if the weakest link is at capacity then jitter really kicks in and call quality is degraded. One solution is DSCP, but in the meantime, any other suggestion to better control/prioritize ports/dest. IP/etc.? I've looked at SQM-QoS and OMR-DSCP, but doesn't appear the tagging makes it across and/or is not handled through both ends.

jwthecolorist, 2019-05-31:
@tentious Yeah same issue here. For some reason, OMR-DSCP+SQM-QoS isn't making a difference for me either BUT I may not have it configured properly. Have to investigate further.

closed 2019-12-20 by Ysurac