Ysurac/openmptcprouter #430: Gateway is up but show up as gateway down on status page

dnwk, 2019-04-13:
The gateway is a Sierra Wireless em7511 LTE module connect to USB. It is operating as QMI cellular. I could confirm the gateway is on. I disconnected all other WAN and I could still access Internet. I could ping out by specifying that interface. However, it will always show up as gateway down on status page. How do I mitigate this problem?

## Specifications

  - OpenMPTCProuter version: (Last is not a version) 0.48
  - OpenMPTCProuter VPS version: (Last is not a version) 0.992 
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)  Linksys WRT32X

Ysurac, 2019-04-13:
Gateway down is when I can't ping the gateway. If it's really the case then it's only a warning, no problem.

dnwk, 2019-04-13:
It's QMI. So, probably no gateway to ping. Connection is definitely up.

closed 2019-05-25 by Ysurac