Ysurac/openmptcprouter #433: When the direct output is active, the default route change takes about one minute

manueldev, 2019-04-14:
## Expected Behavior

When the direct output is active, the default route change should take a few seconds.

## Actual Behavior

When the direct output is active, the default route change takes about one minute to go from wan1 (eth0) to wan3 (usb0). Going from wan3 to wan1 only takes a couple of seconds.

## Steps to Reproduce the Problem

  1. Turn off the VPS
  2. Wait for the direct connection to activate
  3. Disconnect the wan for which the direct connection is being made
  4. Wait for the direct connection to be activated by another wan

## Specifications

  - OpenMPTCProuter version: v0.48 r0+9745-b0395cfc56
  - OpenMPTCProuter VPS version: 0.992
  - OpenMPTCProuter platform: RPI3

Is this behavior normal?

[image]

[image]

https://youtu.be/w0HC_0VqMLM

Ysurac, 2019-04-14:
All settings are in omr-tracker interface, so this can be modified.

manueldev, 2019-04-14:
this? 
[image]

closed 2019-05-25 by Ysurac