Ysurac/openmptcprouter #434: Probleme instabilité

silver9195, 2019-04-15:
## Specifications

  - OpenMPTCProuter version: 0.48
  - OpenMPTCProuter VPS version: 0.991
  - OpenMPTCProuter platform: RPI3

Bonjour Ysurac,

Déjà un GRAND bravo pour la solution.

Je cherchais un moyen de prendre la main sur ma solution domotique et NAS etc à distance à travers un lien 4G (sans ip publique donc) et ta solution le permet sans problème.

Cependant, je rencontre quelques problèmes, par moments le routeur Openmptcp semble ne plus fonctionner et l’accès à extérieure est impossible. 

cela arrive assez souvent , plusieurs fois par jours.

J'ai instauré un reboot toutes les nuits mais cela ne suffit pas.

J'explique quand même mon architecture  à la maison :

un accès SFR connecté en bridge sur mon routeur Synology 1900 (qui s'occupe du lan et du wifi dans la maison) et un accès 4G grâce à une antenne extérieur .

La clés 4G est connectée en direct sur le routeur Openmptcp .

Et je gère les équipements que je veux faire passer par le routeur Openmptcp en changeant la GW en 192.168.100.1

Mais par moments je n'ai plus Internet en passant par le Openmptcp , sans raison accaparantes les deux liens sont Ok tout les voyants dans états sont au vert.

Dis moi de quoi as tu besoin pour aider, je sais que mon problème est un peu vague 

Merci beaucoup.

(La 4G est KO car mes 30Go de data sont off)

[image]

Et voici mes logs :+1: 

Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681677] pinctrl-bcm2835 3f200000.gpio: read 0003ffc0 (36 => gpio_in)
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681681] pinctrl-bcm2835 3f200000.gpio: write 001fffc0 (36 <= alt3)
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681688] pinctrl-bcm2835 3f200000.gpio: request pin 37 (gpio37) for 3f300000.mmc
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681693] pinctrl-bcm2835 3f200000.gpio: read 001fffc0 (37 => gpio_in)
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681697] pinctrl-bcm2835 3f200000.gpio: write 00ffffc0 (37 <= alt3)
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681704] pinctrl-bcm2835 3f200000.gpio: request pin 38 (gpio38) for 3f300000.mmc
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681708] pinctrl-bcm2835 3f200000.gpio: read 00ffffc0 (38 => gpio_in)
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681736] pinctrl-bcm2835 3f200000.gpio: write 07ffffc0 (38 <= alt3)
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681743] pinctrl-bcm2835 3f200000.gpio: request pin 39 (gpio39) for 3f300000.mmc
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681755] pinctrl-bcm2835 3f200000.gpio: read 07ffffc0 (39 => gpio_in)
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681775] pinctrl-bcm2835 3f200000.gpio: write 3fffffc0 (39 <= alt3)
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681875] mmc-bcm2835 3f300000.mmc: GPIO lookup for consumer wp
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681879] mmc-bcm2835 3f300000.mmc: using device tree for GPIO lookup
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681885] of_get_named_gpiod_flags: can't parse 'wp-gpios' property of node '/soc/mmc@7e300000[0]'
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681890] of_get_named_gpiod_flags: can't parse 'wp-gpio' property of node '/soc/mmc@7e300000[0]'
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681894] mmc-bcm2835 3f300000.mmc: using lookup tables for GPIO lookup
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.681899] mmc-bcm2835 3f300000.mmc: lookup for GPIO wp failed
Sun Apr 14 20:40:26 2019 kern.info kernel: [    1.681963] mmc-bcm2835 3f300000.mmc: mmc_debug:0 mmc_debug2:0
Sun Apr 14 20:40:26 2019 kern.info kernel: [    1.689080] mmc-bcm2835 3f300000.mmc: DMA channel allocated
Sun Apr 14 20:40:26 2019 kern.notice kernel: [    1.704919] random: fast init done
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.751280] pinctrl core: add 6 pinctrl maps
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.751319] pinctrl-bcm2835 3f200000.gpio: found group selector 48 for gpio48
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.751339] pinctrl-bcm2835 3f200000.gpio: found group selector 49 for gpio49
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.751357] pinctrl-bcm2835 3f200000.gpio: found group selector 50 for gpio50
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.751382] pinctrl-bcm2835 3f200000.gpio: found group selector 51 for gpio51
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.751400] pinctrl-bcm2835 3f200000.gpio: found group selector 52 for gpio52
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.751426] pinctrl-bcm2835 3f200000.gpio: found group selector 53 for gpio53
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.751438] pinctrl-bcm2835 3f200000.gpio: request pin 48 (gpio48) for 3f202000.mmc
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.751449] pinctrl-bcm2835 3f200000.gpio: read 00200924 (48 => gpio_in)
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.751458] pinctrl-bcm2835 3f200000.gpio: write 04200924 (48 <= alt0)
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.751467] pinctrl-bcm2835 3f200000.gpio: request pin 49 (gpio49) for 3f202000.mmc
Sun Apr 14 20:40:26 2019 kern.debug kernel: [    1.751476] pinctrl-bcm2835 3f200000.gpio: read 04200924 (49 => gpio_in)
[... 680 lines cut ...]
Mon Apr 15 06:50:46 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Mon Apr 15 06:51:39 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Mon Apr 15 06:54:22 2019 daemon.err uhttpd[2702]: luci: accepted login on /admin/system/openmptcprouter for root from 81.255.44.144
Mon Apr 15 07:02:15 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Mon Apr 15 07:02:15 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Mon Apr 15 07:02:22 2019 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Mon Apr 15 07:02:27 2019 user.notice post-tracking-post-tracking: eth1 switched off because ping from 192.168.8.100 error (4.2.2.1,8.8.8.8,8.8.4.4,9.9.9.9)
Mon Apr 15 07:02:27 2019 user.notice post-tracking-post-tracking: Delete vps (51.83.76.13) route via 192.168.8.1 dev eth1
Mon Apr 15 07:02:44 2019 daemon.info glorytun[3614]: STOPPED tun0
Mon Apr 15 07:02:46 2019 daemon.err glorytun[3614]: 51.83.76.13.65001: connected
Mon Apr 15 07:02:56 2019 daemon.err glorytun[3614]: 51.83.76.13.65001: key exchange failed
Mon Apr 15 07:02:56 2019 daemon.info glorytun[3614]: STOPPED tun0
Mon Apr 15 07:02:57 2019 daemon.err glorytun[3614]: 51.83.76.13.65001: connected
Mon Apr 15 07:02:57 2019 daemon.info glorytun[3614]: STARTED tun0
Mon Apr 15 07:03:08 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Mon Apr 15 07:03:11 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Mon Apr 15 07:03:14 2019 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Mon Apr 15 07:03:21 2019 user.notice post-tracking-post-tracking: eth1 switched off because ping from 192.168.8.100 error (4.2.2.1,8.8.8.8,8.8.4.4,9.9.9.9)
Mon Apr 15 07:03:21 2019 user.notice post-tracking-post-tracking: Delete vps (51.83.76.13) route via 192.168.8.1 dev eth1
Mon Apr 15 07:10:47 2019 daemon.info dnsmasq[5495]: read /etc/hosts - 4 addresses
Mon Apr 15 07:10:47 2019 daemon.info dnsmasq[5495]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Mon Apr 15 07:40:54 2019 daemon.notice netifd: omrvpn (3725): udhcpc: sending renew to 10.255.255.1
Mon Apr 15 07:40:54 2019 daemon.notice netifd: omrvpn (3725): udhcpc: bind(UDP): Address not available
Mon Apr 15 07:40:54 2019 daemon.notice netifd: omrvpn (3725): udhcpc: sending renew to 0.0.0.0
Mon Apr 15 07:40:56 2019 daemon.notice netifd: omrvpn (3725): udhcpc: lease of 10.255.255.28 obtained, lease time 7200

Ysurac, 2019-04-15:
Il me faudrait que les logs au moment où ça fonctionne pas.
Sinon je te conseillerais de forcer le master sur la connection SFR et de mettre "Master interface selection" sur "pas de changements" dans Paramètres avancés de Systèmes->OpenMPTCProuter.

silver9195, 2019-04-15:
Ok j'ai modifier l'heure pour les logs elle n'était pas bonne.

C'est fait pour le master :

[image]
 

Merci pour ta réactivité !

Je prend les logs des qu'il y a un soucis

silver9195, 2019-04-16:
Bonjour Ysurac, voici les logs au moments ou j'ai eu pas mal de soucis : 

Mon Apr 15 15:41:07 2019 daemon.notice netifd: omrvpn (3725): udhcpc: lease of 10.255.255.28 obtained, lease time 7200
Mon Apr 15 15:59:38 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:00:00 2019 cron.info crond[2064]: USER root pid 20347 cmd /etc/init.d/vnstat_backup backup
Mon Apr 15 16:00:00 2019 user.info vnstat_backup: backing up database
Mon Apr 15 16:00:15 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:00:35 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:01:05 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:01:39 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:01:59 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:02:39 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:05:21 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:05:52 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:06:35 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:07:05 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:07:26 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:07:45 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:08:15 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:10:25 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:11:04 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:11:36 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:12:09 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:13:30 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:14:10 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Mon Apr 15 16:15:11 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:15:14 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Mon Apr 15 16:16:46 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:17:05 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:17:26 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:17:57 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:20:19 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:22:27 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:23:17 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:25:01 2019 daemon.err /usr/bin/ss-redir[24706]: remote recv: Operation timed out
Mon Apr 15 16:32:28 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:34:16 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:41:07 2019 daemon.notice netifd: omrvpn (3725): udhcpc: sending renew to 10.255.255.1
Mon Apr 15 16:41:07 2019 daemon.notice netifd: omrvpn (3725): udhcpc: bind(UDP): Address not available
Mon Apr 15 16:41:07 2019 daemon.notice netifd: omrvpn (3725): udhcpc: sending renew to 0.0.0.0
Mon Apr 15 16:41:07 2019 daemon.notice netifd: omrvpn (3725): udhcpc: lease of 10.255.255.28 obtained, lease time 7200
Mon Apr 15 16:44:37 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:45:19 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:45:50 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:48:24 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:48:39 2019 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact 103.224.182.242, 198.27.92.1, 88.191.250.176, 151.101.129.164)
Mon Apr 15 16:48:39 2019 user.notice Shadowsocks: Rules DOWN
Mon Apr 15 16:48:44 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:48:55 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:49:00 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Mon Apr 15 16:49:00 2019 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Mon Apr 15 16:49:00 2019 user.notice Shadowsocks: Rules UP
Mon Apr 15 16:49:00 2019 user.notice Shadowsocks: Reload omr-bypass rules
Mon Apr 15 16:49:00 2019 user.notice omr-bypass: Starting OMR-ByPass...
Mon Apr 15 16:49:01 2019 daemon.info dnsmasq[5495]: read /etc/hosts - 4 addresses
Mon Apr 15 16:49:01 2019 daemon.info dnsmasq[5495]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Mon Apr 15 16:49:01 2019 user.notice omr-bypass: OMR-ByPass is running
Mon Apr 15 16:49:27 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
Mon Apr 15 16:50:04 2019 daemon.err /usr/bin/ss-local[24704]: getpeername: Socket not connected
[... cut ...]
Mon Apr 15 20:49:55 2019 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Mon Apr 15 20:49:55 2019 user.notice Shadowsocks: Rules UP
Mon Apr 15 20:49:55 2019 user.notice Shadowsocks: Reload omr-bypass rules
Mon Apr 15 20:49:55 2019 user.notice omr-bypass: Starting OMR-ByPass...
Mon Apr 15 20:49:56 2019 daemon.info dnsmasq[5495]: read /etc/hosts - 4 addresses
Mon Apr 15 20:49:56 2019 daemon.info dnsmasq[5495]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Mon Apr 15 20:49:56 2019 user.notice omr-bypass: OMR-ByPass is running

Ysurac, 2019-04-16:
Il n'arrive pas à tenir la connexion avec le VPS, mais surtout la clef USB s'en va et revient, si tu l'as pas débranché à ce moment là ça ressemble beaucoup à un soucis d'alimentation. Les RPI sont très sensibles niveau USB et la clef consomme peut être un peu trop.

silver9195, 2019-04-16:
Merci de ta rapidité encore !
Je l'ai déconnectée une fois hier pour remettre la carte sim.
Je vais regarder la partie alimentation.
J'ai remarqué aussi qu'il a du mal à garder la connexion avec le VPS tu sais à quoi cela pourrait être due ?
J'ai pas de problème avec l'adsl , la connexion est stable

Ysurac, 2019-04-16:
Le master est bien sur la connexion ADSL ?
Il faudra regarder sur le VPS le log dans /var/log/daemon.log pour voir si il y a un soucis à ce niveau la.

silver9195, 2019-04-16:
J'ai défini le mutipath en "master" sur l'adsl et en "on" sur la 4g c'est cela ?

Voila les logs sur le VPS aux même heures :

Apr 15 14:26:29 vps667853 ss-server[505]:  2019-04-15 14:26:29 ERROR: remote recv: Connection reset by peer
Apr 15 14:28:07 vps667853 ss-server[505]:  2019-04-15 14:28:07 ERROR: remote recv: Connection reset by peer
Apr 15 14:29:40 vps667853 omr-service[23177]: /usr/local/bin/omr-service: line 41: echo: write error: Broken pipe
Apr 15 14:33:23 vps667853 ss-server[505]:  2019-04-15 14:33:23 ERROR: remote recv: Connection reset by peer
Apr 15 14:33:38 vps667853 ss-server[505]:  2019-04-15 14:33:38 ERROR: remote recv: Connection reset by peer
Apr 15 14:38:48 vps667853 ss-server[505]:  2019-04-15 14:38:48 ERROR: getpeername: Transport endpoint is not connected
Apr 15 14:48:29 vps667853 ss-server[505]:  2019-04-15 14:48:29 ERROR: remote recv: Connection reset by peer
Apr 15 15:56:53 vps667853 ss-server[505]:  2019-04-15 15:56:53 ERROR: server recv: Connection reset by peer
Apr 15 15:57:23 vps667853 ss-server[505]:  2019-04-15 15:57:23 ERROR: server recv: Connection reset by peer
Apr 15 16:04:47 vps667853 ss-server[505]:  2019-04-15 16:04:47 ERROR: server recv: Connection reset by peer
Apr 15 16:10:40 vps667853 ss-server[505]:  2019-04-15 16:10:40 ERROR: server recv: Connection reset by peer
Apr 15 16:12:25 vps667853 ss-server[505]:  2019-04-15 16:12:25 ERROR: server recv: Connection reset by peer
Apr 15 16:18:16 vps667853 ss-server[505]:  2019-04-15 16:18:16 ERROR: server recv: Connection reset by peer
Apr 15 16:22:43 vps667853 ss-server[505]:  2019-04-15 16:22:43 ERROR: server recv: Connection reset by peer
Apr 15 16:32:44 vps667853 ss-server[505]:  2019-04-15 16:32:44 ERROR: server recv: Connection reset by peer
Apr 15 16:36:15 vps667853 ss-server[505]:  2019-04-15 16:36:15 ERROR: server recv: Connection reset by peer
Apr 15 16:40:07 vps667853 ss-server[505]:  2019-04-15 16:40:07 ERROR: server recv: Connection reset by peer
Apr 15 16:46:29 vps667853 ss-server[505]:  2019-04-15 16:46:29 ERROR: server recv: Connection reset by peer
Apr 15 16:54:39 vps667853 ss-server[505]:  2019-04-15 16:54:39 ERROR: getpeername: Transport endpoint is not connected
Apr 15 16:56:58 vps667853 ss-server[505]:  2019-04-15 16:56:58 ERROR: remote recv: Connection reset by peer
Apr 15 17:06:46 vps667853 ss-server[505]:  2019-04-15 17:06:46 ERROR: server recv: Connection reset by peer
Apr 15 17:07:26 vps667853 ss-server[505]:  2019-04-15 17:07:26 ERROR: server recv: Connection reset by peer
Apr 15 17:13:13 vps667853 glorytun-tcp-run[23144]: read: Connection timed out
Apr 15 17:13:21 vps667853 glorytun-tcp-run[23144]: 109.17.118.238.46747: connected
Apr 15 17:18:33 vps667853 ss-server[505]:  2019-04-15 17:18:33 ERROR: server recv: Connection reset by peer
Apr 15 17:18:35 vps667853 ss-server[505]:  2019-04-15 17:18:35 ERROR: server recv: Connection reset by peer
Apr 15 17:18:35 vps667853 ss-server[505]:  2019-04-15 17:18:35 ERROR: server recv: Connection reset by peer
Apr 15 17:18:40 vps667853 ss-server[505]:  2019-04-15 17:18:40 ERROR: getpeername: Transport endpoint is not connected
Apr 15 17:22:22 vps667853 ss-server[505]:  2019-04-15 17:22:22 ERROR: server recv: Connection reset by peer
Apr 15 17:31:19 vps667853 ss-server[505]:  2019-04-15 17:31:19 ERROR: server recv: Connection reset by peer
Apr 15 17:34:35 vps667853 ss-server[505]:  2019-04-15 17:34:35 ERROR: server recv: Connection reset by peer
Apr 15 17:34:40 vps667853 ss-server[505]:  2019-04-15 17:34:40 ERROR: server recv: Connection reset by peer
Apr 15 17:38:36 vps667853 ss-server[505]:  2019-04-15 17:38:36 ERROR: server recv: Connection reset by peer
Apr 15 18:17:30 vps667853 dhclient[422]: DHCPREQUEST of 51.83.76.13 on ens3 to 51.83.72.1 port 67
Apr 15 18:17:30 vps667853 dhclient[422]: DHCPACK of 51.83.76.13 from 51.83.72.1
Apr 15 18:17:30 vps667853 dhclient[422]: bound to 51.83.76.13 -- renewal in 39073 seconds.
Apr 15 18:30:13 vps667853 ss-server[505]:  2019-04-15 18:30:13 ERROR: getpeername: Transport endpoint is not connected
Apr 15 18:30:21 vps667853 ss-server[505]:  2019-04-15 18:30:21 ERROR: getpeername: Transport endpoint is not connected
Apr 15 19:30:06 vps667853 ss-server[505]:  2019-04-15 19:30:06 ERROR: server recv: Connection reset by peer
Apr 15 20:07:41 vps667853 omr-service[23177]: /usr/local/bin/omr-service: line 41: echo: write error: Broken pipe
Apr 15 20:45:07 vps667853 ss-server[505]:  2019-04-15 20:45:07 ERROR: remote recv: Connection reset by peer
Apr 15 20:53:28 vps667853 ss-server[505]:  2019-04-15 20:53:28 ERROR: server recv: Connection reset by peer

Ysurac, 2019-04-16:
oui c'est bien ça. Et dans "Systéme->OpenMPTCProuter" puis paramètres avancés tu peux mettre "Master interface selection" à "Pas de changement" pour éviter qu'il soit modifié.
Sinon les logs confirment un soucis de liaison entre le VPS et le routeur.

silver9195, 2019-04-17:
C'est déjà fait pour le master.
Une idée de la raison de la perte de liaison ?

Ysurac, 2019-04-17:
Il faudrait essayer avec la clef 4G de débrancher et voir si le soucis est toujours présent ou pas.

silver9195, 2019-04-19:
Salut Ysurac, désolé du retard je n'ai pas vu ton message.

Alors hier j'ai fait un netoyage dans mon reseau local et j'avais certainement un soucis car mes temps de reponse en lan etaient mauvais.

Maintenant cela à l'air mieux pour l'instant.

Juste un soucis le VPN ne veut pas monter maintenant :

[image]

Voici les logs au cas ou : 

Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.124277] libphy: Fixed MDIO Bus: probed
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.126714] usbcore: registered new interface driver lan78xx
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.129141] usbcore: registered new interface driver smsc95xx
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.131498] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.161661] dwc_otg 3f980000.usb: base=0xf0980000
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.364047] Core Release: 2.80a
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.366242] Setting default values for core params
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.368445] Finished setting default values for core params
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.570814] Using Buffer DMA mode
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.572949] Periodic Transfer Interrupt Enhancement - disabled
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.575204] Multiprocessor Interrupt Enhancement - disabled
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.577511] OTG VER PARAM: 0, OTG VER FLAG: 0
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.579798] Dedicated Tx FIFOs mode
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.582448] WARN::dwc_otg_hcd_init:1046: FIQ DMA bounce buffers: virt = 0xba504000 dma = 0xfa504000 len=9024
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.587202] FIQ FSM acceleration enabled for :
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.587202] Non-periodic Split Transactions
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.587202] Periodic Split Transactions
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.587202] High-Speed Isochronous Endpoints
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.587202] Interrupt/Control Split Transaction hack enabled
Fri Apr 19 08:30:31 2019 kern.debug kernel: [    0.598282] dwc_otg: Microframe scheduler enabled
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.598335] WARN::hcd_init_fiq:459: FIQ on core 1 at 0x8043f74c
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.600652] WARN::hcd_init_fiq:460: FIQ ASM at 0x8043fa68 length 36
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.602937] WARN::hcd_init_fiq:486: MPHI regs_base at 0xf0006000
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.605303] dwc_otg 3f980000.usb: DWC OTG Controller
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.607640] dwc_otg 3f980000.usb: new USB bus registered, assigned bus number 1
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.610041] dwc_otg 3f980000.usb: irq 62, io mem 0x00000000
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.612426] Init: Port Power? op_state=1
Fri Apr 19 08:30:31 2019 kern.warn kernel: [    0.614738] Init: Power Port (0)
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.617156] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.619526] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.621899] usb usb1: Product: DWC OTG Controller
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.624203] usb usb1: Manufacturer: Linux 4.14.108 dwc_otg_hcd
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.626523] usb usb1: SerialNumber: 3f980000.usb
Fri Apr 19 08:30:31 2019 kern.debug kernel: [    0.629304] hub 1-0:1.0: no of_node; not parsing pinctrl DT
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.629329] hub 1-0:1.0: USB hub found
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.631590] hub 1-0:1.0: 1 port detected
Fri Apr 19 08:30:31 2019 kern.debug kernel: [    0.634146] dwc_otg: FIQ enabled
Fri Apr 19 08:30:31 2019 kern.debug kernel: [    0.634150] dwc_otg: NAK holdoff enabled
Fri Apr 19 08:30:31 2019 kern.debug kernel: [    0.634154] dwc_otg: FIQ split-transaction FSM enabled
Fri Apr 19 08:30:31 2019 kern.debug kernel: [    0.634162] Module dwc_common_port init
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.634304] usbcore: registered new interface driver uas
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.636499] usbcore: registered new interface driver usb-storage
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.638730] mousedev: PS/2 mouse device common for all mice
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.641442] bcm2835-wdt 3f100000.watchdog: Broadcom BCM2835 watchdog timer
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.643936] bcm2835-cpufreq: min=600000 max=1200000
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.646508] sdhci: Secure Digital Host Controller Interface driver
Fri Apr 19 08:30:31 2019 kern.info kernel: [    0.648771] sdhci: Copyright(c) Pierre Ossman
Fri Apr 19 08:30:31 2019 kern.debug kernel: [    0.651126] pinctrl core: add 12 pinctrl maps
Fri Apr 19 08:30:31 2019 kern.debug kernel: [    0.651157] pinctrl-bcm2835 3f200000.gpio: found group selector 34 for gpio34
Fri Apr 19 08:30:31 2019 kern.debug kernel: [    0.651184] pinctrl-bcm2835 3f200000.gpio: found group selector 35 for gpio35
[... cut ...]
Fri Apr 19 08:47:51 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Fri Apr 19 09:03:45 2019 daemon.err /usr/bin/ss-redir[5117]: remote recv: Operation timed out
Fri Apr 19 09:04:06 2019 daemon.err /usr/bin/ss-redir[5117]: remote recv: Operation timed out
Fri Apr 19 09:04:21 2019 daemon.err /usr/bin/ss-redir[5116]: remote recv: Operation timed out
Fri Apr 19 09:04:25 2019 daemon.err /usr/bin/ss-redir[5115]: remote recv: Operation timed out
Fri Apr 19 09:04:25 2019 daemon.err /usr/bin/ss-redir[5117]: remote recv: Operation timed out

closed 2019-05-25 by Ysurac