Ysurac/openmptcprouter #435: shadowsocks.conf

user747, 2019-04-15:
VPS 
[fs.file-max = 512000](https://github.com/Ysurac/openmptcprouter-vps/blob/cee6434234cded0d8537b9f4061e142a05a9246a/shadowsocks.conf#L3)

OMR
[fs.file-max = 512000](https://github.com/Ysurac/openmptcprouter-feeds/blob/708a245e5f920959d3666ddd78e42712e57de8a5/shadowsocks-libev/files/shadowsocks.conf#L3)

Should shadowsocks-libev.init also mirror this value?  
[procd_set_param limits nofile="51200 51200"](https://github.com/Ysurac/openmptcprouter-feeds/blob/c24e7ed3d6a882c636564633030763a972ee5626/shadowsocks-libev/files/shadowsocks-libev.init#L227)

Ysurac, 2019-04-15:
True, this will be modified in next release.

user747, 2019-04-16:
https://github.com/Ysurac/openmptcprouter-feeds/commit/8d0e8f82e6c6ca13efb1040d48e3158140dc31fa

closed 2019-04-16 by user747