Ysurac/openmptcprouter #437: help build serdisplib-2.01

kapook007, 2019-04-17:
Hello
i can not build serdisplib-2.01 

git clone https://github.com/Ysurac/openmptcprouter.git
cd openmptcprouter
OMR_TARGET="bpi-r2" ./build.sh

This is my error 
../include/serdisplib/serdisp_connect_usb.h:35:10: fatal error: usb.h: No such file or directory
 #include <usb.h>
          ^~~~~~~
compilation terminated.

my system is debain 9.8  64bit with
apt-get install git-core build-essential libssl-dev libncurses5-dev unzip gawk zlib1g-dev busybox

How to  build serdisplib-2.01 ?

Build bpi-r2 , rpi3 , wrt32x  is same problem !!

Ysurac, 2019-04-17:
It's why it's not more build in current master for now.

kapook007, 2019-04-18:
Now working 
Thank you

closed 2019-04-18 by Ysurac