Ysurac/openmptcprouter #438: TCP traffic works, not UDP

olaulau, 2019-04-17:
## Expected Behavior
being able to communicate through UDP-based protocols (such as DNS)

## Actual Behavior
only TCP works (such as HTTP)

## Steps to Reproduce the Problem
curl ipconfig.io : works
nslookup free.fr 1.1.1.1 : doesn't work

I have been able to make UDP work by unsetting ss-redir for UDP.

## Specifications
  - OpenMPTCProuter version: v0.48
  - OpenMPTCProuter VPS version: 0.992
  - OpenMPTCProuter platform: RPI3+

Ysurac, 2019-04-17:
Shadowsocks is used for UDP only when VPN is down. By default OpenMPTCProuter use Glorytun TCP. Do you make any change here ? If not, what do you have in system log ?

olaulau, 2019-04-17:
Overview » Instance "vpn"
enable
X

port
65001
 
dev
tun0
 
host
46.xx.Xx.xx
 
localip
10.255.255.2
 
Local tunnel ip address
remoteip
10.255.255.1
 
Remote tunnel ip address
key
40xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 
proto
 TCP

listener
 O

mtuauto
 X

mptcp
X
 
chacha20
X

olaulau, 2019-04-17:
Wed Apr 17 17:29:17 2019 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/glorytun-udp
Wed Apr 17 17:29:17 2019 user.notice ucitrack: Setting up /etc/config/upnpd reload trigger for non-procd /etc/init.d/miniupnpd
Wed Apr 17 17:29:17 2019 user.notice ucitrack: Setting up /etc/config/shadowsocks-libev reload dependency on /etc/config/omr-tracker
Wed Apr 17 17:29:17 2019 user.notice firewall: Reloading firewall due to ifup of wan1 (wan1)
Wed Apr 17 17:29:17 2019 user.notice ucitrack: Setting up /etc/config/vnstat reload trigger for non-procd /etc/init.d/vnstat
Wed Apr 17 17:29:17 2019 user.notice mptcp: Reloading mptcp config due to ifup of wan1 (wan1)
Wed Apr 17 17:29:18 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Apr 17 17:29:18 2019 daemon.info glorytun[3092]: INITIALIZED tun0
Wed Apr 17 17:29:18 2019 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Apr 17 17:29:18 2019 daemon.notice netifd: Network device 'tun0' link is up
Wed Apr 17 17:29:18 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Apr 17 17:29:18 2019 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Apr 17 17:29:18 2019 daemon.notice netifd: omrvpn (3148): udhcpc: started, v1.30.1
Wed Apr 17 17:29:18 2019 daemon.notice netifd: omrvpn (3148): udhcpc: sending discover
Wed Apr 17 17:29:18 2019 user.notice mptcp: Reloading mptcp config due to ifup of loopback (lo)
Wed Apr 17 17:29:19 2019 user.notice firewall: Reloading firewall due to ifup of omr6in4 (6in4-omr6in4)
Wed Apr 17 17:29:19 2019 daemon.err glorytun[3092]: 46.105.34.59.65001: connected
Wed Apr 17 17:29:19 2019 daemon.info glorytun[3092]: STARTED tun0

Wed Apr 17 17:42:05 2019 daemon.info glorytun[3092]: STOPPED tun0
Wed Apr 17 17:42:05 2019 daemon.notice netifd: Network device 'tun0' link is down
Wed Apr 17 17:42:05 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Apr 17 17:42:05 2019 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Apr 17 17:42:05 2019 daemon.notice netifd: omrvpn (3148): udhcpc: SIOCGIFINDEX: No such device
Wed Apr 17 17:42:05 2019 daemon.notice netifd: omrvpn (3148): udhcpc: received SIGTERM
Wed Apr 17 17:42:06 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Apr 17 17:42:06 2019 daemon.info glorytun[16353]: INITIALIZED tun0
Wed Apr 17 17:42:06 2019 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Apr 17 17:42:06 2019 daemon.notice netifd: Network device 'tun0' link is up
Wed Apr 17 17:42:06 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Apr 17 17:42:06 2019 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Apr 17 17:42:06 2019 daemon.notice netifd: omrvpn (16388): udhcpc: started, v1.30.1
Wed Apr 17 17:42:06 2019 daemon.notice netifd: omrvpn (16388): udhcpc: sending discover
Wed Apr 17 17:42:07 2019 daemon.err glorytun[16353]: 46.105.34.59.65001: connected
Wed Apr 17 17:42:07 2019 user.notice omr-bypass: Starting OMR-ByPass...
Wed Apr 17 17:42:08 2019 daemon.info dnsmasq[4716]: read /etc/hosts - 4 addresses
Wed Apr 17 17:42:08 2019 daemon.info dnsmasq[4716]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Wed Apr 17 17:42:08 2019 daemon.info dnsmasq-dhcp[4716]: read /etc/ethers - 0 addresses
Wed Apr 17 17:42:08 2019 user.notice omr-bypass: OMR-ByPass is running
Wed Apr 17 17:42:08 2019 daemon.info glorytun[16353]: STARTED tun0

Ysurac, 2019-04-17:
Strange, what do you have in OpenMPTCProuter status page ?

olaulau, 2019-04-17:
and now, enabling back ss-redir for UDP, it seems to work again ...
rebooted the VPS and the router, UDP doesn't work.

olaulau, 2019-04-17:
[image]

Ysurac, 2019-04-17:
You shouldn't use shadowsocks for UDP.
I will need the config from "Show all settings" in System->OpenMPTCProuter (don't forget to remove server keys).

olaulau, 2019-04-17:
disabling ss-redir for UDP make DNS queries working.
if I'm not mistaken, ss-redir for UDP was enabled by default.

olaulau, 2019-04-17:
All OpenMPTCProuter settings
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_default='0'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='Serveur'
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='2C:4D:54:55:8F:61'
dhcp.@host[0].ip='192.168.1.2'
dhcp.@host[0].gw='192.168.1.254'
dhcp.@host[1]=host
dhcp.@host[1].name='Copieur'
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='00:80:91:CC:9F:D6'
dhcp.@host[1].ip='192.168.1.50'
dhcp.@host[1].gw='192.168.1.254'
dhcp.@host[2]=host
dhcp.@host[2].name='VpnNomade'
dhcp.@host[2].dns='1'
dhcp.@host[2].mac='B8:27:EB:5E:CC:3F'
dhcp.@host[2].ip='192.168.1.245'
dhcp.@host[2].gw='192.168.1.254'
dhcp.@host[3]=host
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='AABBCCDDEEFF'
etherwake.@target[0].wakeonboot='off'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@defaults[0].forward='REJECT'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
[... 965 lines cut ...]
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'

Ysurac, 2019-04-17:
This seems correct.
You have no DNS resolution ? if "curl ipconfig.io" this seems that you have DNS resolution, so UDP seems to work.
1.1.1.1, 9.9.9.9,8.8.8.8,... are used by omr-tracker to test if connection is up or down, so they can give some strange result.

olaulau, 2019-04-17:
DNS resolution through OMR (using OMR as DNS : `nslookup free.fr`) always works great. 
`nslookup free.fr 1.1.1.1` is just an example of outgoing UDP traffic (querying directly an external DNS such as 1.1.1.1).

i've also tester with a netcat client/server in UDP mode :
`nc -ul 6666 `(on remote host)
`nc -u remote.host 6666` (on client behind OMR)

so for you, ss-redir for UDP should be unset, and this should go through glorytun?
I don't remember having changed this, think it was enabled by default, by maybe I'm wrong ...

Ysurac, 2019-04-17:
In your config ss-redir is not used for UDP because rules doesn't forward UDP traffic to ss-redir (it's the default).
If DNS is working, then UDP traffic is working. You can test iperf udp.

olaulau, 2019-04-17:
with a DHCP config, default DNS is OMR.this always works, but this is not real internet UDP traffic through OMR.
when I run `nslookup free.fr 1.1.1.1`, I'm querying directly 1.1.1.1, which is real internet UDP traffic.

But maybe I changed a default setting by mistake, sorry. Maybe I'll try a fresh install tomorrow to be sure.

Ysurac, 2019-04-17:
Maybe a MTU problem.

closed 2019-05-25 by Ysurac