Ysurac/openmptcprouter #440: Failing to connect any longer to MPTCProuter when assigning fixed ip address to client PC

jfh75, 2019-04-18:
## Expected Behavior

  In English please.

## Actual Behavior
On a new opentrouter install with everything let "by default" everything is fine.
Changing routeur LAN IP is ok.
Changing client PC IP address, assigning a fixed address in cgi-bin/luci/admin/network/dhcp is not working. Switching network on/off in ubuntu works, getting the fixed IP, but router cannot be reached smootlhy any more. Internet, LAN devices are reachable, only the router is not reachable.
There seems to be a connectivity problem as : 
I can quite often get the router login page loading
I can rarely move to another page
I can never change anything any more

Firefox error message is "la connexion a été réinitialisée"

## Steps to Reproduce the Problem

  1. assigning fixed DHCP lease with MAC/IP in openmptcprouter
  2.
  3.

## Specifications

  - OpenMPTCProuter version: (0.48)
  - OpenMPTCProuter VPS version: (.91)
  - OpenMPTCProuter platform: (zotac Zbox PC/x86_64)

Ysurac, 2019-04-18:
Check system log. You can also reset browser cache.

jfh75, 2019-04-18:
I have emptied browser cache and it works now.

In the meantime, I have activated ipv6 on my FFTH VPS.

Anyway, it is working fine now.

Thank you for your kind help

Best regards

JFH

closed 2019-04-18 by Ysurac