Ysurac/openmptcprouter #441: When there is a wan network quality is bad, it will affect all download speeds, but the upload speed is still very good.

suyuan168, 2019-04-20:
## Expected Behavior
When there is a wan network quality is bad, it will affect all download speeds, but the upload speed is still very good.

## Actual Behavior
The correct openmptcprouter settings, I have wan1 and wan2 and wan3. 3 wan ports. The mptcp settings are enabled for all three wans. My wan1 is very good quality ADSL line, uploading 20M/S download has 50M/S, delay is 40ms. My wan2 is a usb 4g signal is very poor, upload 10M / S, download 2M / s, delay 1000ms. My wan3 is also adsl, upload 10M / S download has 20M / S delay 10ms.

Now the problem is that there are many ways to wan, when one of the wan download speed is very bad, it will make the openmptcprouter overall download speed slower, but does not affect the upload speed.
Is there any way to make a wan with a very bad download speed affect other wan speeds?
The same is true without affecting the upload speed. I think there is definitely a reason.

## Steps to Reproduce the Problem

  1. If I follow the correct upload of 40M/S, the download has 72M/s. In theory, this is the case.
  2. But the problem is, if my wan2 signal is very bad, this will drag down the download speed of the entire openmptcprouter. The download speed will be reduced by half, which is about 30M/S. However, the upload speed does not affect 40M/S.
  3. If I am disconnecting the network cable of wan2 now. Then the upload speed is 30M/S. The download speed is 70M/S.

## Specifications

  - OpenMPTCProuter version: (Last is not a version) 4.8
  - OpenMPTCProuter VPS version: (Last is not a version) 0.922
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) RPI3

Ysurac, 2019-04-20:
I think the problem is more about RTT than download speed. If the RTT is not stable at all this can make this problem even if MPTCP default scheduler should always choose connection with best RTT...
This may be better in a next release with 4.19 kernel.

suyuan168, 2019-04-20:
However, he will not affect the upload speed. The upload speed is always the best. It only affects the download speed, which is very annoying.

suyuan168, 2019-04-20:
The next version of openmptcprouter is 4.19 kernel?

Ysurac, 2019-04-20:
No. Later.

suyuan168, 2019-04-20:
The upload speed is perfect. Can the download speed be unaffected by the speed difference?If there is a slow signal difference, it will affect all speeds. Is there a solution now?

rstanislav, 2019-04-21:
> No. Later.

When to expect release with 4.19 kernel ? There could be great speed improvement when using wireless links, because of new BLEST scheduler

Ysurac, 2019-04-21:
As soon as possible. I'm testing 4.19 patches for RPI now. Then I need to test patches for mvebu (WRT3200ACM and WRT32X). This seems difficult for BPI-R2...

suyuan168, 2019-04-22:
BPI-R2 does not seem to be produced.

rstanislav, 2019-04-22:
> As soon as possible. I'm testing 4.19 patches for RPI now. Then I need to test patches for mvebu (WRT3200ACM and WRT32X). This seems difficult for BPI-R2...

Can you please share link to 32bit rpi image ? I can help with testing i think

suyuan168, 2019-04-22:
I also need RPI 2 haha

suyuan168, 2019-04-24:
hi，https://ylim.mybluemix.net/mptcp_ecf/

suyuan168, 2019-04-24:
I am trying to change the scheduler BBR/CUBIC/OLIA. It seems that there is no impact on the results and performance. Didn't change it.

Ysurac, 2019-04-24:
MPTCP ECF is not yet available for MPTCP 0.94/0.95.
BLEST may help, I'm testing and fixing images with kernel 4.19 and MPTCP 0.95 (that contain BLEST scheduler).

suyuan168, 2019-04-24:
This is really great, look forward to

closed 2019-06-05 by suyuan168