Ysurac/openmptcprouter #442: Use OpenMPTCProuter without VPS

jayanta525, 2019-04-22:
## Expected Behavior

Use OpenMPTCProuter without a VPS on the WAN end. I have multiple 20Mbit lines with the same public IP (college campus internet, limits per MAC 20Mbit/10GB/24hrs). I want to have Multi-path TCP enabled for the interfaces to have aggregated bandwidth. I don't want to use a VPS or MLVPN or GlorytunUDP as I have the same public IP for all the interfaces and I think it shouldn't create any issues. 

**How to use OpenMPTCProuter with only MPTCP enabled on the WAN interfaces?**

## Actual Behavior

Configuration without a VPS simply doesn't work.

## Steps to Reproduce the Problem

  1. Skip VPS configuration on the router (blank fields).
  2. Create multiple MACVLAN interfaces with different MAC address (set to dynamic IP, obtains a lease from DHCP).
  3. Enable MPTCP on the interfaces with one set as master.
  4. No Internet Connectivity.

## Specifications

  - OpenMPTCProuter version: 0.48
  - OpenMPTCProuter VPS version: N/A
  - OpenMPTCProuter platform: RPI3

Thanks.

Ysurac, 2019-04-22:
Even if you have same public IP for all interfaces, you need a VPS to aggregate bandwidth.

jayanta525, 2019-04-22:
Sorry, I don't know much about the working of MPTCP. So, basically, if I want to watch YouTube with MPTCP enabled on my router, the YouTube servers should also be MPTCP capable?

Ysurac, 2019-04-22:
If you don't use a VPS and want aggregated bandwidth, yes.

jayanta525, 2019-04-22:
What if I have an OpenVPN connection to a VPS, will that provide aggregated bandwidth?

Ysurac, 2019-04-22:
no

jayanta525, 2019-04-22:
Okay, got it. Thanks

closed 2019-04-22 by jayanta525

jayanta525, 2019-04-25:
Last time I tried configuration in GoogleVM, I had issues with connections. Can someone confirm if DigitalOcean Standard Droplets works properly?